Koenkk/zigbee2mqtt #24206: Log is flooded with this error

oli737, 2024-10-04:
### What happened?

z2m: Publish 'set' 'state' to 'Stube 1.3' failed: 'Error: ZCL command 0x0017880108bab2b1/11 genLevelCtrl.moveToLevelWithOnOff({"level":0,"transtime":50}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '30722'.)'
[2024-10-04 08:56:46]

And

z2m: Publish 'set' 'brightness' to 'Stube 1.1' failed: 'Error: ZCL command 0x0017880108baa365/11 genLevelCtrl.moveToLevelWithOnOff({"level":15,"transtime":10}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '60861'.)'

Just an example the log is full. I am using an zigbee dongle E with Philips Hue E14 ans E27 Lamps on last Firmware. Some lamps did stay randomly turned on and not switching off as expected.

Version is 1.40.2-1

### How to reproduce it (minimal and precise)

I did try reboot, restart all zigbee devices but the log is still full of errors and lamps won't turn off as expected.

### Zigbee2MQTT version

1.40.2 commit: unknown

### Adapter firmware version

7.4.3 [GA]

### Adapter

EmberZNet

### Setup

Home Assistant OS

Nerivec, 2024-10-05:
That error occurs when the device in question cannot be reached by the adapter.
Can you try updating to 7.4.4. Some users who had issues with Dongle E fixed them with that new firmware version.
If it still doesn't work, please provide a `debug` log from starting z2m to when the error occurs.

senna1992, 2024-10-07:
Same issue here. HA yellow. FW 7.4.4 did not fix it.

`error: 	z2m: Publish 'set' 'state' to 'Licht.3' failed: 'Error: ZCL command 0xa4c1389de1a8b1f5/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":58444,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":199},"zclSequence":82,"commandIdentifier":11} timed out after 10000ms)'`

Similar error to many devices. All working just fine.

Also getting this from time to time on some devices:

`2024-10-07 03:20:06] error: 	z2m: Publish 'get' 'power_on_behavior' to 'Licht.9' failed: 'Error: ZCL command 0x9035eafffec6856c/1 genOnOff.read(["startUpOnOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'`

Nerivec, 2024-10-07:
For `UNSUPPORTED_ATTRIBUTE`, if it does this every time the same `read` is requested, it just means the device does not support that specific configuration. Usually, it's because you have a model variant that doesn't support this, compared to the "base variant" supported by Z2M.

oli737, 2024-12-11:
The problem is still not fixed for me. 
No on:

Zigbee2MQTT-Version
1.42.0 commit: unknown
Coordinator-Typ
EmberZNet
Coordinator-Version
7.4.5 [GA]

Nerivec, 2024-12-11:
Please see my first post.

oli737, 2024-12-11:
[log.log](https://github.com/user-attachments/files/18099413/log.log)

Nerivec, 2024-12-11:
This is not a [`debug` log](https://www.zigbee2mqtt.io/guide/configuration/logging.html#debugging), but from the errors, it looks like your network is overloaded (returning `BUSY` often).
- Do you have any very spammy devices (Tuya, mmWave, etc.)?
- Do you have any group-based triggers (large groups, or triggered often by automation or other)?

You can try updating to 8.0.2 firmware ([darkxst](https://github.com/darkxst/silabs-firmware-builder/releases) or [mine](https://github.com/Nerivec/silabs-firmware-builder/releases)), it sets higher configs in-firmware for the adapter, which can usually help with that sort of issue. Although, it too has its limits, be sure to check the above two points, and try to reduce spam as much as possible (it will help with the overall network's health too).
_[MG24-based adapters](https://www.zigbee2mqtt.io/guide/adapters/emberznet.html#recommended) provide higher specs, which also help in some cases, but of course, that requires buying a new adapter._ 😉

oli737, 2024-12-11:
I updated to version 8 and now everything works like a charm 😀 thank you very much !!!