Koenkk/zigbee2mqtt #24217: Home Assistant button triggers broke in `1.40.2`

alec-pinson, 2024-10-05:
### What happened?

This appeared as `Unknown`, rolling back to `1.40.1` fixed the issue.
[image]

I'm using a [Aqara WXKG11LM](https://www.zigbee2mqtt.io/devices/WXKG11LM.html#aqara-wxkg11lm) button

### What did you expect to happen?

It should show `single`, `double` etc.

### How to reproduce it (minimal and precise)

Upgrade `1.40.1` -> `1.40.2` and it breaks the button

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

na

### Adapter

na

### Setup

Home Assistant

Koenkk, 2024-10-06:
Could you provide the debug log when generating actions on the device?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

djurny, 2024-10-06:
Hi there,
One question; I had similar experience with home assistant, for example I would not see a certain action in that drop-down, but after actually triggering the action from the device, it started showing in HASS.
Is it the same with your situation? Do all the actions show in the drop-down after performing the action on the device itself?
(Not saying that's the way it is supposed to work by the way, just checking it that can help you along right now.)
Groetjes,

Skynetnexus, 2024-10-07:
I had the same issue, all my button actions became unavailable after installing  Home Assistant Core 2024.10.1, revered to 2024.9.3 and all good again - I tried updating twice, both times the same issue ( running Zigbee2MQTT version
 1.40.0-dev commit
 Coordinator revision
 20240710

antbarney, 2024-10-07:
I also noticed this after just updating to 2024.10.1, Button Actions all work tho but show an unknown status, when HA used to list them as "None" according to history.
I think It might be some change of handling in HA from when the action is published as null/blank but that is just a guess. Stuff works fine tho, if I did not have an automation that shows me unknown/unavailable I would have never noticed an issue.

[image]

Skynetnexus, 2024-10-08:
> I also noticed this after just updating to 2024.10.1, Button Actions all work tho but show an unknown status, when HA used to list them as "None" according to history. I think It might be some [...]

Thank you for pointing this out, I am running a template sensor  (  [unavailable-entities-sensor] ) (https://github.com/jazzyisj/unavailable-entities-sensor) and it picked up the button click sensors as being unavailable. Initially I was upgrading remotely from work, so I upgraded whilst I was home and able to test the buttons. Yes the button still work after the upgrade.

nickw444, 2024-10-22:
I've also noticed this for [WXKG11LM](https://www.zigbee2mqtt.io/devices/WXKG11LM.html#aqara-wxkg11lm). The hold action is reported as `""` in Home Assistant:

```yaml
  new_state:
    entity_id: sensor.nicks_bedside_button_action
    state: ""
    attributes:
      action: ""
      battery: 97
      click: null
      device_temperature: 28
      last_seen: "2024-10-22T10:39:52.782Z"
      linkquality: 47
      power_outage_count: 145
      temperature: 23
      voltage: 2995
      icon: mdi:gesture-double-tap
      friendly_name: Nicks Bedside Button Action
```

alec-pinson, 2024-10-26:
Sorry didn't have time to come back to this, I've updated to latest version of this + HA and everything is fine now.

closed 2024-10-26 by alec-pinson