Koenkk/zigbee2mqtt #24218: ts011f_plug_3 not providing power values

senrabyar, 2024-10-05:
### What happened?

Similar to other similar issues which I think I have exhausted trying.

I have 14 TS011F plugs working OK (4 are plug_1, 10 are plug_3).  I purchased a further batch of four and two work fine, but the other two have no power.  The working ones have the expected 1.0.5 firmware, but the broken have 0122052017 firmware.   The physical units are identical, and labels give model WHDZ03.

The plug is the UK 3 pin version.

I have tried changing the poll interval from blank to both 30 and '30' 

OTA says firmware up to date.

The stated  Zigbee Manufacturer is  _TZ3000_o81odtrp  which is not the same as all the working plugs, and in correspondence on the HA Community is not known by other people.
 
Is this a new device / new firmware / corrupt firmware ?

### What did you expect to happen?

Expected same monitoring as the the other TS011F_plug_3 

### How to reproduce it (minimal and precise)

The behaviour is detected by simply pairing device, and viewing in Z2M Devices or Dashboard

(Also fails on ZHA / Conbee )

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20210708

### Adapter

Sonoff ZB Dongle P

### Setup

HA on Pi5

### Debug log

Sorry I cannot find data/log in any HA folder

Here's the stuff Z2M reports

Device type       Router
Zigbee Model   TS011F
Zigbee Manufacturer    _TZ3000_o81odtrp
Description       Smart plug (with power monitoring by polling)
Support status   Supported

IEEE Address      0xa4c138ae715a78fb
Network address  0xC542 / 50498
Firmware version   0122052017
Manufacturer  [Tuya]
Model    [TS011F_plug_3]
Power   
Interview completed   True

{
    "child_lock": "UNLOCK",
    "indicator_mode": "off/on",
    "last_seen": "2024-10-05T10:52:27.304Z",
    "linkquality": 102,
    "power_outage_memory": "off",
    "state": "OFF",
    "update": {
        "installed_version": 268449287,
        "latest_version": 268449287,
        "state": "idle"
    },
    "update_available": false,
    "current": null,
    "energy": null,
    "power": null,
    "voltage": null
}

senrabyar, 2024-10-05:
Further info ..

Discovered that configuration.yaml already had an "advanced:" line, so that prevented the logging debug from working. 

Now found a file in \log\datestamp\log.log which contains what you want see - debug errors !

The file is heavily polluted by the excess data rate the air quality devices produce - but that is a different problem.
[log.log](https://github.com/user-attachments/files/17266965/log.log)

See attached

briodan, 2024-11-03:
seeing the same issue with a US version of the plug

[image]
[image]
[image]

measurement interval set to 60 in configuration.yaml

    measurement_poll_interval: 60

senrabyar, 2024-11-03:
Just to confirm - the problem is still there in 1.41.0

1.41.0 however has broken some other devices which I'll detail in another post.

Koenkk, 2024-11-03:
Could you provide the debug log from starting z2m + 5 minutes after this and the `data/database.db` entry of this device?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

senrabyar, 2024-11-03:
[xlog.log](https://github.com/user-attachments/files/17611213/xlog.log)

This is a redacted version - I've removed things obviously to do with the other Zigbee devices, and changed device friendly names.  The log around the reported error is however all the lines ...

[2024-11-03 14:17:49] debug: 	zh:controller:endpoint: Error: ZCL command 0xa4c138ae715a78fb/1 haElectricalMeasurement.read(["rmsVoltage","rmsCurrent","activePower"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28)
    at Endpoint.zclCommand (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26)
    at Endpoint.read (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:446:29)
    at poll (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:182:13)
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman-converters/src/lib/utils.ts:54:25)

THX

R

Koenkk, 2024-11-05:
`UNSUPPORTED_ATTRIBUTE` indicates the device does not support it. Can you try reading them one by one via the dev console?

[image]

senrabyar, 2024-11-11:
Not surprisingly  I don't get anything from this.  The cluster haElectricalMeasurement doesn't exist.  

Comparing with the working plugs in the same batch, I see very different Clusters.    

The functioning plugs do have an input cluster haElectricalMeasurement on endpoint 1

But the non-functioning have endpoint 242 with an output cluster 'greenPower'   

[image]

[image]

I'm assuming this is a complete failure in the plugs firmware.   Can it be reflashed ?

senrabyar, 2024-11-11:
and for what it's worth - I've bought an actual Zigbee Hub so I can test devices directly to Tuya (instead of using Sonoff Coordinator and Z2M and HA) - and guess what - the plug only shows a switch for PowerOn and PowerOff in the Tuya Smart Things app.    (there's ChildLock and Indicator mode too - but that's tinkering at the edge)   No energy stats.

senrabyar, 2025-01-11:
FWIW  I eventually got my money back for the two non-reporting switches.     Thanks for the help which became very useful with the arrival of Z2M 2.0.  I've learnt so much more about MQTT and can now even make IKEA buttons work in NodeRed without HomeAssistant involved.

closed 2025-01-11 by senrabyar