Koenkk/zigbee2mqtt #24220: [New device support]: Urgent Please help with TS0726 Switch Error

horacecar, 2024-10-05:
### What happened?

[zStack3x0](z2m: Failed to configure 'Office_A_Wall_Switch_1', attempt 3 (TypeError: Cannot read properties of undefined (reading 'bind') at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:6217:33) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Configure.configure (/app/lib/extension/configure.ts:130:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:75:17)))

----
z2m: Failed to configure 'Office_A_Wall_Switch_2', attempt 3 (TypeError: Cannot read properties of undefined (reading 'bind') at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:24) at Object.configure (/app/node_modules/zigbee-herdsman-converters/src/devices/tuya.ts:6217:33) at processTicksAndRejections (node:internal/process/task_queues:95:5) at Configure.configure (/app/lib/extension/configure.ts:130:13) at Immediate.<anonymous> (/app/lib/extension/configure.ts:75:17))

I purchased a one-way and two-way composite switch (that is, Zigbee with scene function, which can switch scenes and switch functions in the Z2M background). Regardless of the 1Switch or 2switch, it is recognized as a 4 switch in the Z2M background. Switch, I guess there is a possibility that the system keeps reading the third and fourth switches that do not exist, causing Z2M to report a large number of errors?  And the Config switch keeps popping up, causing the Z2M system to be extremely slow.

[image]
[image]
[image]

[image]
[image]

### Zigbee2MQTT version

1.40.2-dev commit: 063d462

### Adapter firmware version

20230507

### Adapter

zStack3x0

### Setup

Z2M with docker on Synology ds 224+

### Debug log

[database.db.zip](https://github.com/user-attachments/files/17266131/database.db.zip)

horacecar, 2024-10-09:
is there any update?

horacecar, 2024-10-12:
Export From Tuya  DPID DATA " DPID  scene1:1 ;scene2:2; backlight switch:36 ;relay status all:38 ; relay status1:39 ; relay status2:40, sence id & group id:17; mode 1:18 ; mode 2: 19; switch1:24 switch2:25;countdown1:30 countdown2:31. " ，

	scene_1	Enum	
{
  "range": [
    "scene"
  ]
}
	scene_2	Enum	
{
  "range": [
    "scene"
  ]
}
 scene_id_group_id	Raw	
{}
	mode_1	Enum	
{
  "range": [
    "switch_1",
    "scene_1"
  ]
}
	mode_2	Enum	
{
  "range": [
    "switch_2",
    "scene_2"
  ]
}
	switch_1	Boolean	
"{true,false}"
	switch_2	Boolean	
"{true,false}"
	countdown_1	Integer	
{
  "unit": "s",
  "min": 0,
  "max": 43200,
  "scale": 0,
  "step": 1
}
countdown_2	Integer	
{
  "unit": "s",
  "min": 0,
  "max": 43200,
  "scale": 0,
  "step": 1
}
	switch_backlight	Boolean	
"{true,false}"
	relay_status	Enum	
{
  "range": [
    "power_off",
    "power_on",
    "last"
  ]
}
relay_status_1	Enum	
{
  "range": [
    "power_off",
    "power_on",
    "last"
  ]
}
	relay_status_2	Enum	
[image]
[image]
[image]

{
  "range": [
    "power_off",
    "power_on",
    "last"
  ]
}"

horacecar, 2024-10-12:
@Koenkk Could you please have a look on this? thank you so much!!!!!

Koenkk, 2024-10-13:
Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/f42461ae38525b5f3dd1054a85da043c): 

- save this as file next to `configuration.yaml` as `ext_converter.js`
- add it to `configuration.yaml`:
```yaml
external_converters:
  - ext_converter.js
```
- start z2m, 
- in the frontend, check if the description of this device now ends with `CUSTOM` (this indicates the external converter has been loaded correctly)
- check if issue is fixed

horacecar, 2024-10-14:
> Could you check if the issue is fixed with the following [external converter](https://gist.github.com/Koenkk/f42461ae38525b5f3dd1054a85da043c):
>
> * save this as file next to `configuration.yaml` as [...]

Hello Koenkk, first of all, thank you very much for your help. I Really Appreciate that.!!!!!!!

1. For the 2-Gang {fingerprint('TS0726', ['_TZ3002_xw7xz6ys']}I have tested this code and generally no errors are reported, but there is a small problem with the setting scene mode of Switch 2. No matter whether the Z2M button state of Switch 1 is on or off, when the scene mode is set, the backlight of the button on the left side of the switch will remain orange. When I click it with my hand, the white light will flash four times (this should be correct) . But the right side. After setting the scene mode in Z2M for Switch 2, the backlight of the button on the right side of the panel will turn white when clicked to turn on in the Z2M background, and the backlight will turn orange when clicked to turn off. When I click the switch with my hand The right button does not keep the backlight orange like the left button and flashes four times, telling it is a scene mode. When I physically click the right button with my hand and switch the panel, the color of the light does not change or flash.

2. 1-Gang Zigbee Model:TS0726;Zigbee Manufacturer:_TZ3002_m9sbhhfs .The test is perfect for now 

3. I have a (Same appearance and series) four-way switch  {Zigbee Model:TS0726;Zigbee Manufacturer:_TZ3002_hlqaknde} of the same series as shown in the picture (the four switch buttons independently support changing to scene mode). The same situation as 2-GANG just now. Only the first channel scene mode backlight is normal. After the second, third, and fourth switches are set to scene mode, click the button on the switch panel with my hand. The backlight remains orange and does not flash four times (the Z2M switch is set to off). The backlight remains white and does not flash four times. (Z2M switch is set to ON).The following is the Tuya data for a four-way switch.

`
{
  "result": {
    "category": "cjkg",
    "functions": [
      {
        "code": "scene_1",
        "dp_id": 1,
        "type": "Enum",
        "values": "{\"range\":[\"scene\"]}"
      },
      {
        "code": "scene_2",
        "dp_id": 2,
        "type": "Enum",
        "values": "{\"range\":[\"scene\"]}"
      },
      {
        "code": "scene_3",
        "dp_id": 3,
        "type": "Enum",
        "values": "{\"range\":[\"scene\"]}"
      },
      {
        "code": "scene_4",
        "dp_id": 4,
        "type": "Enum",
        "values": "{\"range\":[\"scene\"]}"
      },
      {
        "code": "scene_id_group_id",
        "dp_id": 17,
        "type": "Raw",
        "values": "{}"
      },
      {
        "code": "mode_1",
        "dp_id": 18,
        "type": "Enum",
        "values": "{\"range\":[\"switch_1\",\"scene_1\"]}"
      },
      {
        "code": "mode_2",
        "dp_id": 19,
        "type": "Enum",
        "values": "{\"range\":[\"switch_2\",\"scene_2\"]}"
      },
      {
        "code": "mode_3",
        "dp_id": 20,
        "type": "Enum",
        "values": "{\"range\":[\"switch_3\",\"scene_3\"]}"
      },
      {
        "code": "mode_4",
        "dp_id": 21,
        "type": "Enum",
        "values": "{\"range\":[\"switch_4\",\"scene_4\"]}"
[... 472 lines cut ...]
            [40, 'relay_status_2', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 2
        ],
    },
};

module.exports = definition;
`

5. Is there any way to set the first and second switches of the same switch in the Home Assistant Entity ID in Z2M? For example, I want the Entity ID of switch 1 in the Home Asssitant to be named light.bedroom_led_1, and the Entity ID of switch 2 in the Home Asssitant to be named light.bedroom_led_2. Is there any way to set it directly in the CONFIG of Z2M? Sync to Home Assistant to quickly recover when disaster strikes? When my Config is set as follows, the prefix of the switch is displayed in HA, light.living_room_2_gang_storeroom_toilet_ceiling_light_1, but I only want light.storeroom_toilet_ceiling_light_1. Is it possible to achieve this?

[image]

`
  '0xa4c138bea73c979b':
    friendly_name: Living_Room_2_Gang
    homeassistant:
      switch_l2:
        type: light
        object_id: storeroom_toilet_ceiling_light_1
        name: storeroom_toilet_ceiling_light_1
        value_template: null
        state_value_template: '{{ value_json.state_l2 }}'

`
Thank you for your patience in watching and answering. Thank you from the bottom of my heart. You Save my day.

Koenkk, 2024-10-15:
- 1/3: I'm not sure if this is the expected behaviour of this switch @kirovilya do you still have this device, if yes, can you confirm this?
- 2: great!
- 4: I'm not sure if this device uses the Tuya datapoint API, it seems to use standard Zigbee commands even though the Tuya portal shows datapoints (it's not the first time I saw this)
- 5: You can change the entity ids directly from HA

horacecar, 2024-10-15:
> * 1/3: I'm not sure if this is the expected behaviour of this switch @kirovilya do you still have this device, if yes, can you confirm this?
> * 2: great!
> * 4: I'm not sure if this device uses the [...]

[image]

https://github.com/user-attachments/assets/2b56146a-62a9-4e91-b0b8-ca3c5f81041a

So the scene mode of the switch backlight, there's no way to fix? Because only L1 responds normally, L2, L3, and L4 are not correct, which makes it very strange to use.

But I have tried pairing this switch with Tuya and using SmartLife OR Tuya. After changing the button to a scene function key, the backlight is indeed normal, and there are also countdown functions and backlight settings. These functions are even in Tuya. Even though the operating platform has found its DPID, it is still unable to add these functions to Home Assistant through Z2M? But my understanding is that as long as the product has this function and its DP ID is found, it should be possible to add these functions to Z2M, right? Because the Tuya cloud platform can also control and debug switch products through relevant IDs?

Koenkk, 2024-10-16:
Can you try with your converter based on dpID and change:

```
[19, 'mode_2', tuya.valueConverterBasic.lookup({'switch_2': 0, 'scene_2': 1})],
```

to

```
[19, 'mode_2', tuya.valueConverterBasic.lookup({'switch': tuya.enum(0), 'scene': tuya.enum(1)})],
```

Does it work then? If not, can you provide the debug logging when setting it?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

horacecar, 2024-10-17:
> Can you try with your converter based on dpID and change:
>
> ```
> [19, 'mode_2', tuya.valueConverterBasic.lookup({'switch_2': 0, 'scene_2': 1})],
> ```
>
> to
>
> ```
> [19, 'mode_2', [...]

I'll get back to you after testing. Do you mean the external converter I compiled myself? However, there is a problem with this external converter. Except for the on and off functions of the smart switch, all other functions cannot be used.

horacecar, 2024-10-17:
> Can you try with your converter based on dpID and change:
>
> ```
> [19, 'mode_2', tuya.valueConverterBasic.lookup({'switch_2': 0, 'scene_2': 1})],
> ```
>
> to
>
> ```
> [19, 'mode_2', [...]

I temporarily used a four-way switch for testing. It is the same brand and series as the previous two-way switch. I used this program code. Except that the function of the switch is normal, the other functions are such as Countdown, Power_on_behave, and backlight settings are all unavailable.
[image]
[image]
[image]
[image]

I am using the following code

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: "TS0726",
            manufacturerName: "_TZ3002_hlqaknde",
        }
    ],
    model: 'TS0726',
    vendor: 'TuYa',
    description: 'TuYa Two Gang Switch (with scene control and backlight) - 1018',
    powerSource: 'Mains (single phase)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [
        tz.on_off,
        tuya.tz.datapoints
    ],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.action(['scene_1', 'scene_2']),  // Scene control
        e.linkquality(),
        tuya.exposes.switch().withEndpoint('switch_1'),  // Switch 1
        tuya.exposes.switch().withEndpoint('switch_2'),  // Switch 2
        tuya.exposes.countdown().withEndpoint('countdown_1'),  // Countdown 1
        tuya.exposes.countdown().withEndpoint('countdown_2'),  // Countdown 2
        e.binary('switch_backlight', ea.STATE_SET, 'ON', 'OFF').withDescription('Backlight Switch'),  // Backlight switch
        e.enum('relay_status', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior'),  // Global power-on behavior
        e.enum('relay_status_1', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior for Switch 1'),  // Power-on behavior for Switch 1
        e.enum('relay_status_2', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior for Switch 2'),  // Power-on behavior for Switch 2
        e.enum('mode_1', ea.STATE_SET, ['switch_1', 'scene_1']).withDescription('Mode for Switch 1'),  // Mode for Switch 1
        e.enum('mode_2', ea.STATE_SET, ['switch_2', 'scene_2']).withDescription('Mode for Switch 2'),  // Mode for Switch 2
    ],
    endpoint: (device) => {
        return {
            'switch_1': 1,
            'switch_2': 2
        };
    },
    meta: {
        multiEndpoint: true,
        disableDefaultResponse: true,
        timeout: 5000,
        tuyaDatapoints: [
            [1, 'scene_1', tuya.valueConverter.static('scene')],
            [2, 'scene_2', tuya.valueConverter.static('scene')],
            [17, 'scene_id_group_id', tuya.valueConverter.raw],
            [18, 'mode_1', tuya.valueConverterBasic.lookup({'switch_1': 0, 'scene_1': 1})],
            [19, 'mode_2', tuya.valueConverterBasic.lookup({'switch': tuya.enum(0), 'scene': tuya.enum(1)})],
            [24, 'switch_1', tuya.valueConverter.onOff],
            [25, 'switch_2', tuya.valueConverter.onOff],
            [30, 'countdown_1', tuya.valueConverter.countdown],  // Countdown 1
            [31, 'countdown_2', tuya.valueConverter.countdown],  // Countdown 2
            [36, 'switch_backlight', tuya.valueConverter.onOff],  // Backlight switch
            [38, 'relay_status', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior
            [39, 'relay_status_1', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 1
            [40, 'relay_status_2', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 2
        ],
    },
};

module.exports = definition;
```

horacecar, 2024-10-17:
> Can you try with your converter based on dpID and change:
>
> ```
> [19, 'mode_2', tuya.valueConverterBasic.lookup({'switch_2': 0, 'scene_2': 1})],
> ```
>
> to
>
> ```
> [19, 'mode_2', [...]

And here is the db and log you need, thank you so much !!

DB 
```
{"id":14,"type":"Coordinator","ieeeAddr":"0x00124b0029dfdf25","nwkAddr":0,"manufId":7,"epList":[1,2,3,4,5,6,8,10,11,12,13,47,110,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"10":{"profId":260,"epId":10,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[1281,10],"outClusterList":[1280,1282],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"47":{"profId":260,"epId":47,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"interviewCompleted":true,"meta":{}}
{"id":15,"type":"Router","ieeeAddr":"0xa4c138bea73c979b","nwkAddr":51693,"manufId":4417,"manufName":"_TZ3002_xw7xz6ys","powerSource":"Mains (single phase)","modelId":"TS0726","epList":[1,2,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u001f\n�.\u0012","65506":54,"65508":0,"stackVersion":0,"dateCode":"","appVersion":82}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":0,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dfdf25","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"2":{"profId":260,"epId":2,"devId":4,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchMode":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dfdf25","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1620506091},"lastSeen":1729027701294}
{"id":16,"type":"Router","ieeeAddr":"0xa4c13813a475e3d0","nwkAddr":51226,"manufId":4417,"manufName":"_TZ3002_hlqaknde","powerSource":"Mains (single phase)","modelId":"TS0726","epList":[1,2,3,4,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":4,"inClusterList":[3,4,5,6,57344,57345,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000e\u0000\u0000\u0000\u0012","65506":54,"65508":0,"65534":0,"modelId":"TS0726","manufacturerName":"_TZ3002_hlqaknde","powerSource":1,"zclVersion":3,"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"65278":0,"onOff":1,"onTime":0,"offWaitTime":0,"tuyaBacklightMode":1,"moesStartUpOnOff":2,"tuyaBacklightSwitch":1}},"manuSpecificTuya_3":{"attributes":{"65278":0,"powerOnBehavior":0,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dfdf25","endpointID":1}],"configuredReportings":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":4,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"65278":0,"onOff":0,"onTime":0,"offWaitTime":0}},"manuSpecificTuya_3":{"attributes":{"65278":0,"powerOnBehavior":0,"switchMode":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dfdf25","endpointID":1}],"configuredReportings":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":4,"inClusterList":[4,5,6,57345],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"65278": [... cut ...] er":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dfdf25","endpointID":1}],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":85,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":710755415},"lastSeen":1728911203864}
```
[log.log](https://github.com/user-attachments/files/17421486/log.log)

Koenkk, 2024-10-18:
Try changing `tuya.exposes.countdown().withEndpoint('countdown_1')` to `tuya.exposes.countdown().withEndpoint('1')`

horacecar, 2024-10-18:
> Try changing `tuya.exposes.countdown().withEndpoint('countdown_1')` to `tuya.exposes.countdown().withEndpoint('1')`

Just tried you telling with using this converter with no success.  In addition, I would like to ask you a question a few days ago. Once the button of the two-way or four-way switch is converted to scene mode, it will not always maintain the orange light like the first channel, and when you press it with your hand, Is there a solution to the white light that flashes four times?
```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: "TS0726",
            manufacturerName: "_TZ3002_hlqaknde",
        }
    ],
    model: 'TS0726',
    vendor: 'TuYa',
    description: 'TuYa Two Gang Switch (with scene control and backlight) - 1019',
    powerSource: 'Mains (single phase)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [
        tz.on_off,
        tuya.tz.datapoints
    ],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.action(['scene_1', 'scene_2']),  // Scene control
        e.linkquality(),
        tuya.exposes.switch().withEndpoint('switch_1'),  // Switch 1
        tuya.exposes.switch().withEndpoint('switch_2'),  // Switch 2
        tuya.exposes.countdown().withEndpoint('1'),  // Countdown 1
        tuya.exposes.countdown().withEndpoint('countdown_2'),  // Countdown 2
        e.binary('switch_backlight', ea.STATE_SET, 'ON', 'OFF').withDescription('Backlight Switch'),  // Backlight switch
        e.enum('relay_status', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior'),  // Global power-on behavior
        e.enum('relay_status_1', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior for Switch 1'),  // Power-on behavior for Switch 1
        e.enum('relay_status_2', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior for Switch 2'),  // Power-on behavior for Switch 2
        e.enum('mode_1', ea.STATE_SET, ['switch_1', 'scene_1']).withDescription('Mode for Switch 1'),  // Mode for Switch 1
        e.enum('mode_2', ea.STATE_SET, ['switch_2', 'scene_2']).withDescription('Mode for Switch 2'),  // Mode for Switch 2
    ],
    endpoint: (device) => {
        return {
            'switch_1': 1,
            'switch_2': 2
        };
    },
    meta: {
        multiEndpoint: true,
        disableDefaultResponse: true,
        timeout: 5000,
        tuyaDatapoints: [
            [1, 'scene_1', tuya.valueConverter.static('scene')],
            [2, 'scene_2', tuya.valueConverter.static('scene')],
            [17, 'scene_id_group_id', tuya.valueConverter.raw],
            [18, 'mode_1', tuya.valueConverterBasic.lookup({'switch_1': 0, 'scene_1': 1})],
            [19, 'mode_2', tuya.valueConverterBasic.lookup({'switch': tuya.enum(0), 'scene': tuya.enum(1)})],
            [24, 'switch_1', tuya.valueConverter.onOff],
            [25, 'switch_2', tuya.valueConverter.onOff],
            [30, 'countdown_1', tuya.valueConverter.countdown],  // Countdown 1
            [31, 'countdown_2', tuya.valueConverter.countdown],  // Countdown 2
            [36, 'switch_backlight', tuya.valueConverter.onOff],  // Backlight switch
            [38, 'relay_status', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior
            [39, 'relay_status_1', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 1
            [40, 'relay_status_2', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 2
        ],
    },
};

[... 81 lines cut ...]
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1352:27)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:306:52)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:144:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:150:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:147:27)
    at WebSocket.emit (node:events:517:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:517:28)
[2024-10-19 03:22:24] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/0xa4c13813a475e3d0/set' with data '{"countdown_countdown_2":3}'
[2024-10-19 03:22:24] debug: 	z2m: Publishing 'set' 'countdown_countdown_2' to '0xa4c13813a475e3d0'
[2024-10-19 03:22:24] error: 	z2m: Publish 'set' 'countdown_countdown_2' to '0xa4c13813a475e3d0' failed: 'Error: No datapoint defined for 'countdown_countdown_2''
[2024-10-19 03:22:24] debug: 	z2m: Error: No datapoint defined for 'countdown_countdown_2'
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/src/lib/tuya.ts:1352:27)
    at Publish.onMQTTMessage (/app/lib/extension/publish.ts:306:52)
    at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:23)
    at EventEmitter.emit (node:events:529:35)
    at EventBus.emitMQTTMessage (/app/lib/eventBus.ts:144:22)
    at MQTT.onMessage (/app/lib/mqtt.ts:150:27)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.ts:147:27)
    at WebSocket.emit (node:events:517:28)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:1220:20)
    at Receiver.emit (node:events:517:28)
```

Koenkk, 2024-10-19:
In the log you provided, does setting the countdown for endpoint 1 work?

horacecar, 2024-10-19:
> In the log you provided, does setting the countdown for endpoint 1 work?

Nothing happened. And no error.

horacecar, 2024-10-19:
Also mode setting to switch or scene is error. This converter only switch 1 and switch turn on and off is working
Error 2024-10-20 02:54:43z2m: Publish 'set' 'mode_2' to '0xa4c13813a475e3d0' failed: 'Error: Value: 'switch_2' not found in: [switch, scene]'
Error 2024-10-20 02:54:44z2m: Publish 'set' 'mode_2' to '0xa4c13813a475e3d0' failed: 'Error: Value: 'scene_2' not found in: [switch, scene]'

Koenkk, 2024-10-20:
To fix that error, use `e.enum('mode_2', ea.STATE_SET, ['switch', 'scene']).withDescription('Mode for Switch 2'),`

horacecar, 2024-10-20:
The converter below 

> To fix that error, use `e.enum('mode_2', ea.STATE_SET, ['switch', 'scene']).withDescription('Mode for Switch 2'),`

It can only control the switch function of the first and second channels. The SCENE function of the first and second channels cannot be used normally. Not to mention that I asked before that I can use the program code you gave me "https://gist.github.com/Koenkk/f42461ae38525b5f3dd1054a85da043c" to switch ch1. and ch2 scene mode is success, but the backlight function display mode is incorrect

So I would like to ask about the two-way switch and the four-way switch in my hand. Even if the relevant DP ID or product JS information of this product is found on Tuya’s operating platform, it is impossible to program the Zigbee2MQTT converter so that all functions of the product can fully operate normally. ?

The log and custom converter below.  

[_Zigbee2MQTT_logs-4.txt](https://github.com/user-attachments/files/17451699/_Zigbee2MQTT_logs-4.txt)

```
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: "TS0726",
            manufacturerName: "_TZ3002_hlqaknde",
        }
    ],
    model: 'TS0726',
    vendor: 'TuYa',
    description: 'TuYa Two Gang Switch (with scene control and backlight) - 1021',
    powerSource: 'Mains (single phase)',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [
        tz.on_off,
        tuya.tz.datapoints
    ],
    configure: tuya.configureMagicPacket,
    exposes: [
        e.action(['scene_1', 'scene_2']),  // Scene control
        e.linkquality(),
        tuya.exposes.switch().withEndpoint('switch_1'),  // Switch 1
        tuya.exposes.switch().withEndpoint('switch_2'),  // Switch 2
        tuya.exposes.countdown().withEndpoint('1'),  // Countdown 1
        tuya.exposes.countdown().withEndpoint('2'),  // Countdown 2
        e.binary('switch_backlight', ea.STATE_SET, 'ON', 'OFF').withDescription('Backlight Switch'),  // Backlight switch
        e.enum('relay_status', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior'),  // Global power-on behavior
        e.enum('relay_status_1', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior for Switch 1'),  // Power-on behavior for Switch 1
        e.enum('relay_status_2', ea.STATE_SET, ['power_off', 'power_on', 'last']).withDescription('Power-on behavior for Switch 2'),  // Power-on behavior for Switch 2
        e.enum('mode_1', ea.STATE_SET, ['switch', 'scene']).withDescription('Mode for Switch 1'),  // Mode for Switch 1
        e.enum('mode_2', ea.STATE_SET, ['switch', 'scene']).withDescription('Mode for Switch 2'),  // Mode for Switch 2
    ],
    endpoint: (device) => {
        return {
            'switch_1': 1,
            'switch_2': 2
        };
    },
    meta: {
        multiEndpoint: true,
        disableDefaultResponse: true,
        timeout: 5000,
        tuyaDatapoints: [
            [1, 'scene_1', tuya.valueConverter.static('scene')],
            [2, 'scene_2', tuya.valueConverter.static('scene')],
            [17, 'scene_id_group_id', tuya.valueConverter.raw],
            [18, 'mode_1', tuya.valueConverterBasic.lookup({'switch_1': 0, 'scene_1': 1})],
            [19, 'mode_2', tuya.valueConverterBasic.lookup({'switch': tuya.enum(0), 'scene': tuya.enum(1)})],
            [24, 'switch_1', tuya.valueConverter.onOff],
            [25, 'switch_2', tuya.valueConverter.onOff],
            [30, 'countdown_1', tuya.valueConverter.countdown],  // Countdown 1
            [31, 'countdown_2', tuya.valueConverter.countdown],  // Countdown 2
            [36, 'switch_backlight', tuya.valueConverter.onOff],  // Backlight switch
            [38, 'relay_status', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior
            [39, 'relay_status_1', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 1
            [40, 'relay_status_2', tuya.valueConverterBasic.lookup({'power_off': 0, 'power_on': 1, 'last': 2})],  // Power-on behavior for Switch 2
        ],
    },
};

module.exports = definition;
```

Koenkk, 2024-10-21:
I expect that the Tuya gateway does not use the DP id, to find out what the Tuya gateway sends, you need to sniff the traffic when changing the switch type of endpoint 2: https://www.zigbee2mqtt.io/advanced/zigbee/04_sniff_zigbee_traffic.html

closed 2024-12-31 by github-actions[bot]