Koenkk/zigbee2mqtt #24221: Xiaomi ZNCLDJ11LM state is always open

ASNNetworks, 2024-10-05:
### What happened?

After updating Zigbee2mqtt to 1.29.1 this [curtain (motor)](https://www.zigbee2mqtt.io/devices/ZNCLDJ11LM.html) always shows the state as open. When closing the curtain, the position attribute works fine and is set at 0%, however the state of the curtain remains as open.

This was an old bug that now appeared again, same as here: https://github.com/Koenkk/zigbee2mqtt/issues/15926

I noticed that if you wait for a few minutes (like 2-5 min from my testing), the curtain changes itself to the correct state (off). It should be immediately however. I tried changing the report settings, but it had no effect.

### What did you expect to happen?

I expected it to show the correct state when closed.

### How to reproduce it (minimal and precise)

Close the curtain using Z2M dashboard or HA dashboard or automations.

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

20221226

### Adapter

Sonoff Dongle-P

### Setup

Z2M as add-on installed on HA

Drafteed, 2024-10-05:
Let's start over:
1. What fw version is currently installed on the device?
2. Are there any OTA updates for the device?
3. Check in the Z2M devices map whether the device is connected directly to coordinator or through some other device-router?

ASNNetworks, 2024-10-05:
> Let's start over:
> 
> 1. What fw version is currently installed on the device?
> 2. Are there any OTA updates for the device?
> 3. Check in the Z2M devices map whether the device is connected directly to coordinator or through some other device-router?

1. It has the same firmware since I purchased it years ago. It doesn't show a version in frontend, only a date: 04-13-2017. Also, Z2M says there is an update since a while ago (before that it never got an update), but that update never installs. I looked into it, and seems to be some kind of bug, where it actually has no firmware update. So I have never updated the firmware, since it's not available.
2. 2. see above.
3. 3. it's to both the coordinator and multiple others when I check the map: 
[image]
When I only tick is parent, then I see only one line from the device to the coordinator. So it seems it is directly hooked up to that.

Drafteed, 2024-10-06:
> 1. It has the same firmware since I purchased it years ago. It doesn't show a version in frontend, only a date: 04-13-2017. Also, Z2M says there is an update since a while ago (before that it never [...]

Actual firmware build date for ZNCLDJ11LM: 01-25-2022. 
After purchasing, I also had build from ~2017-2019, which was then updated to 01-25-2022 via OTA.

ASNNetworks, 2024-10-06:
> Actual firmware build date for ZNCLDJ11LM: 01-25-2022. After purchasing, I also had build from ~2017-2019, which was then updated to 01-25-2022 via OTA.

Yes, but as I mentioned, it never updates. When its finished, it's the same firmware and still available. Have tried it like 30x in the past year. Most likely there is a small difference in devices (perhaps mine is older, since I own it for like 5 years), but Z2M sees them as the same.

Drafteed, 2024-10-06:
If you are having trouble with OTA, and if you have enthusiasm, you can flash the device via TTL and JN51xxProgrammer.exe. It's not very difficult there, but it requires some soldering skills:
https://1-re--engr-ru.translate.goog/aqara_curtain/?_x_tr_enc=1&_x_tr_sl=ru&_x_tr_tl=en&_x_tr_hl=ru&_x_tr_pto=wapp
Fresh FW (07-17-2020): https://github.com/re-engr/zigbee_firmware/blob/main/aqara/curtain/lumi_curtain_0034_07172020.bin

ASNNetworks, 2024-10-06:
> Actual firmware build date for ZNCLDJ11LM: 01-25-2022. After purchasing, I also had build from ~2017-2019, which was then updated to 01-25-2022 via OTA.

thanks, but I don't want to mess with soldering, just to install an OTA. Also this devices worked fine, and suddenly it has the same bug from 2 years ago, that was fixed with a dev build like in my previous ticket. So I am hoping there is an easy fix from Z2M side. Also since the device worked fine with the same FW.

Drafteed, 2024-10-06:
I think it's a bug in your firmware, because it appeared again after repairing. I don't see any changes from Z2M regarding this device in the last few months.

Drafteed, 2024-10-06:
> thanks, but I don't want to mess with soldering, just to install an OTA, which might not even fix anything

You can also try connecting the device to the original Aqara/Xiaomi hub - maybe it will update it.

closed 2024-12-19 by github-actions[bot]