Koenkk/zigbee2mqtt #24228: [New device support]: THIRDREALITY Zigbee Smart Color Bulb ZL1

MiAutomations, 2024-10-05:
### Link

https://3reality.com/product/smart-color-bulb-zl1/

### Database entry

{"id":184,"type":"Router","ieeeAddr":"0xb40ecfd31e8a0000","nwkAddr":12462,"manufId":5127,"manufName":"Thirdreality","powerSource":"Mains (single phase)","modelId":"3RCB01057Z","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"3RCB01057Z","manufacturerName":"Thirdreality","powerSource":0,"zclVersion":8,"appVersion":32,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"v1.00.32"}},"genOnOff":{"attributes":{"onOff":1,"startUpOnOff":1}},"lightingColorCtrl":{"attributes":{"currentHue":11,"currentSaturation":254,"colorTemperature":1000,"colorCapabilities":25,"colorTempPhysicalMin":142,"colorTempPhysicalMax":454}},"genLevelCtrl":{"attributes":{"currentLevel":254}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":32,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"v1.00.32","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1728150472137}

### Zigbee2MQTT version

1.40.2

### Comments

I tried to add this new device but appears in the lista as not supported yet, can you please include this device in the list of supported devices ? Thank you 

### External definition

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['3RCB01057Z'],
    model: '3RCB01057Z',
    vendor: 'Thirdreality',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[142,454]},"color":{"modes":["xy","hs"]}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

The colors are wrong and the effects doesn't work

Koenkk, 2024-10-06:
Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/third_reality.ts)?

MiAutomations, 2024-10-07:
> Could you make a PR by clicking [here](https://github.com/Koenkk/zigbee-herdsman-converters/edit/master/src/devices/third_reality.ts)?

Yes, but can you please let me know how can i proceed with this kind of request? 

When I try to follow the link I have this message... 

[image]

Thank you

Koenkk, 2024-10-07:
Sorry, you have to make a fork first

JoeVaz81-CLI, 2024-10-27:
Is there any updates on this ticket?

closed 2025-01-03 by github-actions[bot]