Koenkk/zigbee2mqtt #24234: [New device support]: Tuya 6 gang smart switch (NovaDigital with Tuya Whitelabel)

rodrigodrds, 2024-10-05:
### Link

https://www.novadigitalsmart.com.br/produtos/interruptor-inteligente-zigbee-tecla-fisica-4x4-fzb-4-6

### Database entry

{"id":7,"type":"Router","ieeeAddr":"0xa4c1384ce4d3d0ba","nwkAddr":20991,"manufId":4417,"manufName":"_TZE204_g4au0afs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u000b\u0000\u0000\u0000i\u000b\u0000\u0000\u0000i\f\u0000\u0000\u0000i\f\u0000\u0000\u0000i\r\u0000\u0000\u0000i\r\u0000\u0000\u0000i\u000e\u0000\u0000\u0000i\u000e\u0000\u0000\u0000i\u000f\u0000\u0000\u0000i\u000f\u0000\u0000\u0000i","65506":56,"65508":0,"65534":0,"stackVersion":0,"dateCode":"","appVersion":74,"manufacturerName":"_TZE204_g4au0afs","zclVersion":3,"modelId":"TS0601","powerSource":1}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1728166948098}

### Zigbee2MQTT version

1.40.2

### Comments

I tried to import the model TS0601_switch_6_gang (https://github.com/Koenkk/zigbee-herdsman-converters/blob/06c82b69d57358d0a879928deaff05305501366c/src/devices/tuya.ts#L2708C9-L2708C14)
Some basic parts worked, I used the fingerprint "_TZE204_g4au0afs" I don't know if it was correct

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_g4au0afs',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

With the default external nothing works, with this one below the basic one works but still gives some bind errors
js```
{
    fingerprint: tuya.fingerprint('TS0601', ['_TZE204_g4au0afs']),
    model: 'TS0601_switch_6_gang',
    vendor: 'Tuya',
    description: '6 gang switch',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    extend: [tuya.modernExtend.tuyaOnOff()],
    configure: async (device, coordinatorEndpoint) => {
        await tuya.configureMagicPacket(device, coordinatorEndpoint);
        await reporting.bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(3), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(4), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(5), coordinatorEndpoint, ['genOnOff']);
        await reporting.bind(device.getEndpoint(6), coordinatorEndpoint, ['genOnOff']);
        device.powerSource = 'Mains (single phase)';
        device.save();
    },
    exposes: [
        e.switch().withEndpoint('l1'),
        e.switch().withEndpoint('l2'),
        e.switch().withEndpoint('l3'),
        e.switch().withEndpoint('l4'),
        e.switch().withEndpoint('l5'),
        e.switch().withEndpoint('l6'),
    ],
    endpoint: (device) => {
        return {l1: 1, l2: 1, l3: 1, l4: 1, l5: 1, l6: 1};
    },
    whiteLabel: [tuya.whitelabel('Mercator Ikuü', 'SSW06G', '6 Gang switch', ['_TZE200_wnp4d4va'])],
    meta: {
        multiEndpoint: true,
        tuyaDatapoints: [
            [1, 'state_l1', tuya.valueConverter.onOff],
            [2, 'state_l2', tuya.valueConverter.onOff],
            [3, 'state_l3', tuya.valueConverter.onOff],
            [4, 'state_l4', tuya.valueConverter.onOff],
            [5, 'state_l5', tuya.valueConverter.onOff],
            [6, 'state_l6', tuya.valueConverter.onOff],
        ],
    }
}
```

rodrigodrds, 2024-10-05:
Error bind, when refresh

```js
z2m: Failed to configure 'Interruptor Sala', attempt 5 (Error: Bind 0xa4c1384ce4d3d0ba/1 genOnOff from '0x8c65a3fffe5255ec/1' failed (Status 'INVALID_EP') at Endpoint.bind (/app/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:566:23) at processTicksAndRejections (node:internal/process/task_queues:95:5) at runNextTicks (node:internal/process/task_queues:64:3) at processImmediate (node:internal/timers:447:9) at Object.bind (/app/node_modules/zigbee-herdsman-converters/src/lib/reporting.ts:42:9) at Object.configure (/app/data/extension/externally-loaded.js:25:9) at Configure.configure (/app/lib/extension/configure.ts:130:13) at Configure.onMQTTMessage (/app/lib/extension/configure.ts:57:21) at EventEmitter.wrappedCallback (/app/lib/eventBus.ts:206:17))
```

Koenkk, 2024-10-06:
The `configure` part shouldn't be needed for this type of Tuya device.

closed 2024-12-19 by github-actions[bot]