Koenkk/zigbee2mqtt #24235: docker: Error: Cannot find module 'unix-dgram'

cpktmpkt, 2024-10-05:
### What happened?

zigbee2mqtt doesn't start. jugt getting Error: Cannot find module 'unix-dgram'

### What did you expect to happen?

zigbee2mqtt should start

### How to reproduce it (minimal and precise)

starting an zigbee2mqtt docker with the following configuration.yaml

```
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://192.168.10.41:1883
  user: mqtt
  password: '!secret.yaml mqtt_password'
serial:
  port: tcp://192.168.20.11:6638
  baudrate: 115200
  adapter: zstack
  disable_led: false
  transmit_power: 20
frontend:
  port: 8080
homeassistant:
  legacy_entity_attributes: true
advanced:
  log_level: debug
  log_output:
    - console
    - syslog
  log_directory: /var/log/zigbee2mqtt
  log_syslog:
    facility: daemon
    path: /dev/log
    protocol: unix
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    pid: process.pid
    port: 514
    type: '5424'
  network_key: '!secret.yaml network_key'
  pan_id: 36108
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  last_seen: ISO_8601_local
devices: devices.yaml
groups: groups.yaml
device_options:
  legacy: false
availability: true
```

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

20240710

### Adapter

SLZB-06 // CC2652P 

### Setup

Docker container on alpine Linix 3.19 LXC

### Debug log

```
Starting Zigbee2MQTT without watchdog.

[2024-10-06 01:39:27] info: 	z2m: Logging to console, syslog

Error: Cannot find module 'unix-dgram'

Require stack:

- /app/node_modules/winston-syslog/lib/winston-syslog.js

- /app/dist/util/logger.js

- /app/dist/eventBus.js

- /app/dist/controller.js

- /app/index.js

    at Function.Module._resolveFilename (node:internal/modules/cjs/loader:1140:15)

    at Function.Module._load (node:internal/modules/cjs/loader:981:27)

    at Module.require (node:internal/modules/cjs/loader:1231:19)

    at require (node:internal/modules/helpers:177:18)

    at Syslog.connectDgram (/app/node_modules/winston-syslog/lib/winston-syslog.js:309:21)

    at Syslog.connect (/app/node_modules/winston-syslog/lib/winston-syslog.js:352:19)

    at Syslog.log (/app/node_modules/winston-syslog/lib/winston-syslog.js:216:10)

    at Syslog._write (/app/node_modules/winston-transport/modern.js:103:17)

    at writeOrBuffer (/app/node_modules/winston-transport/node_modules/readable-stream/lib/internal/streams/writable.js:334:12)

    at _write (/app/node_modules/winston-transport/node_modules/readable-stream/lib/internal/streams/writable.js:283:10)
```

djurny, 2024-10-06:
Hi there,
Any reason you have `unix` as `protocol`?
Groetjes,

cpktmpkt, 2024-10-06:
to be honest, no. i just followed any tutorial so far and do not have any problems before. But I switched from an bare mateal Alpine Linux installation to the docker image and lust copied my config.

djurny, 2024-10-06:
Hi there,
Would advise to use `udp4` or `tcp4` as it seems there might not be domain socket support for writing to syslog in your Docker env.
Or leave it out from your config to have z2m use default protocol setting for syslog.
Hope that helps,
Groetjes,

cpktmpkt, 2024-10-10:
thank you. A change to udp4 let the container start again without errors

closed 2024-12-19 by github-actions[bot]