Koenkk/zigbee2mqtt #24237: Does not work Loginovo ZG-101ZL

codinv, 2024-10-06:
### What happened?

After update Zigbee2MQTT to 1.40.2 my button Loginovo ZG-101ZL stop working
When I press the button, Z2M returns empty string 
Maybe this is due to the changes in #24013
[image]

### Zigbee2MQTT version

1.40.2 commit: e06848d

### Adapter firmware version

7.4.4 [GA]

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus-E (EFR32MG21)

### Setup

Docker

codinv, 2024-10-06:
That's how it doesn't work
```
triggers:
  - domain: mqtt
    device_id: 11111111111111111111111111
    type: action
    subtype: single
    trigger: device
```

But that's how it works

```
triggers:
  - trigger: state
    entity_id:
      - sensor.button_one_action
    to: single
```

closed 2025-02-20 by github-actions[bot]

carlosnh96, 2025-03-20:
I have the same problem and it has not been solved with the latest versions.