Koenkk/zigbee2mqtt #24238: Bosch - BTH RA - Not resetting target temperature after Window detection off

sebi1412, 2024-10-06:
### What happened?

When resetting Window Detection to "off" after longer periods (~30 Min) of window open, the thermostat will stay at target temperature (Occupied heating setpoint) 8°C, instead of reverting to the previous target e.g. 20°C.

This error does not happen if the window detection was only active for e.g. 30 seconds.

### What did you expect to happen?

Reset to previous target temperature, e.g. 20°C. 
This expected behavior was also present in earlier versions of Z2M.

### How to reproduce it (minimal and precise)

1. Set thermostat to heating mode and ~20°C target temperature
2. Set Window Detection "ON"
3. Wait longer time ~30 Mins
4. Reset Window Detection to "OFF".
5. --> Target temp stays at 8°C

### Zigbee2MQTT version

1.30.2

### Adapter firmware version

7.4.4.0 build 0

### Adapter

Dongle-E - EMBER

### Setup

Addon in Homeassistant on PI4

PatLaba, 2024-10-12:
**Exactly** the same happening with all of my Bosch thermostats. Was wondering until I saw this post...

Zigbee2MQTT version
1.40.2-1 

Adapter firmware version
6.10.3.0 build 297

Adapter
Sonoff EZSP v8

Setup
Addon on HA on proxmox

Debug log
No response

mpyzara, 2024-10-24:
Same problem here. Sometimes turning off window mode reverting previous temperature but sometimes it stays on 8 degrees:

### **Zigbee2MQTT version**
1.40.2

### **Adapter firmware version**
core: v2.3.6 / zigbee: 20240710

### **Adapter**
SMLIGHT SLZB-06 [Ethernet mode]

### **Bosch Thermostat II (BTH-RA) firmware version:**
3.05.09

### **Setup**
Docker container on Ubuntu SRV

wovka88, 2024-11-07:
I noticed the same thing. I would say it was running fine with Deconz before I switched to Z2M.

Is there any approach where I can analyze?

-----

Zigbee2MQTT version
1.41.0-1

Adapter firmware version
7.4.4

Adapter
Home Assistant Connect 

Setup
Addon on HA on proxmox

carsten-blanke, 2024-11-10:
Same here: 
Zigbee2MQTT version
1.41.0 and 2 Versions below

Adapter firmware version
7.4.4

Adapter
Dongle E with Ember driver

Setup
Addon on HA on proxmox

I've done some testing to specify the error.

- I have an automation which pushed the remote temperature every 10 Minutes (xx:00, xx:10, ...) to the thermostat.
- I have testet between the "update windows" (open xx:03 and close xx:05) and over a update window (open xx:09 close xx:11) to see if pushing the remote temperature affects the "occupied_heating_setpoint"
- I try to open and close the window
- I try to switch window detection in z2m dashboard
- I try to switch windows detection in mqtt

I have the following results. Below 1 minute the room temperature restores in nearly 100% to the old set one.
Above 1min it seems like a random decision if the temperature is correctly resetting or not with decreasing probability to get the correct temperature with increasing time with open window.

I try to analyze the logs in z2m. Unfortunately I only see the MQTT published, I don't know how to see the zigbee-logs.
Every open and close leads to 3 published zu MQTT. Here is the sequence:

1. Window detection -> Set to true
2. MQTT Publish
   a. Publish old heating setpoint (21°C) and old window detection state (false)
   b. Publish old heating setpoint (21°C) and new window detection state (true)
   c. Publish new heating setpoint (8°C) and new window detection state (true)
3. Window detection -> Set to false
   a. Publish old heating setpoint (8°C) and old window detection state (true)
   b. Publish old heating setpoint (8°C) and new window detection state (false)
   c. Publish new heating setpoint (21°C) and new window detection state (false)

And in the error case, Number 3c stays with 8°C.

Is there any possibility to analyze zigbee actions? Maybe I can find something there.

carsten-blanke, 2024-11-10:
I found the logs but every time, i activete the debug logging, the thermostats work as I want -.-

carsten-blanke, 2024-11-11:
Found something. I paste Raw Date as logfile, but here only the main finding:
`[2024-11-11 12:51:44] debug: 	z2m: Received Zigbee message from 'OG_Badezimmer_Senstor_Fenster', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":0}' from endpoint 1 with groupID 0
[2024-11-11 12:51:44] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/OG_Badezimmer_Senstor_Fenster', payload '{"battery":100,"battery_low":false,"contact":true,"last_seen":"2024-11-11T12:51:44+01:00","linkquality":128,"tamper":false,"voltage":3100}'`
Window-Senstor says -> Window Closed

In Background an automation says: To Thermostat -> Window open state -> False

`[2024-11-11 12:51:46] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/OG_Badezimmer_Heizung/set/window_detection' with data 'OFF'
[2024-11-11 12:51:46] debug: 	z2m: Publishing 'set' 'window_detection' to 'OG_Badezimmer_Heizung'`

Now i have several Thermostate status updates which leads to MQTT Update messages like
`[2024-11-11 12:51:47] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/OG_Badezimmer_Heizung', payload '{"battery":59,"boost":"OFF","boost_heating":"OFF","child_lock":"UNLOCK","display_brightness":7,"display_ontime":10,"display_orientation":"flipped","displayed_temperature":"target","last_seen":"2024-11-11T12:51:47+01:00","linkquality":124,"local_temperature":20.5,"local_temperature_calibration":0,"occupied_heating_setpoint":8,"operating_mode":"manual","pi_heating_demand":0,"remote_temperature":20.5,"running_state":"idle","setpoint_change_source":"externally","system_mode":"off","update":{"installed_version":889787668,"latest_version":889787668,"state":"idle"},"update_available":null,"valve_adapt_process":false,"valve_adapt_status":"success","window_detection":"ON","window_open":"OFF"}'`

As you can see: occupied_heating_setpoint:8

And then, out of nowhere, 90 seconds later:
`[2024-11-11 12:53:26] debug: 	zh:ember:ezsp: <=== [CBFRAME: ID=63:"MESSAGE_SENT_HANDLER" Seq=136 Len=22]
[2024-11-11 12:53:26] debug: 	zh:ember:ezsp: ezspMessageSentHandler(): callback called with: [status=OK], [type=DIRECT], [indexOrDestination=1173], [apsFrame={"profileId":260,"clusterId":1794,"sourceEndpoint":1,"destinationEndpoint":1,"options":4352,"groupId":0,"sequence":206}], [messageTag=83]
[2024-11-11 12:53:28] debug: 	z2m:mqtt: Received MQTT message on 'zigbee2mqtt/OG_Badezimmer_Heizung/set/occupied_heating_setpoint' with data '8.0'
[2024-11-11 12:53:28] debug: 	z2m: Publishing 'set' 'occupied_heating_setpoint' to 'OG_Badezimmer_Heizung'`

Why in the hell does anyone publish 8.0 degrees to the thermostate?

So i try to find other logs, but only found this one in home assistant progress(?)
[image]

The german "Soll-Temperatur" is the occupied_heating_setpoint

And the 20 degrees you see at 12:54, was manually set by me after this test.

closed 2025-01-19 by github-actions[bot]

Aragur, 2025-02-17:
I found the issue in my automation. My contact sensor did send the `window_detection: 'ON'` multiple times. This overwrote the`thermostatTemperatureSetpoint`. Using `Send only if value changes` in my node-red automation fixes the issue for me. 

[image]