Koenkk/zigbee2mqtt #24241: [Feature request]: Nous B4Z ZigBee Smart Curtain module : missing wall switch mode type selection

Barjea, 2024-10-06:
### Is your feature request related to a problem? Please describe

NOUS B4Z : does not have a way to select wall switch mode such as  two positions pulse or three position up, stop, down.
Without this features these module are useless for lots of household that want to do automation on already existing curtains with most common curtain wall switches.

### Describe the solution you'd like

Expose switch mode so that it can be modified

### Describe alternatives you've considered

set 'by default' three position wall switch type.

### Additional context

const {onOff, windowCovering} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS130F'],
    model: 'TS130F',
    vendor: '_TZ3000_yruungrl',
    description: 'Automatically generated definition',
    extend: [onOff({"powerOnBehavior":false}), windowCovering({"controls":["lift","tilt"]})],
    meta: {},
};

module.exports = definition;

closed 2024-12-19 by github-actions[bot]

mayask, 2025-01-05:
I think I’m having an issue with this module, wall curtain switch stop position stopped working after setting up the device. I don’t use Nous app, only Home Assistant and there doesn’t seem to be a way to change this behavior

mdraaijer, 2025-03-13:
The Nous B4Z works fine with the zigbee2mqqt software.
You could try that if you have not many other modules.
If you have many modules it can indeed be cumbersome to re-arrange all stuff to this new platform.