Koenkk/zigbee2mqtt #24249: Lots of 'BUSY' reports

corporategoth, 2024-10-07:
### What happened?

I setup an automation in Home Assistant, it's fairly simple.

1. I have setup 6 groups in Z2M (each for a cluster of 3 smart bulbs).
2. I setup an automation that, every 2 seconds, will cause ONE of the groups to change the bulb colors (with a 1 second transition).

The effect should be that I slowly see a transition of colors across the 6 groups of bulbs over the course of 10 seconds.
This is setup to repeat every 15 seconds.

I am seeing a LOT of BUSY errors:
```
[2024-10-07 05:17:39] error:    zh:ember: Failed to register group '17' in multicast table with status=INVALID_STATE.
[2024-10-07 05:17:39] error:    zh:ember: Failed to register group '17' in multicast table with status=INVALID_STATE.
[2024-10-07 05:17:40] error:    z2m: Publish 'set' 'brightness' to 'post_6_lights' failed: 'Error: Command 18 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}) failed (~x~> [ZCL GROUP groupId=18] Failed to send with status=BUSY.)'
[2024-10-07 05:17:40] error:    z2m: Publish 'set' 'color' to 'post_6_lights' failed: 'Error: Command 18 lightingColorCtrl.moveToColor({"transtime":10,"colorx":14287,"colory":5112}) failed (~x~> [ZCL GROUP groupId=18] Failed to send with status=BUSY.)'
[2024-10-07 05:17:44] error:    z2m: Publish 'set' 'color' to 'post_1_lights' failed: 'Error: Command 13 lightingColorCtrl.moveToColor({"transtime":10,"colorx":14287,"colory":5112}) failed (~x~> [ZCL GROUP groupId=13] Failed to send with status=BUSY.)'
[2024-10-07 05:17:46] error:    z2m: Publish 'set' 'brightness' to 'post_2_lights' failed: 'Error: Command 14 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}) failed (~x~> [ZCL GROUP groupId=14] Failed to send with status=BUSY.)'
[2024-10-07 05:17:46] error:    z2m: Publish 'set' 'color' to 'post_2_lights' failed: 'Error: Command 14 lightingColorCtrl.moveToColor({"transtime":10,"colorx":40042,"colory":24576}) failed (~x~> [ZCL GROUP groupId=14] Failed to send with status=BUSY.)'
[2024-10-07 05:17:52] error:    z2m: Publish 'set' 'brightness' to 'post_5_lights' failed: 'Error: Command 17 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}) failed (~x~> [ZCL GROUP groupId=17] Failed to send with status=BUSY.)'
[2024-10-07 05:17:52] error:    z2m: Publish 'set' 'color' to 'post_5_lights' failed: 'Error: Command 17 lightingColorCtrl.moveToColor({"transtime":10,"colorx":14287,"colory":5112}) failed (~x~> [ZCL GROUP groupId=17] Failed to send with status=BUSY.)'
[2024-10-07 05:17:54] error:    z2m: Publish 'set' 'brightness' to 'post_6_lights' failed: 'Error: Command 18 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}) failed (~x~> [ZCL GROUP groupId=18] Failed to send with status=BUSY.)'
[2024-10-07 05:17:54] error:    z2m: Publish 'set' 'color' to 'post_6_lights' failed: 'Error: Command 18 lightingColorCtrl.moveToColor({"transtime":10,"colorx":40042,"colory":24576}) failed (~x~> [ZCL GROUP groupId=18] Failed to send with status=BUSY.)'
[2024-10-07 05:18:02] error:    z2m: Publish 'set' 'brightness' to 'post_2_lights' failed: 'Error: Command 14 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}) failed (~x~> [ZCL GROUP groupId=14] Failed to send with status=BUSY.)'
[2024-10-07 05:18:02] error:    z2m: Publish 'set' 'color' to 'post_2_lights' failed: 'Error: Command 14 lightingColorCtrl.moveToColor({"transtime":10,"colorx":14287,"colory":5112}) failed (~x~> [ZCL GROUP groupId=14] Failed to send with status=BUSY.)'
[2024-10-07 05:18:08] error:    z2m: Publish 'set' 'color' to 'post_5_lights' failed: 'Error: Command 17 lightingColorCtrl.moveToColor({"transtime":10,"colorx":40042,"colory":24576}) failed (~x~> [ZCL GROUP groupId=17] Failed to send with status=BUSY.)'
```

Also not sure what the INVALID_STATE thing is all about too - as when I look at group 17, it looks fine.

### What did you expect to happen?

I expect that each group of 3 bulbs will change color (transitioning over 1 second), a different group every 2 seconds.  Without errors.

### How to reproduce it (minimal and precise)

Run the automation.  It's very reproducible in my environment.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.1

### Adapter

ember

### Setup

Running in it's own docker env on TrueNAS.

### Debug log

```
[2024-10-07 05:22:06] debug:    z2m:mqtt: Received MQTT message on 'zigbee2mqtt/post_4_lights/set' with data '{"state":"ON","color":{"x":0.218,"y":0.078},"transition":1.0,"brightness":254}'
[2024-10-07 05:22:06] debug:    z2m: Publishing 'set' 'brightness' to 'post_4_lights'
[2024-10-07 05:22:06] debug:    zh:controller:group: Command 16 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10})
[2024-10-07 05:22:06] debug:    zh:ember: ~~~> [ZCL GROUP apsFrame={"profileId":260,"clusterId":8,"sourceEndpoint":1,"destinationEndpoint":255,"options":4416,"groupId":16,"sequence":0} header={"frameControl":{"reservedBits":0,"frameType":1,"direction":0,"disableDefaultResponse":true,"manufacturerSpecific":false},"transactionSequenceNumber":156,"commandIdentifier":4}]
[2024-10-07 05:22:06] debug:    zh:ember:ezsp: ===> [FRAME: ID=56:"SEND_MULTICAST" Seq=46 Len=26]
[2024-10-07 05:22:06] debug:    zh:ember:uart:ash: ---> [FRAME type=DATA frmTx=6 frmRx=3](ackRx=6)
[2024-10-07 05:22:06] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-10-07 05:22:06] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7](ackRx=6 frmTx=7)
[2024-10-07 05:22:06] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=7 frmNum=3](frmRx=3) Added to rxQueue
[2024-10-07 05:22:06] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=4](ackRx=7)
[2024-10-07 05:22:06] debug:    zh:ember:ezsp: <=== [FRAME: ID=56:"SEND_MULTICAST" Seq=46 Len=7]
[2024-10-07 05:22:06] debug:    zh:ember:ezsp: ~~~> [SENT type=MULTICAST apsSequence=93 messageTag=99 status=BUSY]
[2024-10-07 05:22:06] debug:    zh:controller:group: Error: Command 16 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}) failed (~x~> [ZCL GROUP groupId=16] Failed to send with status=BUSY.)
[... 66 lines cut ...]
[2024-10-07 05:22:07] debug:    z2m: Device 'Post 4 Light - Medium' requested OTA
[2024-10-07 05:22:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-10-07 05:22:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2024-10-07 05:22:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=6](frmRx=6) Added to rxQueue
[2024-10-07 05:22:07] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=7](ackRx=0)
[2024-10-07 05:22:07] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=89:"INCOMING_ROUTE_RECORD_HANDLER" Seq=47 Len=20]
[2024-10-07 05:22:07] debug:    zh:ember:ezsp: ezspIncomingRouteRecordHandler(): callback called with: [source=16982], [sourceEui=0x6c5cb1fffe5e7ca7], [lastHopLqi=192], [lastHopRssi=-52], [relayCount=1], [relayList=24915]
[2024-10-07 05:22:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA]
[2024-10-07 05:22:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0](ackRx=0 frmTx=0)
[2024-10-07 05:22:07] debug:    zh:ember:uart:ash: <--- [FRAME type=DATA ackNum=0 frmNum=7](frmRx=7) Added to rxQueue
[2024-10-07 05:22:07] debug:    zh:ember:uart:ash: ---> [FRAME type=ACK frmRx=0](ackRx=0)
[2024-10-07 05:22:07] debug:    zh:ember:ezsp: <=== [CBFRAME: ID=69:"INCOMING_MESSAGE_HANDLER" Seq=47 Len=33]
[2024-10-07 05:22:07] debug:    zh:ember:ezsp: ezspIncomingMessageHandler(): callback called with: [type=UNICAST], [apsFrame={"profileId":260,"clusterId":2820,"sourceEndpoint":1,"destinationEndpoint":1,"options":64,"groupId":0,"sequence":139}], [packetInfo:{"senderShortId":51526,"senderLongId":"0xFFFFFFFFFFFFFFFF","bindingIndex":255,"addressIndex":255,"lastHopLqi":140,"lastHopRssi":-65,"lastHopTimestamp":0}], [messageContents=18010a0b0529a600]
[2024-10-07 05:22:07] debug:    zh:controller: Received payload: clusterID=2820, address=51526, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=140, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":1,"commandIdentifier":10},"payload":[{"attrId":1291,"dataType":41,"attrData":166}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[... cut ...]
[2024-10-07 05:22:07] debug:    zh:controller: Received payload: clusterID=25, address=11290, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=136, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":65,"commandIdentifier":1},"payload":{"fieldControl":0,"manufacturerCode":4655,"imageType":257,"fileVersion":16908818},"command":{"ID":1,"response":2,"parameters":[{"name":"fieldControl","type":32},{"name":"manufacturerCode","type":33},{"name":"imageType","type":33},{"name":"fileVersion","type":35}],"name":"queryNextImageRequest"}}

```

corporategoth, 2024-10-07:
Updated firmware to 8.0.2, and the errors disappeared.

closed 2024-10-07 by corporategoth

IsaacWG, 2024-10-12:
Thanks @corporategoth! I was having similar issues with Philips Hue light bulbs in groups and the experimental 8.0.2 firmware helped me as well with my Sonoff ZBDongle-E adapter. I still get some `Failed to send with status=BUSY` errors when sending commands to the lights too quickly, but it is at least much more reasonable and it seems to recover faster.

rbnet, 2024-11-25:
@corporategoth @IsaacWG Sorry for the trivial question, but where can I download firmware 8.0.2 for the SonOFF Dongle-E from? I have the same problem with groups of Philips Hue lamps.

corporategoth, 2024-11-25:
@rbnet https://github.com/Nerivec/silabs-firmware-builder

rbnet, 2024-11-26:
The update to firmware 8.0.2 completely fixed the problem.
Thank you.

Ryuisnod, 2024-12-05:
> The update to firmware 8.0.2 completely fixed the problem. Thank you.

would you be able to share this firmware? the build looks rather complicated. i also have the E dongle and have flashed before with secureCRT

rbnet, 2024-12-05:
@Ryuisnod You can find the firmware build at the link provided by corporategoth. Just click on "Release" and look through the assets.

Ryuisnod, 2024-12-06:
> @Ryuisnod You can find the firmware build at the link provided by corporategoth. Just click on "Release" and look through the assets.

thanks!

vhkristof, 2024-12-07:
I got similar issues on a SLZB-06M (connected via network)

remisharrock, 2024-12-16:
have the same issue with SLZB-06M (over ethernet) too

remisharrock, 2024-12-19:
and I tried with https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update3

hsmanninen, 2024-12-20:
Same problem with SLZB-06M over Ethernet

Zigbee firmware version: 20241127

quodo, 2025-01-20:
> Updated firmware to 8.0.2, and the errors disappeared.

**Thanks**! This comment was the answer to my problem with having a lot of `Failed to send with status=BUSY errors`. 

**System**

- Home Assistant v2025.1.2
- Zigbee2MQTT v2.0.0-2, ember, "Home Assistant leacy action sensors" unchecked
- Sonoff Zigbee 3.0 USB Dongle E
- Firmware Zigbee dongle: `ncp-uart-hw-v7.4.5.0-zbdonglee-115200.gbl` from https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee.

**Example error**
`z2m: Publish 'set' 'state' to 'Lampje TV dressoir' failed: 'Error: Command 2 genOnOff.on({}) failed (~x~> [ZCL GROUP groupId=2] Failed to send with status=BUSY.)'`

**Updating the firmware** of my Sonoff Zigbee 3.0 USB Dongle E to 
`sonoff_zbdonglee_zigbee_ncp_8.0.2.0_115200_sw_flow.gbl` found here https://github.com/Nerivec/silabs-firmware-builder/releases using the Firmware Flasher found here https://darkxst.github.io/silabs-firmware-builder/ solves my problem with `Failed to send with status=BUSY errors`.

Nerivec, 2025-01-20:
Just a quick note. BUSY in this scenario usually means you are relying quite heavily on groups. Newer firmware tend to solve the immediate issue because they have larger buffering capabilities, but it still means you are sending a lot of group messages over the network, which can lead to slow downs and other quirks.
You will want to check a few things:
- Make sure automations are optimized, and not called unnecessarily, too fast, or too many times
- Avoid using devices like dimmers in groups (these are usually the no.1 cause of BUSY networks because they rarely implement proper throttling of messages and just send every tiny increment as a message over the network, resulting in potentially hundreds of messages in just a few milliseconds)

pitkaha, 2025-01-20:
> Just a quick note. BUSY in this scenario usually means you are relying quite heavily on groups. Newer firmware tend to solve the immediate issue because they have larger buffering capabilities, but it still means you are sending a lot of group messages over the network, which can lead to slow downs and other quirks. You will want to check a few things:
> 
> * Make sure related automations are optimized, and not called unnecessarily, too fast, or too many times
> * Avoid using devices like dimmers in groups (these are usually the no.1 cause of BUSY networks because they rarely implement proper throttling of messages and just send every tiny increment as a message over the network, resulting in potentially hundreds of messages in just a few milliseconds)

Figured as much, my problems were worse when controlling big groups of lights (50-80 bulbs), no dimming, just brightness and temperature with 1 sec transition. One change to all those three lead to about 5-10 sec before the next command could be successfully issued, but no problems any more, thanks to your firmware builder.

Do you happen to know does the 1 sec transition load the system? How about the transition in device settings on zigbee2mqtt?

quodo, 2025-01-20:
@Nerivec, can I ask when one is relying too heavily on groups?

I only changed the Zigbee adapter protocol from ezsp protocol (firmware 6.10.3.0 build 297 on the dongle when I bought it) to ember this week. With ezsp things worked fine. I had/have five groups and two buttons. 

Button one toggles two groups on/off with two light bulbs each, so four Zigbee devices. This did not go well with ember v7.

Button two toggles five groups on/off with one to three light bulbs each, with a total of nine Zigbee devices. This did not go well with ember v7 either.

All these Zigbee devices are Ikea light bulbs. There are no dimmers. The buttons I tried were Ikea Styrbar (via a blueprint) and Sonoff SNZB-01 (directly from an automation). 

Is that "relying quite heavily on groups"? 
Is there a better way?

Nerivec, 2025-01-20:
Per ZigBee spec, a broadcast is expected to propagate through a network in ~9 seconds. This of course changes based on the size of the network, and how deeply devices are nested. Also per spec, the limit for "over-the-air" broadcasts is around 8 per 9-second window. Over that, stacks are forced to buffer them, or drop them (if no buffering, or buffer full).
These are meant to avoid basically ruining the network with rapid-succession broadcasts.

@quodo the switch between `ezsp` and `ember` can, in some cases, bring about a (partial) network rebuild (does it itself, but can be disrupting for a while). Was that issue you experienced observed even after several hours (to let the network settle down)? We're not quite sure what these old firmware had for configuration (closed source builds), so, hard to say exactly how that would affect the change. On top of that, the mere stability of the network matters too (nesting, interference, etc.). _If you have some `debug` logs from those issues, I can take a look if you want._

pitkaha, 2025-01-21:
> Per ZigBee spec, a broadcast is expected to propagate through a network in ~9 seconds. This of course changes based on the size of the network, and how deeply devices are nested. Also per spec, the [...]

Great, seems like the dropping was happening with 7.x fw, thank you for you help!

quodo, 2025-01-22:
> Was that issue you experienced observed even after several hours (to let the network settle down)?

@Nerivec, I searched in my notes for the timeline. The issue with BUSY errors was even there after a few days:
- Jan. 15 about 21.00: Updated firmware from `ezsp` v6 to `ember` v7
- Jan. 19: Automation fails with busy error on Jan. 19 [Github comment with screenshot of errors](https://github.com/EPMatt/awesome-ha-blueprints/discussions/623#discussioncomment-11881566). There are several groups with lights.
[One button with all lights in a single group](https://github.com/EPMatt/awesome-ha-blueprints/discussions/623#discussioncomment-11882116) seemed to work though.
- Jan. 20: Still BUSY errors, so updated firmware from ember v7 to v8, no more BUSY errors, with the exact same automation.

If it would help I'd be happy to try to find logs, however, I need some help where to find what. I suppose the Debug log given in the first post in this thread will be better than what I could give you since I don't have debug enabled? Its error is the same as mine.

Nerivec, 2025-01-22:
Previous firmware had a buffer of 15, which should mean 15+8 broadcasts would have to be "in-transit" to produce the BUSY signal, which, based on your automations, sounds a little unlikely. However, there are other sources for broadcasts (outside of your control, like routers acting as concentrator), and I suspect a lot of people that had this error may have had a combination.

jeffreyvangorkum, 2025-12-09:
So I run into this issue with the Sonoff Dongle-M which currently sits on firmware 7.4.5. It just feels weird to have this issue while pushing a group to turn on or to have Home Assistant poke 4 bulbs to go on.

Unfortunately no firmware 8.x exists for the Dongle-M :(

jeffreyvangorkum, 2025-12-18:
Fair enough. And works perfectly without the groups but it does feel a bit counter intuitive :)

I did a scan on my Dongle-M and it's not doing any secret WiFI broadcasting. However, I just connected the wifi to my network and because the UTP cable is connected it disabled WiFi. Maybe that's the trick.

jonzuchin, 2026-04-26:
@Nerivec Hello! I'm new on this business, I have a Sonoff Dongle-M (m24) as a coordinator MQTT<>Z2M, to flash with 9.0.1 firmware, which version should I use? NPC, OpenThread or router?
Thanks in advance!