Koenkk/zigbee2mqtt #24250: Tuya ZP01 TS0202 - Occupancy not changing

jesade-vbg, 2024-10-07:
### What happened?

The PIR sensor is added nicely, but Occupancy does not change, ever. I can see the sensor blink but the state does not update.

generated_external_definition 
```
const {battery, iasZoneAlarm, commandsOnOff} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['TS0202'],
    model: 'TS0202',
    vendor: '_TZ3000_lf56vpxj',
    description: 'Automatically generated definition',
    extend: [battery(), iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), commandsOnOff()],
    meta: {},
};

module.exports = definition;
```

### What did you expect to happen?

I expect that the Occupancy state changes whrn the PIR sensor triggers.

### How to reproduce it (minimal and precise)

Just try to trigger a response from the PIR sensor

### Zigbee2MQTT version

1.40.2 

### Adapter firmware version

EmberZNet 7.4.3 [GA]

### Adapter

Sonoff ZBDongle-E

### Setup

HA on HA Green

wuppiwuppi, 2024-10-25:
I can confirm this. Occupancy never changes from N/A. 
Device however blinks on movement detection.

psycoticninja, 2024-10-29:
Same here - thought it was a DOA item.

wuppiwuppi, 2024-10-29:
Dead Or Alive?

psycoticninja, 2024-10-29:
> Dead Or Alive?

Dead on Arrival.

wuppiwuppi, 2024-10-29:
ah. I believe not. 
As I stated: it blinks on movement detection. So it seems to be alive. 
I lack other ZigBee hubs to test it, but I believe it is a communication issue. Also: some other ppl on Amazon seem to get it working, I guess in different environments.

closed 2025-01-06 by github-actions[bot]

QazCetelic, 2025-08-15:
Were you able to get it working?