Koenkk/zigbee2mqtt #24252: Lixee: missing discovery topic for register_breakout

KipK, 2024-10-07:
### What happened?

I can see the register breakout data published in /zigbee2mqtt/linky topic:
But Discovery topic do not expose those breakout data.
Could it be possible to add those to the discovery? 
For example the "depassement_ref_pow" is a really important data to have, it's the signal your overpowering.

{
	"active_energy_out_d01": 37493.8,
	"active_energy_out_d02": 1166.6,
(...)
	"status_register": "003A4801",
	"status_register_breakout": {
		"cache_borne_dist": "ferme",
		"comm_euridis": "activee avec securite",
		"contact_sec": "ouvert",
		"depassement_ref_pow": 0,
		"etat_cpl": "nouveau_verrouille",
		"horloge": "correcte",
		"organe_coupure": "ferme",
		"pointe_mobile": "AUCUN",
		"preavis_pointe_mobile": "AUCUN",
		"producteur": 0,
		"sens_energie_active": "positive",
		"surtension_phase": 0,
		"sync_cpl": "non_synchronise",
		"tarif_dist": "index_2",
		"tarif_four": "index_3",
		"tempo_demain": "UNDEF",
		"tempo_jour": "UNDEF",
		"type_tic": "standard"
	},
(...)
}

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

ember 8.0.1

### Adapter

Ember Sonff-E

### Setup

z2m in lxc, HASS in vm

ralmn, 2024-10-07:
I don't know if we can expose data who are on object.

@Koenkk can we expose some attribute from attribute object key ? 
```
e.binary('DEP_REF_POW', ea.STATE, true, false).withProperty('statusRegister_breakout.depassement_ref_pow')
 ```
 I just tried it on my Z2M instance with an external convert but the stay on `N/A`...

KipK, 2024-10-07:
@ralmn  I've also tried to manually create a discovery topic, but failed same as you. Thought I've missed something on my side.

KipK, 2024-10-08:
Interesting, for whatever reason home assistant didn't parse nested json with dot path. 
However, when using brackets it then works ok with discovery topic.

this works:

```
{
  "availability": [
    {
      "topic": "zigbee2mqtt/bridge/state",
      "value_template": "{{ value_json.state }}"
    },
    {
      "topic": "zigbee2mqtt/Linky/availability",
      "value_template": "{{ value_json.state }}"
    }
  ],
  "availability_mode": "all",
  "device": {
    "identifiers": [
      "zigbee2mqtt_0x00158d000638e0dc"
    ],
    "manufacturer": "LiXee",
    "model": "Lixee ZLinky (ZLinky_TIC)",
    "name": "Linky",
    "sw_version": "4000-0014",
    "via_device": "zigbee2mqtt_bridge_0xe0798dfffea81c6e"
  },
  "enabled_by_default": true,
  "name": "DEPASSEMENT",
  "object_id": "linky_status_register_breakout_depassement_ref_pow",
  "origin": {
    "name": "Zigbee2MQTT",
    "sw": "1.40.2",
    "url": "https://www.zigbee2mqtt.io"
  },
  "state_topic": "zigbee2mqtt/Linky",
  "unique_id": "0x00158d000638e0dc_status_register_breakout_depassement_zigbee2mqtt",
  "value_template": "{{ value_json['status_register_breakout']['depassement_ref_pow'] }}"
}

```

closed 2024-12-19 by github-actions[bot]