Koenkk/zigbee2mqtt #24254: [New device support]: Neo Smart Water Valve NAS-WV05B2 Tuya _TZE204_z7a2jmyy

TechsDomadoo, 2024-10-07:
### Link

https://www.domadoo.fr/en/water-consumption/7084-neo-zigbee-30-tuya-smart-watering-controller.html

### Database entry

{"id":84,"type":"EndDevice","ieeeAddr":"0xa4c138233850d042","nwkAddr":5571,"manufId":4417,"manufName":"_TZE204_z7a2jmyy","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65487":14400,"65506":56,"65508":1,"modelId":"TS0601","manufacturerName":"_TZE204_z7a2jmyy","powerSource":3,"zclVersion":3,"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1728308733645}

### Zigbee2MQTT version

1.40.2-1

### Comments

The device was supported before. It seems it is a different firmware version as it is recognized as not supported now.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_z7a2jmyy',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

I didn't try it

TechsDomadoo, 2024-12-11:
It seems there is a solution here : https://github.com/Koenkk/zigbee2mqtt/issues/22813#issuecomment-2382832814

Koenkk, 2024-12-11:
Added in https://github.com/Koenkk/zigbee2mqtt/issues/22813

closed 2024-12-11 by Koenkk

Quenbo, 2025-06-28:
(first post on Github so please forgive me if I am wrong)

Hi Koenkk,

My device is incorrectly recognized in Zigbee2MQTT. Here are the details:

- **Expected model**: NAS-WV05B2 (smart sprinkler timer)
- **ModelID reported**: TS0601
- **ManufacturerName reported**: _TZE204_uab532m0
- **Current detected as**: NAS-WV03B
- **Link to official supported device**: https://www.zigbee2mqtt.io/devices/NAS-WV05B2.html

Could you please add this fingerprint to the NAS-WV05B2 definition?

**Fingerprint:**

```json
{
  "modelID": "TS0601",
  "manufacturerName": "_TZE204_uab532m0"
}

Thanks

Koenkk, 2025-06-29:
@Quenbo `_TZE204_uab532m0` has already been added to the dev branch.

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Quenbo, 2025-06-29:
Thanks Koen !
Fortunately lext 1st of month is soon...
Have a nice WE.

Quenbo, 2025-07-18:
> [@Quenbo](https://github.com/Quenbo) `_TZE204_uab532m0` has already been added to the dev branch.
> 
> > Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

Hi Koen, I tried to reconnect my device after updating Z2M to 2.5.1, but it is still recognized as a NAS-WV03B instead of a NAS-WV05B2.
Could you please confirm if the fingerprint for _TZE204_uab532m0 has been added, or is planned for the next release?
Thanks for the support.
```
{
  "modelID": "TS0601",
  "manufacturerName": "_TZE204_uab532m0"
}
```
Did I do something wrong

Quenbo, 2025-07-19:
> [@Quenbo](https://github.com/Quenbo) is it just a cosmetic issue? (do all features work)?

Unfortunately not.
The majority of features does not work at all.

There are a lot of entities that are N/A. (battery, status, fault status, quaitative watering, single watering amount, duration, etc). Water total is stuck at 0 L, and the water current flow is shown in L/min but the numbers shown are in fact gallon/min

Only the switch ON/OFF works... 

I read somewhere that the good version "NAS-WV05B2" can correct some of these problems, and among other things battery status and water metering in L/min.

I struggle to export the Z2M Expose datas that are in french, but if you need it I can try again

Koenkk, 2025-07-19:
Could you try to:

- Stop Z2M
- Open `data/database.db` with a text editor and change all occurrences of `_TZE204_uab532m0` to `_TZE284_nnhwcvbk`
- Start Z2M and see if the device now works
- If it does not work try with `_TZE284_z7a2jmyy` instead of `_TZE284_nnhwcvbk`

Quenbo, 2025-07-19:
Thanks.
So, it worked better with the second replacement proposed : _TZE284_z7a2jmyy.
The picture shown is the right one, and I have now access to:
- Status : disabled (instead of N/A)
- Water current is shown in x,xxx L/min (instaed of x xxx L/min)
- Battery is now showed as nul % (instead of N/A) (maybe I will have to wait for it to refresh)

```
{
    "countdown": 1,
    "countdown_left": 1,
    "flow_switch": "ON",
    "last_seen": "2025-07-19T20:17:12+02:00",
    "linkquality": 172,
    "quantitative_watering": 0,
    "state": "ON",
    "switch_enabled": "OFF",
    "total_flow_reset_switch": "OFF",
    "water_current": 2.122,
    "water_total": 0,
    "battery": null,
    "child_lock": "ON",
    "current_switch": "ON",
    "reset_switch": "ON",
    "status": "app_manual"
}
```
Water total is still stuck at zero.

Thanks again.

Koenkk, 2025-07-20:
Can you capture the debug logs of 24 hours, use some water and see if you see any log statement like below in the log?

```
Datapoint XX not defined for 'XX' with value XX
```

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

If not, we would need to reverse engineer the correct datapoints via the Tuya hub: https://www.zigbee2mqtt.io/advanced/support-new-devices/03_find_tuya_data_points.html