Koenkk/zigbee2mqtt #24256: Issues with disable/enable devices via Web UI

Great-Chart, 2024-10-07:
### What happened?

Via Device / Settings tab ticking the Disabled checkbox results in it immediately un-ticking itself before it is possible to submit.
I have found a workaround in that (on desktop PC) checking and immediately pressing enter key applies the revision; but not sure of a mobile app capability

### What did you expect to happen?

Select a tick in "Disabled" checkbox and have it stick long enough to submit

### How to reproduce it (minimal and precise)

The current problem is depicted in screengrab / video form within https://github.com/nurikk/zigbee2mqtt-frontend/issues/2135.
It has also been referenced within the last week or so under https://github.com/Koenkk/zigbee2mqtt/discussions/16314.
I have commented under both to indicate the workaround I have been using for info to those parties; as its been a while since I experienced the issue myself and somehow stumbled over the above workaround.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20230507

### Adapter

Sonoff Dongle Plus P

### Setup

x86 mini PC

CasperVerswijvelt, 2024-10-27:
Having the same issue, very annoying

Great-Chart, 2024-10-28:
@CasperVerswijvelt 
It's been acknowledged as an issue https://github.com/Koenkk/zigbee2mqtt/discussions/16314 so I expect it'll get resolved in due course - workaround has been OK for me thus far

closed 2025-01-05 by github-actions[bot]