Koenkk/zigbee2mqtt #24261: (BLE-YL01) no update data after verison Z2m 1.40.2

aperezva, 2024-10-08:
### What happened?

after update my z2m to 1.40.2  my (BLE-YL01) pool monitor doesn´t send information. Ihave checked with a smartlife hub and it´s working.

### What did you expect to happen?

data updates after 10 muinutes.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

 20221226

### Adapter

slzb 06

### Setup

lxc proxmox

aperezva, 2024-10-09:
Aby feedback or information @Koenkk ?

Koenkk, 2024-10-09:
Does it work again after downgrading z2m?

aperezva, 2024-10-09:
Yes
 Correct. 
 
And I have two z2m databases in both is failing

aperezva, 2024-10-09:
This command works before. Now it´s not working in both Z2m 

[image]

Vaarinvaliliha, 2024-10-10:
Same thing here, interview goes fine but no data is moving :)

aperezva, 2024-10-10:
> Does it work again after downgrading z2m?

But I don´t want to downgrade it for only one devices.

Any alternatives?

A new version?

Koenkk, 2024-10-10:
Found the issue and fixed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-10-10 by Koenkk

aperezva, 2024-10-10:
Thanks

Will you upload to production, later this month?

BR

Koenkk, 2024-10-10:
Yes, 1 November

theimo1221, 2024-10-13:
@Koenkk It still might be happening, as even with herdsman V20.25.0 I don't get any values after pairing TS0202 and device sometimes get's stuck in a weird loop in which device leaves network directly after configure during interview.

Koenkk, 2024-10-13:
@theimo1221 please open a new issue for that, it's not related to this device.

theimo1221, 2024-10-13:
Thought it might be connected as it is a Tuya as well (considering your commit message)
Follow up issue:
https://github.com/Koenkk/zigbee2mqtt/issues/24314

andresgomezfrr, 2024-12-31:
hey @aperezva @Koenkk, have you checked if the issue is solved after the latest release? I'm trying to pair the BLE-YL01 device and I cant send the command, this is the error:

```
zigbee2mqtt  | [2024-12-31 12:01:50] error: 	z2m: Publish 'set' 'command' to 'Piscina_Sensor' failed: 'Error: No datapoint defined for 'command''
```

I also see this error on the logs related to the same device:

```
zigbee2mqtt  | [2024-12-31 11:47:41] error: 	z2m: EventBus error 'OnEvent/deviceMessage': ZCL command 0x0cae5ffffee8c740/1 manuSpecificTuya.mcuVersionRequest({"seq":2}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'No network route' (205))
```

Any help? please! 🙏🏼  and happy new year!! 🎉

aperezva, 2024-12-31:
Hi,

It was solved versions before o n november versión i guess.

Remove from network and pair it again.

ThePhantom79, 2025-02-15:
Hi, 
I a, slo using BLE-YL01 with Z2m. Pairing was successfull and I get one time the correct data. Sometimes after a while it will be renewed once or twice, but then : nothing. I dont get any new data from device. 

Z2M version:
2.1.1
Coordinator-Typ
zStack3x0
Coordinator-Version
20240710
Coordinator IEEE Adresse
0x00124b0031df5380
Frontend-Version
0.9.4
Zigbee Herdsman Konverter Version
21.31.0
Zigbee Herdsman Version
3.2.5

I dont know, what to do. With original Tuya Gateway everything worked well for the last weeks.

Any hints?

juslex, 2025-05-16:
Some problem here

Sylv157, 2025-05-31:
Same problem here. Can you solve this please?

juslex, 2025-06-08:
please check here https://github.com/Koenkk/zigbee2mqtt/issues/23946#issuecomment-2949219614 There is a new external converter that seems to have solved the problem

Koenkk, 2025-06-08:
The fix is available in the dev branch, can you check if it works?

> Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now and in the next release which is every 1st of the month.

ThePhantom79, 2025-06-08:
For me unfortunately it doesnt  work ...
Repairing, Reconfigering,  Interviewing successfull, but then no data receiving ... :-(

OE5DRO, 2025-06-09:
i dont use dev branche but external converter, but it does not work:

[image]

xvolte, 2025-08-07:
any news on this ? i have same problem