Koenkk/zigbee2mqtt #24264: SIN-4-FP-21 power value intermittently drops to 0

jojeju9428, 2024-10-08:
### What happened?

Module SIN-4-FP-21 is not reporting power and energy anymore.
It's stuck to 0.
Tried to reconfigure.
Tried to reset module
Tried to delete from z2m and add again

### What did you expect to happen?

It was previously working as expected in version 1.39.1 or 1.40.1

### How to reproduce it (minimal and precise)

After switching to comfort mode, the power value intermittently drops to 0.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

20230507

### Adapter

SONOFF Zigbee 3.0 USB Dongle Plus

### Setup

Raspberry Pi

### Debug log

`[2024-10-07 12:29:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:27.282Z","linkquality":76,"pilot_wire_mode":null,"power":0,"state":"OFF","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:27.287Z","linkquality":76,"pilot_wire_mode":null,"power":0,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:27.287Z","linkquality":76,"pilot_wire_mode":"comfort","power":0,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:27.304Z","linkquality":76,"pilot_wire_mode":"comfort","power":0,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:28.404Z","linkquality":76,"pilot_wire_mode":"comfort","power":983,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:32.425Z","linkquality":76,"pilot_wire_mode":"comfort","power":983,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:33.382Z","linkquality":76,"pilot_wire_mode":"comfort","power":982,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:37] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:37.408Z","linkquality":76,"pilot_wire_mode":"comfort","power":982,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:38] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:38.365Z","linkquality":76,"pilot_wire_mode":"comfort","power":985,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:42] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:42.388Z","linkquality":72,"pilot_wire_mode":"comfort","power":985,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:43] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:43.344Z","linkquality":76,"pilot_wire_mode":"comfort","power":971,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:46] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:46.943Z","linkquality":76,"pilot_wire_mode":"comfort","power":971,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:47] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:47.369Z","linkquality":76,"pilot_wire_mode":"comfort","power":971,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:48] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:48.330Z","linkquality":76,"pilot_wire_mode":"comfort","power":993,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:51] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:51.237Z","linkquality":76,"pilot_wire_mode":"comfort","power":993,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:51] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0,"last_seen":"2024-10-07T10:29:51.237Z","linkquality":76,"pilot_wire_mode":"comfort","power":993,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:52] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.01,"last_seen":"2024-10-07T10:29:52.356Z","linkquality":72,"pilot_wire_mode":"comfort","power":993,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:53] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.01,"last_seen":"2024-10-07T10:29:53.313Z","linkquality":76,"pilot_wire_mode":"comfort","power":978,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:56] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.01,"last_seen":"2024-10-07T10:29:56.657Z","linkquality":76,"pilot_wire_mode":"comfort","power":978,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:57] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.01,"last_seen":"2024-10-07T10:29:57.332Z","linkquality":76,"pilot_wire_mode":"comfort","power":978,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:58] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.01,"last_seen":"2024-10-07T10:29:58.288Z","linkquality":76,"pilot_wire_mode":"comfort","power":972,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:29:59] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.01,"last_seen":"2024-10-07T10:29:59.645Z","linkquality":76,"pilot_wire_mode":"comfort","power":972,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:30:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.01,"last_seen":"2024-10-07T10:30:02.315Z","linkquality":76,"pilot_wire_mode":"comfort","power":972,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:30:02] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.01,"last_seen":"2024-10-07T10:30:02.636Z","linkquality":76,"pilot_wire_mode":"comfort","power":972,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[... cut ...]
[2024-10-07 12:31:28] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.03,"last_seen":"2024-10-07T10:31:28.602Z","linkquality":156,"pilot_wire_mode":"comfort","power":913,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
[2024-10-07 12:31:29] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt-2/Chauffage salle de bain 1', payload '{"energy":0.03,"last_seen":"2024-10-07T10:31:29.881Z","linkquality":156,"pilot_wire_mode":"comfort","power":913,"state":"ON","update":{"installed_version":771,"latest_version":771,"state":"idle"},"update_available":null}'
`

ParusCristatus, 2024-10-12:
Hey @jojeju9428,

I am noticing the same behaviour on my `SIN-4-FP-21` modules. In my case, I've come to think this is not a reporting issue, but that the heater actually cuts off its power to 0 every now and then to regulate the temperature. I've set up some graphs which show this behaviour quite clearly. Every morning when the heater starts cold, power stays at maximum level for a while. Then, when it starts to get hot, the power get cuts off to 0 for a few seconds, more and more frequently until the desired temperature is reached.
[image]
My heaters are Intuis Chamane, but I'm guessing that many modern radiators work like this.

AlexisPolegato, 2024-11-19:
> ### What happened?
> Module SIN-4-FP-21 is not reporting power and energy anymore. It's stuck to 0. Tried to reconfigure. Tried to reset module Tried to delete from z2m and add again

@jojeju9428 The issue has been fixed on the V3.5.0 of the Firmware's Device ; you can try to update it and then reconfigure it

closed 2025-01-27 by github-actions[bot]