Koenkk/zigbee2mqtt #24266: WXCJKG12LM, WXCJKG11LM, WXCJKG13LM - all action entities are now showing as Unknown, although all functionality is still working

dnedeltchev, 2024-10-08:
### What happened?

Upgraded to 1.40.2-1 and now core to 2024.10.1 and all my *action entities for these Aqara switches have changed their default status to Unknown which is triggering my automation to alert on Unknown, offline entities/devices. I can exclude them from the automation but I wonder what's changed? I can't find any changes but again, this happened after the Core update.

### What did you expect to happen?

When idle, the action state should be None

### How to reproduce it (minimal and precise)

Core 2024.10.1 and z2m 1.40.2-1

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

20230507

### Adapter

zzh!

### Setup

Home Assistant Blue (ODROID N2+)

### Debug log

[image]
nothing in the log but this is a screenshot from before the Core upgrade and after.

Chartreusito, 2024-10-09:
Same issue as https://github.com/Koenkk/zigbee2mqtt/issues/24195

Koenkk, 2024-10-09:
Could you provide the debug log when the action goes to unknown?

See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

dnedeltchev, 2024-10-09:
> Could you provide the debug log when the action goes to unknown?
> 
> See [this](https://www.zigbee2mqtt.io/guide/configuration/logging.html) on how to enable debug logging.

Is this sufficient? The device I am testing with is 0x54ef441000179f97

[2024-10-09 18:08:59] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,3,203,1,223,236,126]
[2024-10-09 18:08:59] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [3,203,1,223,236] - 126
[2024-10-09 18:08:59] debug: 	zh:zstack:znp: <-- AREQ: ZDO - srcRtgInd - {"dstaddr":51971,"relaycount":1,"relaylist":[60639]}
[2024-10-09 18:08:59] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-09 18:08:59] debug: 	zh:zstack:unpi:parser: <-- [254,25,68,129,0,0,6,0,3,203,1,1,0,83,0,93,215,216,0,0,5,24,16,11,1,0,223,236,28,59]
[2024-10-09 18:08:59] debug: 	zh:zstack:unpi:parser: --- parseNext [254,25,68,129,0,0,6,0,3,203,1,1,0,83,0,93,215,216,0,0,5,24,16,11,1,0,223,236,28,59]
[2024-10-09 18:08:59] debug: 	zh:zstack:unpi:parser: --> parsed 25 - 2 - 4 - 129 - [0,0,6,0,3,203,1,1,0,83,0,93,215,216,0,0,5,24,16,11,1,0,223,236,28] - 59
[2024-10-09 18:08:59] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":51971,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":83,"securityuse":0,"timestamp":14210909,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[24,16,11,1,0]}}
[2024-10-09 18:08:59] debug: 	zh:controller: Received payload: clusterID=6, address=51971, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=83, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":16,"commandIdentifier":11},"payload":{"cmdId":1,"statusCode":0},"command":{"ID":11,"name":"defaultRsp","parameters":[{"name":"cmdId","type":32},{"name":"statusCode","type":32}]}}
[2024-10-09 18:08:59] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-09 18:08:59] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d0002a6082c', payload '{"brightness":254,"color_mode":"color_temp","color_temp":370,"linkquality":83,"power_on_behavior":null,"state":"ON"}'
[2024-10-09 18:09:13] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,18,0,253,5,1,1,0,120,0,235,115,229,0,0,8,24,167,10,85,0,33,1,0,39,33,28]
[2024-10-09 18:09:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,18,0,253,5,1,1,0,120,0,235,115,229,0,0,8,24,167,10,85,0,33,1,0,39,33,28]
[2024-10-09 18:09:13] debug: 	zh:zstack:unpi:parser: <-- [228]
[2024-10-09 18:09:13] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,18,0,253,5,1,1,0,120,0,235,115,229,0,0,8,24,167,10,85,0,33,1,0,39,33,28,228]
[2024-10-09 18:09:13] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,253,5,1,1,0,120,0,235,115,229,0,0,8,24,167,10,85,0,33,1,0,39,33,28] - 228
[2024-10-09 18:09:13] debug: 	zh:zstack:znp: <-- AREQ: AF - incomingMsg - {"groupid":0,"clusterid":18,"srcaddr":1533,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":120,"securityuse":0,"timestamp":15037419,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[24,167,10,85,0,33,1,0]}}
[2024-10-09 18:09:13] debug: 	zh:controller: Received payload: clusterID=18, address=1533, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=120, frame={"header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":167,"commandIdentifier":10},"payload":[{"attrId":85,"dataType":33,"attrData":1}],"command":{"ID":10,"name":"report","parameters":[{"name":"attrId","type":33},{"name":"dataType","type":32},{"name":"attrData","type":1000}]}}
[2024-10-09 18:09:13] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-09 18:09:13] debug: 	z2m: Received Zigbee message from '0x54ef441000179f97', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 1 with groupID 0
[2024-10-09 18:09:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x54ef441000179f97', payload '{"action":"button_1_single","battery":94,"linkquality":120,"operation_mode":"command","power_outage_count":21,"voltage":2991}'
[2024-10-09 18:09:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x54ef441000179f97', payload '{"action":"","battery":94,"linkquality":120,"operation_mode":"command","power_outage_count":21,"voltage":2991}'
[2024-10-09 18:09:13] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x54ef441000179f97/action', payload 'button_1_single'
[2024-10-09 18:09:18] debug: 	zh:zstack:unpi:parser: <-- [254,28,68,129,0,0,18,0,253,5,1,1,0,116,0,99,190,234,0,0,8,24,168,10,85,0,33,1,0,39,33,28]
[2024-10-09 18:09:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,18,0,253,5,1,1,0,116,0,99,190,234,0,0,8,24,168,10,85,0,33,1,0,39,33,28]
[2024-10-09 18:09:18] debug: 	zh:zstack:unpi:parser: <-- [173]
[2024-10-09 18:09:18] debug: 	zh:zstack:unpi:parser: --- parseNext [254,28,68,129,0,0,18,0,253,5,1,1,0,116,0,99,190,234,0,0,8,24,168,10,85,0,33,1,0,39,33,28,173]
[2024-10-09 18:09:18] debug: 	zh:zstack:unpi:parser: --> parsed 28 - 2 - 4 - 129 - [0,0,18,0,253,5,1,1,0,116,0,99,190,234,0,0,8,24,168,10,85,0,33,1,0,39,33,28] - 173
[... cut ...]
[2024-10-09 18:09:27] debug: 	zhc:lumi: MCCGQ11LM: Processed data into payload {"voltage":3045,"battery":100,"device_temperature":23,"power_outage_count":8,"trigger_count":0,"contact":false}
[2024-10-09 18:09:27] debug: 	zhc:lumi: MCCGQ11LM: Processed data into payload {"voltage":3045,"battery":100,"device_temperature":23,"power_outage_count":8,"trigger_count":0,"contact":false}
[2024-10-09 18:09:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x00158d000ae14e64', payload '{"battery":100,"contact":false,"device_temperature":23,"linkquality":61,"power_outage_count":8,"trigger_count":0,"voltage":3045}'

Koenkk, 2024-10-10:
Does it go from `None` to `Unknown` in HA with this? Because I still see `"action":""`. Note that the `action` sensor will be removed in Z2M 2.0.0: https://github.com/Koenkk/zigbee2mqtt/discussions/24198

dnedeltchev, 2024-10-10:
Yes, it goes from None to Unknown. But since these are legacy and will be
replaced by event, then leave it. No need to spend time fixing something
that will be removed eventually. I have excluded these entities from
alerting.
Many thanks!

closed 2024-12-19 by github-actions[bot]