Koenkk/zigbee2mqtt #24267: Zemismart ZN-LC1E showing as TS0601'

pereiraru, 2024-10-08:
### What happened?

Zemistart smart curtain switch not listed in Zigbee2mqtt properly

this is the device, and confirmed as supported here: https://www.zigbee2mqtt.io/devices/ZN-LC1E.html

in zigbee2mqtt (tried dev as well with no luck, the device is identified as below:

[image]

what it displays under developer

[image]

`
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE204_ettzhnwk',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;`

### What did you expect to happen?

that the switch is recognize.

### How to reproduce it (minimal and precise)

put device in pair mode by pressing the middle button for 5 seconds, activate search in zigbee 2 mqtt

### Zigbee2MQTT version

Zigbee2MQTT version 1.40.2-dev commit: 3bb4af2

### Adapter firmware version

20240710

### Adapter

zStack3x0

### Setup

homeassistant addon on HA in a nuc, SLZB-06 on the latest firmware version

rp2019433, 2024-11-06:
It seems your device has the same code (_TZE204_ettzhnwk) as this one [24150](https://github.com/Koenkk/zigbee2mqtt/issues/24150).

Still looking how to support it on Z2M. Mine is not working.

battosai30, 2024-11-26:
I have the same problem. But the model is not the same as described here : https://www.zigbee2mqtt.io/devices/ZN-LC1E.html 
I have both "real" ZN-LCE1E" (no problem with it) and this "new" version which is not supported. 
I noticed a difference between this models : the "new" one has RGB leds.

closed 2025-02-02 by github-actions[bot]