Koenkk/zigbee2mqtt #24268: Action label - Some devices don't work anymore with Node Red

gregolin, 2024-10-08:
### What happened?

I'm facing issue with some devices and Node Red.
After last upgrade, it seems NodeRed is using the node label "action" to themselves, so any value there (string) result in error message (Node Red log):

"InputError: Invalid action format: unlock"

And all actions following the node result in error. Is "action" a right/best label for this node (for a door lock for instance)? I'm thinking if ACTION could be some protected word, something similar that.

Here is another user reporting the same issue:
(zachowj/node-red-contrib-home-assistant-websocket#1489)
Is there any way that I could change the label by myself or it must be done in the add-on level?
This is the object which is returned:

> action: "unlock"
action_source: 2
action_source_name: "manual"
action_user: 0
auto_relock_time: null
battery: 100
battery_low: null
linkquality: 120
lock_state: "locked"
pin_code: null
sound_volume: null
state: "LOCK"

Thanks.

### What did you expect to happen?

The node returns with some other label than "action", which seems used as a reserved word nowadays.

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

2024.10.1

### Adapter

Sonoff

### Setup

Add-on working under Intel NUC

mundschenk-at, 2024-10-08:
Seems like a Node Red issue.

mundschenk-at, 2024-10-08:
No idea. There is probably a specific plugin for Node Red that interfaces with Z2M? "Label" has no meaning in MQTT.

gregolin, 2024-10-08:
> No idea. There is probably a specific plugin for Node Red that interfaces with Z2M? "Label" has no meaning in MQTT.

Label I mean the name/id/definition of the sensor (in the HA), which is created when we add some device in the HA by Z2M. If I could change this name/label, the problem is fixed.

gregolin, 2024-10-08:
For this "action", under EXPOSES tab:

[image]

gregolin, 2024-10-09:
I have imagined about that... so my question is about some possibility to change the Z2M for this device (maybe there are other reserved words as well), in the add-on level.

mundschenk-at, 2024-10-09:
You need to talk to whatever Node Red plugin you use to connect Node Red and Home Assistant or Z2M.

gregolin, 2024-10-09:
I use the most popular Node Red add-on for HA [(Git)](https://github.com/hassio-addons/addon-node-red) certainly they are using ACTION label for some reason, so the easier way to fix it would be to crate this node with a different word.
I really think easier to change this word, since it is created when some device is added in the HA.

mundschenk-at, 2024-10-09:
That's not how this works. You can rename all entities yourself in HA (but really the issue lies with the Node Red plugin that interfaces with HA and should be fixed there).

gregolin, 2024-10-09:
This is the point. It seems isn't related to entities name, which I can change it. See the entity's name below:

[image]

[image]

But when I access the device by Node Red, the node still comes as ACTION:

[image]

I really can't understand where this ACTION is defined/created.

mundschenk-at, 2024-10-09:
But just to be sure, do you interface Z2M with Node Red directly or via Home Assistant? The last screenshot appears to show something that is not mediated by HA.

gregolin, 2024-10-09:
Via HA.
The last screenshot is the NodeRed debug screen, over HA.

mundschenk-at, 2024-10-09:
Well, it's just that zigbee2mqtt/Sala_Fechadura looks more like an MQTT topic (prefix) than something coming from Home Assistant.

gregolin, 2024-10-09:
> Well, then the issue obviously hasn't got anything to do with Z2M nor HA entity naming and needs to be found elsewhere (Node Red respectively the Node Red plugin/component that interfaces with Home Assistant - that doesn't mean the HA add-on, which is just a fancy packaging for a Docker container).
> 
> Most likely, the issue you experience is related to the recent renaming of service calls to actions in Home Assistant.

I think you are right about this suspicion.
https://www.home-assistant.io/blog/2024/10/02/release-202410/#improved-yaml-syntax-for-automations
It changed from SERVICE to ACTION.
Thanks.

closed 2024-12-19 by github-actions[bot]

samfrjn11, 2025-01-05:
This issue has indeed nothing to do with Z2M. However, since this thread is on the internet and as I encountered the same issue; If anyone would stumble upon this issue as well:
Edit your action-node(s) and check the checkbox next to "Block input overrides".

carpii, 2025-01-15:
> Edit your action-node(s) and check the checkbox next to "Block input overrides".

I had the same problem. This was a frustrating breaking-change which I didnt see documented anywhere. 

Without this option ticked, any incoming message will ovrride any value you set in the actual node. Makes no sense to me but whatever.