Koenkk/zigbee2mqtt #24271: ‎F20T60A negative value

robyevolution, 2024-10-08:
### What happened?

Hi, I have been using this power meter for more than a year without problems, today I updated the firmware of the device and the consumption value became negative, how can I change it to positive?

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

.

### Adapter

SLZB-06M

### Setup

NUC i5

alexchini, 2024-10-10:
Having the same problem here  with the last OTA update the device meter send a negative power value, all the others values are ok, this is the mqtt topic:
{
  "linkquality": 32,
  **"power": -308,**
  "power_alarm": null,
  "power_alarm_active": false,
  "power_alarm_active_value": 0,
  "power_alarm_enabled": 1,
  "power_alarm_wh_threshold": 3000,
  "power_apparent": 392,
  "update": {
    "installed_version": 6637055,
    "latest_version": 6636035,
    "state": "idle"
  },
  "update_available": false
}

This is a biticino device equivalent to Legrand 412015. Don't know if there is any way to convert the negative value as i suppose is not possible to downgrade the firmware  in OTA.

tommaineult, 2024-10-13:
Hello @robyevolution @alexchini ,
I encountered the same problem : just have to change the direction of  the current transformer.
Hope this will help.

alexchini, 2024-10-14:
> Hello @robyevolution @alexchini , I encountered the same problem : just have to change the direction of the current transformer. Hope this will help.

Ah great i used an HA helper to convert in positive the value but your way is for sure better i will try it, thanks.

closed 2024-12-22 by github-actions[bot]