Koenkk/zigbee2mqtt #24283: Lidl remote FB20-002 doesn't register first press after being idle for a while

Virusmater, 2024-10-10:
### What happened?

If Lidl remote FB20-002 is idle for a while (a few days, no idea what exact period) then it doesn't register the first button press. All consequent button presses work as expected.

Logs when the first press isn't registered:
```
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: <-- [254,12,69,202,68,88,164,178,17,254,255,249,227,180,180,198,69]
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,12,69,202,68,88,164,178,17,254,255,249,227,180,180,198,69]
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --> parsed 12 - 2 - 5 - 202 - [68,88,164,178,17,254,255,249,227,180,180,198] - 69
[2024-10-10 19:13:49] debug: 	zh:zstack:znp: AREQ: <-- ZDO - tcDeviceInd - {"nwkaddr":22596,"extaddr":"0xb4e3f9fffe11b2a4","parentaddr":50868}
[2024-10-10 19:13:49] debug: 	zh:controller: Device '0xb4e3f9fffe11b2a4' joined
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-10 19:13:49] debug: 	zh:controller: Device '0xb4e3f9fffe11b2a4' accepted by handler
[2024-10-10 19:13:49] debug: 	zh:controller: Not interviewing '0xb4e3f9fffe11b2a4', completed 'true', in progress 'false'
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: <-- [254,5,69,196,180,198,1,211,47,11]
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,5,69,196,180,198,1,211,47,11]
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --> parsed 5 - 2 - 5 - 196 - [180,198,1,211,47] - 11
[2024-10-10 19:13:49] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":50868,"relaycount":1,"relaylist":[12243]}
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: <-- [254,13,69,193,68,88,68,88,164,178,17,254,255,249,227,180,128,161]
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,13,69,193,68,88,68,88,164,178,17,254,255,249,227,180,128,161]
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --> parsed 13 - 2 - 5 - 193 - [68,88,68,88,164,178,17,254,255,249,227,180,128] - 161
[2024-10-10 19:13:49] debug: 	zh:zstack:znp: AREQ: <-- ZDO - endDeviceAnnceInd - {"srcaddr":22596,"nwkaddr":22596,"ieeeaddr":"0xb4e3f9fffe11b2a4","capabilities":128}
[2024-10-10 19:13:49] debug: 	zh:zstack: Discovering route to 22596
[2024-10-10 19:13:49] debug: 	zh:zstack:znp: SREQ: --> ZDO - extRouteDisc - {"dstAddr":22596,"options":0,"radius":30}
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:writer: --> frame [254,4,37,69,68,88,0,30,102]
[2024-10-10 19:13:49] debug: 	zh:controller: Device announce '0xb4e3f9fffe11b2a4'
[2024-10-10 19:13:49] debug: 	z2m: Device 'Guest Blinds Remote' announced itself
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-10 19:13:49] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Guest Blinds Remote","ieee_address":"0xb4e3f9fffe11b2a4"},"type":"device_announce"}'
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: <-- [254,1,101,69,0,33]
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,101,69,0,33]
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 5 - 69 - [0] - 33
[2024-10-10 19:13:49] debug: 	zh:zstack:znp: SRSP: <-- ZDO - extRouteDisc - {"status":0}
[2024-10-10 19:13:49] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

Example of properly registered press logs:
```
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: <-- [254,7,69,196,68,88,2,180,198,211,47,22]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,7,69,196,68,88,2,180,198,211,47,22]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --> parsed 7 - 2 - 5 - 196 - [68,88,2,180,198,211,47] - 22
[2024-10-10 19:25:30] debug: 	zh:zstack:znp: AREQ: <-- ZDO - srcRtgInd - {"dstaddr":22596,"relaycount":2,"relaylist":[50868,12243]}
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: <-- [254,24,68,129,0,0,6,0,68,88,1,1,0,72,0,58,221,175,0,0,4,1,110,253,0,211,47,27,182]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,24,68,129,0,0,6,0,68,88,1,1,0,72,0,58,221,175,0,0,4,1,110,253,0,211,47,27,182]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --> parsed 24 - 2 - 4 - 129 - [0,0,6,0,68,88,1,1,0,72,0,58,221,175,0,0,4,1,110,253,0,211,47,27] - 182
[2024-10-10 19:25:30] debug: 	zh:zstack:znp: AREQ: <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":22596,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":72,"securityuse":0,"timestamp":11525434,"transseqnumber":0,"len":4,"data":{"type":"Buffer","data":[1,110,253,0]}}
[2024-10-10 19:25:30] debug: 	zh:controller: Received payload: clusterID=6, address=22596, groupID=0, endpoint=1, destinationEndpoint=1, wasBroadcast=false, linkQuality=72, frame={"header":{"frameControl":{"frameType":1,"manufacturerSpecific":false,"direction":0,"disableDefaultResponse":false,"reservedBits":0},"manufacturerCode":null,"transactionSequenceNumber":110,"commandIdentifier":253},"payload":{"value":0,"data":{"type":"Buffer","data":[]}},"command":{"ID":253,"parameters":[{"name":"value","type":32},{"name":"data","type":1008}],"name":"tuyaAction"}}
[2024-10-10 19:25:30] debug: 	zh:controller:endpoint: ZCL command 0xb4e3f9fffe11b2a4/1 genOnOff.defaultRsp({"cmdId":253,"statusCode":0}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":110,"writeUndiv":false})
[2024-10-10 19:25:30] debug: 	zh:zstack: sendZclFrameToEndpointInternal 0xb4e3f9fffe11b2a4:22596/1 (0,0,1)
[2024-10-10 19:25:30] debug: 	zh:zstack:znp: SREQ: --> AF - dataRequest - {"dstaddr":22596,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":236,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[24,110,11,253,0]}}
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:writer: --> frame [254,15,36,1,68,88,1,1,6,0,236,0,30,5,24,110,11,253,0,71]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-10 19:25:30] debug: 	z2m: Received Zigbee message from 'Guest Blinds Remote', type 'commandTuyaAction', cluster 'genOnOff', data '{"data":{"data":[],"type":"Buffer"},"value":0}' from endpoint 1 with groupID 0
[2024-10-10 19:25:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Guest Blinds Remote', payload '{"action":"on","linkquality":72}'
[2024-10-10 19:25:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Guest Blinds Remote', payload '{"action":"","linkquality":72}'
[2024-10-10 19:25:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Guest Blinds Remote/action', payload 'on'
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: <-- [254,1,100,1,0,100]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,1,100,1,0,100]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --> parsed 1 - 3 - 4 - 1 - [0] - 100
[2024-10-10 19:25:30] debug: 	zh:zstack:znp: SRSP: <-- AF - dataRequest - {"status":0}
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: <-- [254,3,68,128,0,1,236,42]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --- parseNext [254,3,68,128,0,1,236,42]
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --> parsed 3 - 2 - 4 - 128 - [0,1,236] - 42
[2024-10-10 19:25:30] debug: 	zh:zstack:znp: AREQ: <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":236}
[2024-10-10 19:25:30] debug: 	zh:zstack:unpi:parser: --- parseNext []
```

### What did you expect to happen?

The first button press works fine even after not being touched for a while.

### How to reproduce it (minimal and precise)

1. Pair FB20-002
2. Check that buttons register
3. Don't touch the remote for a day or two
4. Press any button

### Zigbee2MQTT version

1.39.1-1

### Adapter firmware version

20210708

### Adapter

ITead Sonoff Zigbee 3.0 USB Dongle Plus

### Setup

HomeAssistant OS on proxmox

Virusmater, 2024-10-11:
In my case I cannot reproduce it as easy as just restart zigbee2mqtt. It works fine after restart and breaks after being idle without any restarts.

merlinschumacher, 2024-10-12:
There is a known issue with this remote regarding the on button:
https://github.com/Koenkk/zigbee2mqtt/issues/6509

Virusmater, 2024-10-12:
The initial issue described in https://github.com/Koenkk/zigbee2mqtt/issues/6509 was fixed. 
It got followed up with another, unrelated problem described in the current issue. The current issue was created to no hijack the unrelated issue that got fixed.

closed 2024-12-22 by github-actions[bot]