Koenkk/zigbee2mqtt #24287: EnOcean PTM 215Z not registering button clicks (action)

Atiunderscore, 2024-10-11:
### What happened?

I've bought a 3-pack Friends of Hue Smart Switches and I'm in the process of connecting these in Home Assistant through Z2M. Connecting the first one was a breeze and everything functions as expected:

[image]
[image]

So I just the 'Permit join (All) button, click 2 random buttons and then hold the top-right button to join. Once joined I can use the action trigger in automations. Also when I press one of the buttons I can see within the device information in Z2M the 'Action' value rapidly changing to whatever button I pressed and reverting to status 'N/A'.

Joining the second switch also went easy, but from some reason this switch is nog registering the buttons when pressed. The state of action stays 'N/A':

[image]
[image]

Because the action is not being registered I am also unable to select these as a trigger in an automation:

[image]

In comparison to the first switch I joined:

[image]

I tried restarting Home Assistant, rejoining through various other Philips Hue devices and factory resetting the switch. Nothing seems to work. Is there anyone here that can help me resolve this issue?

Version and adapter information:

[image]

[image]

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.2

### Adapter

Nabu_Casa_SkyConnect_v1.0 (ember driver)

### Setup

Add-on on Home Assistant OS

chris-1243, 2024-10-11:
I would factory reset all your PTM first. Then, strictly follow the pairing process. While pairing, do not play with the device. Only the buttons which are required during this process.

https://www.zigbee2mqtt.io/devices/PTM_215Z.html#enocean-ptm%2520215z

Some tips😉 check this [link](https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices) before any pairing try. Avoid to pair thoses devices as `brodacast` as you will see some latency in your network. Try to use the `unicast` mode.

Never pair via the `permit join all`. Always select a router as close as possible and use the `permit join via "router_xy"`. If your router is not listed, don't worry and check the cluster tab. Find if the endpoint 242 is present. At least the `output cluster, greenPower` should be sufficient.

I have 7 PTM215Z and 1 PTM215ZE and they are all working perfectly.

Atiunderscore, 2024-10-14:
> I would factory reset all your PTM first. Then, strictly follow the pairing process. While pairing, do not play with the device. Only the buttons which are required during this process. [...]

Thank you for the reply. I force deleted the PTM215Z switch from Z2M. After that I factory resetted the switch. I then checked the list on the following website to see which of my other Zigbee devices is best to pair with:

https://www.zigbee2mqtt.io/advanced/zigbee/01_zigbee_network.html#green-power-devices

I have a couple of "Philips 9290022169" bulbs, so I picked the one closest to the switch. Afer that I joined the switch on one of the closests bulbs by following the steps from page you linked, but still no luck. I also tried joining it to every single other Zigbee router, but also no success.

You are also talking about "endpoint 242", but I don't know exactly what this means. I do see it stated under clusters and dev console:

[image]

[image]

Now I know these can work, because I have a PTM215Z switch with configurable actions. But sadly 2 of the same other switches don't. Do you or anyone else have any tips that may help?

Atiunderscore, 2024-10-14:
Wow.. I was breaking my head on this one. I've read the instructions so many times, but I somehow read over the following:

"Once the device is paired you need to confirm the channel. To do this press A1 and B0 together. Important: don't press any other buttons between this and the pairing."

After confirming the channel it now works. Thank you for helping out @chris-1243

closed 2024-10-14 by Atiunderscore

chris-1243, 2024-10-14:
> "Once the device is paired you need to confirm the channel. To do this press A1 and B0 together. Important: don't press any other buttons between this and the pairing."

That is the most important part of the process !

Glad everything is now working