Koenkk/zigbee2mqtt #24290: After installing zigbee2mqtt on Raspberry PI 4B, zboss was used as adapter, but startup failed

chenyixi1997, 2024-10-11:
### What happened?

English is not my native language. Please forgive me for any mistakes
My device is raspberry PI 4b.Adapter is nRF52840 DK, write the firmware in the https://github.com/kardia-as/nrf-zboss-ncp/tree/main/nRF52840_dongle. After starting zigbee2mqtt(npm start) according to the tutorial,It shows that the startup is not successful, there are some errors, it seems that the command is not transmitted normally, how to solve it?For configuration information, I set up my mqtt, as well as serial.

### What did you expect to happen?

I expect to be able to start zigbee2mqtt normally

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

v2.6.0_ncp_zboss_max_mem_nrf52840_dongle

### Adapter

zboss

### Setup

Pi 4

### Debug log

[image]

chenyixi1997, 2024-10-11:
debug_log:
> zigbee2mqtt@1.40.2 start
> node index.js

Starting Zigbee2MQTT without watchdog.
[2024-10-11 17:54:37] info: 	z2m: Logging to console, file (filename: log.log)
[2024-10-11 17:54:37] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #e06848d)
[2024-10-11 17:54:37] info: 	z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-11 17:54:38] info: 	zh:zboss: ZBOSS Adapter starting
[2024-10-11 17:54:38] info: 	zh:zboss:driv: Driver connecting
[2024-10-11 17:54:38] info: 	zh:zboss:uart: NCP reset
[2024-10-11 17:54:38] info: 	zh:zboss:uart: Serial port opened
[2024-10-11 17:54:38] info: 	zh:zboss:uart: UART starting
[2024-10-11 17:54:38] info: 	zh:zboss:driv: Driver connected
[2024-10-11 17:54:38] info: 	zh:zboss:driv: Driver startup
[2024-10-11 17:54:40] error: 	zh:zboss:uart: --> Error: Error: 0 after 2000ms
[2024-10-11 17:54:40] error: 	zh:zboss:uart: -!- break waiting (0)
[2024-10-11 17:54:40] error: 	zh:zboss:uart: Can't send DATA frame (0,0,0): dead0c0006c07c7df70000140001
[2024-10-11 17:54:50] error: 	zh:zboss:driv: ==> Error: Error: {"commandId":20,"tsn":1} after 10000ms
[2024-10-11 17:54:50] error: 	z2m: Error while starting zigbee-herdsman
[2024-10-11 17:54:50] error: 	z2m: Failed to start zigbee
[2024-10-11 17:54:50] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-10-11 17:54:50] error: 	z2m: Exiting...
[2024-10-11 17:54:50] error: 	z2m: Error: Failure send 20:{"version":0,"type":0,"commandId":20,"tsn":1,"payload":{}}
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/zboss/driver.ts:284:23
    at Queue.execute (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/utils/queue.ts:36:20)
    at ZBOSSDriver.execCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/zboss/driver.ts:265:16)
at ZBOSSDriver.getNetworkInfo (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/zboss/driver.ts:157:22)
    at ZBOSSDriver.needsToBeInitialised (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/zboss/driver.ts:143:24)
    at ZBOSSDriver.startup (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/zboss/driver.ts:91:13)
    at ZBOSSAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/zboss/adapter/zbossAdapter.ts:137:16)
at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:161:27)

kirovilya, 2024-10-11:
Please record the startup debug log.
```
advanced:
  log_level: debug
```

chenyixi1997, 2024-10-11:
> Please record the startup debug log.
> 
> ```
> advanced:
>   log_level: debug
> ```

Thank you for your reply
Do I copy the following into my configuration.yaml file and then start zigbee2mqtt again?
advanced:
  log_level: debug

kirovilya, 2024-10-11:
yes, if you don't have a similar section in your config file

chenyixi1997, 2024-10-12:
When I add debug, this is new and different log information.I checked my installation steps，but I don't understand why such a problem occurs：

Starting Zigbee2MQTT without watchdog.
[2024-10-12 09:49:25] info:     z2m: Logging to console, file (filename: log.log)
[2024-10-12 09:49:25] debug:     z2m: Can't load state from file /opt/zigbee2mqtt/data/state.json (doesn't exist)
[2024-10-12 09:49:25] info:     z2m: Starting Zigbee2MQTT version 1.40.2 (commit #e06848d)
[2024-10-12 09:49:25] debug:     z2m: sd-notify loaded
[2024-10-12 09:49:25] info:     z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-12 09:49:25] debug:     z2m: Using zigbee-herdsman with settings: '"{\"network\":{\"panID\":58682,\"extendedPanID\":[38,194,254,210,69,100,223,174],\"channelList\":[11],\"networkKey\":\"HIDDEN\"},\"databasePath\":\"/opt/zigbee2mqtt/data/database.db\",\"databaseBackupPath\":\"/opt/zigbee2mqtt/data/database.db.backup\",\"backupPath\":\"/opt/zigbee2mqtt/data/coordinator_backup.json\",\"serialPort\":{\"path\":\"/dev/serial/by-id/usb-SEGGER_J-Link_000683762835-if00\",\"adapter\":\"zboss\"},\"adapter\":{\"disableLED\":false}}"'
[2024-10-12 09:49:25] debug:     zh:zboss: Adapter concurrent: 8
[2024-10-12 09:49:25] debug:     zh:controller: Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":"HIDDEN","panID":58682,"extendedPanID":[38,194,254,210,69,100,223,174],"channelList":[11]},"serialPort":{"path":"/dev/serial/by-id/usb-SEGGER_J-Link_000683762835-if00","adapter":"zboss"},"adapter":{"disableLED":false},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json"}'
[2024-10-12 09:49:25] info:     zh:zboss: ZBOSS Adapter starting
[2024-10-12 09:49:25] info:     zh:zboss:driv: Driver connecting
[2024-10-12 09:49:25] info:     zh:zboss:uart: NCP reset
[2024-10-12 09:49:25] debug:     zh:zboss:uart: Opening serial port with {"path":"/dev/serial/by-id/usb-SEGGER_J-Link_000683762835-if00","baudRate":115200,"rtscts":false,"autoOpen":false}
[2024-10-12 09:49:25] info:     zh:zboss:uart: Serial port opened
[2024-10-12 09:49:25] info:     zh:zboss:uart: UART starting
[2024-10-12 09:49:25] info:     zh:zboss:driv: Driver connected
[2024-10-12 09:49:25] info:     zh:zboss:driv: Driver startup
[2024-10-12 09:49:25] debug:     zh:zboss:driv: ==> GET_JOINED(20): {}
[2024-10-12 09:49:25] debug:     zh:zboss:driv: ==> FRAME: {"version":0,"type":0,"commandId":20,"tsn":1,"payload":{}}
[2024-10-12 09:49:25] debug:     zh:zboss:uart: --> FRAME: 0000140001
[2024-10-12 09:49:25] debug:     zh:zboss:uart: --> package type 6, flags c0{"isACK":false,"retransmit":false,"sequence":0,"ACKseq":0,"isFirst":true,"isLast":true}
[2024-10-12 09:49:25] debug:     zh:zboss:uart: --> PACK: dead0c0006c07c7df70000140001
[2024-10-12 09:49:25] debug:     zh:zboss:uart: --> DATA (0,0,0): dead0c0006c07c7df70000140001
[2024-10-12 09:49:25] debug:     zh:zboss:uart: --> [dead0c0006c07c7df70000140001]
[2024-10-12 09:49:25] debug:     zh:zboss:uart: -?- waiting (0)
[2024-10-12 09:49:27] error:     zh:zboss:uart: --> Error: Error: 0 after 2000ms
[2024-10-12 09:49:27] error:     zh:zboss:uart: -!- break waiting (0)
[2024-10-12 09:49:27] error:     zh:zboss:uart: Can't send DATA frame (0,0,0): dead0c0006c07c7df70000140001
[2024-10-12 09:49:27] debug:     zh:zboss:uart: --> error Error: sendDATA error: try 1: Error: 0 after 2000ms

Thank you for your help~

chenyixi1997, 2024-10-12:
I am a newbie and looking at these logs, I feel that zigbee2mqtt is unable to send Get_JOINED this Frame to my Adapter. So the boot failed, right?

kirovilya, 2024-10-12:
> /dev/serial/by-id/usb-SEGGER_J-Link_000683762835-if00

you have a strange port specified. as if it is connected via jlink. perhaps it is necessary to connect differently.
I see that the port is opened and data is sent to it, but there is no response. it does not work.

chenyixi1997, 2024-10-12:
Thank you, when I writed the hex file to nRF52840 using nRF Connect for Desktop, I connected it directly to my Raspberry PI via USB, and it showed this by-id, is this operation correct? What should I do if it's not correct?

chenyixi1997, 2024-10-12:
I tried to change the way I connected to the nRF52840 DK, but it didn't work. In the process of writing the firmware, I used J-Link to do it, and then in the Linux environment, the by-id displayed is like this:
usb-SEGGER_J-Link_000683762835-if00
I tried to change the configuration file, and /dev/ttyACM0 got the same error.

kirovilya, 2024-10-13:
strange. I checked the work on nrf usb-sticks, in your case DK is used and it may need a different firmware, similar to the nRF5340 DK https://github.com/kardia-as/nrf-zboss-ncp/tree/main/nRF5340DK
but you used https://github.com/kardia-as/nrf-zboss-ncp/tree/main/nRF52840_dongle which is for dongles

chenyixi1997, 2024-10-14:
Thank you for your reply. After testing, I learned that the problem was indeed caused by my failure to use proper firmware.At the moment, I change for the website during the firmware writing to nRF52840DK(https://docs.nordicsemi.com/bundle/ncs-latest/page/nrf/protocols/zigbee/architectures.html#ug-zigbee-platform-design-ncp-details),and then it can be identified as Zigbee NCP normally. At present, starting zigbee2mqtt can enter the step of connecting to MQTT server. It seems that there are still mqtt problems to be solved, and I am trying to deal with them. Thank you for your help!

chenyixi1997, 2024-10-14:
I was able to connect to my MQTT Server after modifying the configuration file, but then the following log appeared:

[2024-10-14 17:15:33] info:     z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload '{"state":"online"}'
[2024-10-14 17:15:33] debug:     zh:zboss:driv: ==> GET_MODULE_VERSION(1): {}
[2024-10-14 17:15:33] debug:     zh:zboss:driv: ==> FRAME: {"version":0,"type":0,"commandId":1,"tsn":29,"payload":{}}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> FRAME: 000001001d
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> package type 6, flags c4{"isACK":false,"retransmit":false,"sequence":1,"ACKseq":0,"isFirst":true,"isLast":true}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> PACK: dead0c0006c484b891000001001d
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> DATA (1,3,0): dead0c0006c484b891000001001d
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> [dead0c0006c484b891000001001d]
[2024-10-14 17:15:33] debug:     zh:zboss:uart: -?- waiting (1)
[2024-10-14 17:15:33] debug:     zh:zboss:read: <<<  DATA [dead05000611c0]
[2024-10-14 17:15:33] debug:     zh:zboss:read: <<< FRAME [05000611c0]
[2024-10-14 17:15:33] debug:     zh:zboss:uart: <-- package type 6, flags 11{"isACK":true,"retransmit":false,"sequence":0,"ACKseq":1,"isFirst":false,"isLast":false}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: <-- ACK (1)
[2024-10-14 17:15:33] debug:     zh:zboss:uart: -+- waiting (1) success
[2024-10-14 17:15:33] debug:     zh:zboss:read: <<<  DATA [dead1a0006c48a416c000101001d00000100000063630b0315010000]
[2024-10-14 17:15:33] debug:     zh:zboss:read: <<< FRAME [1a0006c48a416c000101001d00000100000063630b0315010000]
[2024-10-14 17:15:33] debug:     zh:zboss:uart: <-- package type 6, flags c4{"isACK":false,"retransmit":false,"sequence":1,"ACKseq":0,"isFirst":true,"isLast":true}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> ACK (1)
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> package type 6, flags 11{"isACK":true,"retransmit":false,"sequence":0,"ACKseq":1,"isFirst":false,"isLast":false}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: -->  ACK: dead05000611c0
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> DATA (2,1,0): dead05000611c0
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> [dead05000611c0]
[2024-10-14 17:15:33] debug:     zh:zboss:uart: <-- FRAME: 000101001d00000100000063630b0315010000
[2024-10-14 17:15:33] debug:     zh:zboss:driv: <== Frame: {"version":0,"type":1,"commandId":1,"tsn":29,"payload":{"category":0,"status":0,"fwVersion":1,"stackVersion":51077987,"protocolVersion":277}}
[2024-10-14 17:15:33] debug:     zh:zboss:driv: ==> GET_COORDINATOR_VERSION(36): {}
[2024-10-14 17:15:33] debug:     zh:zboss:driv: ==> FRAME: {"version":0,"type":0,"commandId":36,"tsn":30,"payload":{}}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> FRAME: 000024001e
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> package type 6, flags c8{"isACK":false,"retransmit":false,"sequence":2,"ACKseq":0,"isFirst":true,"isLast":true}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> PACK: dead0c0006c8e9a599000024001e
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> DATA (2,1,0): dead0c0006c8e9a599000024001e
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> [dead0c0006c8e9a599000024001e]
[2024-10-14 17:15:33] debug:     zh:zboss:uart: -?- waiting (2)
[2024-10-14 17:15:33] debug:     zh:zboss:read: <<<  DATA [dead0500062111dead0f0006c86afa67000124001e000000]
[2024-10-14 17:15:33] debug:     zh:zboss:read: <<< FRAME [0500062111]
[2024-10-14 17:15:33] debug:     zh:zboss:uart: <-- package type 6, flags 21{"isACK":true,"retransmit":false,"sequence":0,"ACKseq":2,"isFirst":false,"isLast":false}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: <-- ACK (2)
[2024-10-14 17:15:33] debug:     zh:zboss:read: <<< TAIL [dead0f0006c86afa67000124001e000000]
[2024-10-14 17:15:33] debug:     zh:zboss:read: <<< FRAME [0f0006c86afa67000124001e000000]
[2024-10-14 17:15:33] debug:     zh:zboss:uart: <-- package type 6, flags c8{"isACK":false,"retransmit":false,"sequence":2,"ACKseq":0,"isFirst":true,"isLast":true}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> ACK (2)
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> package type 6, flags 21{"isACK":true,"retransmit":false,"sequence":0,"ACKseq":2,"isFirst":false,"isLast":false}
[2024-10-14 17:15:33] debug:     zh:zboss:uart: -->  ACK: dead0500062111
[2024-10-14 17:15:33] debug:     zh:zboss:uart: -+- waiting (2) success
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> DATA (3,2,0): dead0500062111
[2024-10-14 17:15:33] debug:     zh:zboss:uart: --> [dead0500062111]
[2024-10-14 17:15:33] debug:     zh:zboss:uart: <-- FRAME: 000124001e000000
[2024-10-14 17:15:33] debug:     zh:zboss:driv: <== Frame: {"version":0,"type":1,"commandId":36,"tsn":30,"payload":{"category":0,"status":0,"version":0}}
[2024-10-14 17:15:33] error:     z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'zh')
    at Bridge.publishDevices (/opt/zigbee2mqtt/lib/extension/bridge.ts:781:43)
    at Bridge.start (/opt/zigbee2mqtt/lib/extension/bridge.ts:175:20)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Controller.callExtensions (/opt/zigbee2mqtt/lib/controller.ts:399:17)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:218:9)
    at start (/opt/zigbee2mqtt/index.js:154:5))
[2024-10-14 17:15:33] info:     z2m: Started frontend on port 8080

I can't understand：
 error:     z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'zh')
What should this error be caused by?

kirovilya, 2024-10-14:
try to run it again. this shouldn't be happening

chenyixi1997, 2024-10-15:
I have tried many times, but this problem still occurs, is it caused by MQTT server?

chenyixi1997, 2024-10-15:
I found this problem with similar log information, does it seem to be caused by the same cause?
https://github.com/Koenkk/zigbee2mqtt/issues/24317

chenyixi1997, 2024-10-15:
I have checked the log information and the information received in my mqtt server, and it seems that after connected to MQTT server, a message with the topic: zigbee2mqtt/bridge/state and the load: {"state":"online"} needs to be sent to MQTT server. Indicates that the server is successfully connected and is online. This process also sends some MQTT messages, including theme respectively topics: zigbee2mqtt/bridge/info and zigbee2mqtt/bridge/extensions, zigbee2mqtt/bridge/info contains a large number of relevant information. The MQTT server can receive these messages, but the extensions message is empty(I am not sure that's right), and some error occurred during the process, which caused the zigbee2mqtt/bridge/state message to not be published successfully. So the payload that MQTT server finally receives is {"state":"offline"}.

kirovilya, 2024-10-15:
@chenyixi1997  Can you check it on the dev version (edge)? Maybe they've already fixed it.

chenyixi1997, 2024-10-15:
I don't understand what you mean, could you please tell me how to check it?

kirovilya, 2024-10-15:
https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html

chenyixi1997, 2024-10-15:
Do you mean the firmware version of my nRF52840DK?

chenyixi1997, 2024-10-15:
Do you mean that I need to Switch to dev branch according to the Linux part of Switch to the dev branch, right? Looks like the instruction in the last step: 
**_sudo systemctl start zigbee2mqtt_**
I need to switch it to the "**_npm start_**" instruction, right? 
Thank you for your help. Thank you so much.

chenyixi1997, 2024-10-15:
I use the following two instructions will get the following error, it seems that there is no way to switch branches:
instructions:
git fetch origin dev
git fetch origin --unshallow
error:
error: pathspec 'dev' did not match any file(s) known to git

chenyixi1997, 2024-10-16:
Thanks for your help, I tried to reinstall zigbee2mqtt on my Raspberry PI today, and my problem can be solved after the re-installation.

chenyixi1997, 2024-10-17:
I found that whenever I re-installed zigbee2mqtt, my first npm start started normally, but after I closed zigbee2mqtt by ctrl+c, I started npm start again, and there was an error：
**_error:     z2m: Failed to call 'Bridge' 'start'_**

chenyixi1997, 2024-10-18:
I find that every time I use npm start to start zigbee2mqtt, I need to delete data/database.db before I can start zigbee2mqtt normally. I don't know the specific reasons

jaymemaurice, 2024-11-30:
> Thank you for your reply. After testing, I learned that the problem was indeed caused by my failure to use proper firmware.At the moment, I change for the website during the firmware writing to [...]

I have the exact same issue. I have the makerdiary nrf52840-mdk dongle and flashed the firmware as recommended for the ZHA [zigpy-zboss](https://github.com/kardia-as/zigpy-zboss) experimental radios as released by @DamKast in https://github.com/kardia-as/nrf-zboss-ncp . I would imagine at some point, the issue of which firmware is supported will lead to many complaints. 

@chenyixi1997 Did you find a pre-built firmware binary on nordicsemi's website!?! Can you provide a direct link here?

AapoTahkola, 2024-12-10:
I get similar error on home assistant OS on proxmox virtual machine both on esp32c6 based coordinator and also on nRF52840 based coordinator. I ended up installing zigbee2mqtt on debian container and pointing mqtt to ha and voila everything works now. i am using nRF52840 haven't tested esp32c6 since resolving this.

closed 2025-02-17 by github-actions[bot]