Koenkk/zigbee2mqtt #24295: [Feature request]: manual ota update from UI by uploading our own file.

KipK, 2024-10-11:
### Is your feature request related to a problem? Please describe

Sometime we want to downgrade, sometime the ota implementation for a device prevent to use local otaindex file ( lixee.js as example ), and sometime you just use custom fw

### Describe the solution you'd like

It would be a good thing having the ability to select and upload our own ota file for a device in ota page, and force an update.

### Describe alternatives you've considered

This should test the ota header metas to prevent mess-up with wrong file, but this would be a good feature on UX point of view, 

### Additional context

[image]

Nerivec, 2024-10-12:
What prevents [local OTA index](https://www.zigbee2mqtt.io/guide/usage/ota_updates.html#local-ota-index-and-firmware-files) from working?

PS: that ZLinky_TIC is in the official OTA index from what I can see: https://github.com/Koenkk/zigbee-OTA/blob/b8dee47ba1ff6e79ec6f249b666b9631d71c406e/index.json#L1668

KipK, 2024-10-12:
I think it's this https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/src/lib/ota/lixee.ts

On what I se it just download the same version whatever the Type meta is

closed 2024-12-22 by github-actions[bot]