Koenkk/zigbee2mqtt #24297: color（HS）use  wrong zigbee command

mosanwen, 2024-10-12:
### What happened?

when set color by "color(X/Y)" or "color (hs)",both of them send “move to color （XY）”
[image]
[image]

### What did you expect to happen?

 “color（hs）”should send “MOVE_TO_HUE_AND_SATURATION ”。

### Zigbee2MQTT version

1.40.1

### Adapter firmware version

EZSP v8 6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0 _USB_Dongle_Plus_V2_20221029145826-if00

### Setup

on homeassistant

Oggymator, 2024-12-30:
push

Koenkk, 2024-12-30:
This is a frontend issue, please create issues related to the frontend in this repository: https://github.com/nurikk/zigbee2mqtt-frontend

closed 2024-12-30 by Koenkk

mosanwen, 2024-12-31:
> I used some GLEDOPTO products and has the same problem, using ZigStar USB and newest Version of Z2M. Anybody an idea?
go to MQTT->CONFIGIRE->Publish a pagclet.
[image]