Koenkk/zigbee2mqtt #24299: LGYCDD01LM: zigbee2mqtt/FRIENDLY_NAME/get - No converter available for 'length' ("")

StevenLiew, 2024-10-12:
### What happened?

Tried to `/get` and `/set` the `length` but I am getting the error "No converter available for 'length' ("")".

I also noticed that the frontend UI does not have the option for `length`. 

### What did you expect to happen?

Able to `/get` and `/set` the `length` option correctly.

### How to reproduce it (minimal and precise)

`zigbee2mqtt/FRIENDLY_NAME/get` with the payload `{"length": ""}` as per documentation.

### Zigbee2MQTT version

1.33.2

### Adapter firmware version

6.10.3.0 build 297

### Adapter

SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20230329160812

### Setup

HA (VM) running on NAS Synology 923+

SuddenGunter, 2024-10-21:
have similar issue, unable to properly use /get endpoint

where should it send a response anyway? on regular zigbee2mqtt/FRIENDLY_NAME topic?

StevenLiew, 2024-10-31:
Has the `length` been removed from the API?

closed 2025-01-07 by github-actions[bot]