Koenkk/zigbee2mqtt #24303: Earu/Avatto TRV06 - running_state always idle.

studio404pl, 2024-10-12:
### What happened?

The Avatto TRV06 properly pairs and returns all of the available configuration. The `Exposes` are properly updating the device state and the currently pulled states are reflecting real data of the device (temperature, heat mode, etc.)

[image]

The problem is that it can't start heat (no flame indicator on the device, `running_state` is always `idle`).

Example state values:

```
{
    "child_lock": "LOCK",
    "current_heating_setpoint": 24,
    "frost_protection": "OFF",
    "linkquality": 255,
    "local_temperature": 20,
    "local_temperature_calibration": 2,
    "running_state": "idle",
    "scale_protection": "ON",
    "system_mode": "heat"
}
```

The problem isn't related to child lock mode (tried turned on and off), it doesn't work in programmable (clock icon on the device) and manual modes. I can only trigger the heat action with the flame indicator by manually operating the device.

Any thermostat actions with a properly set temperature higher than the current one don't change the `running_state` value to `heat` and valve remains closed.

### What did you expect to happen?

The raised `current_heating_setpoint` to value higher than `local_temperature` should trigger the `running_state` value change to `heat`.

### How to reproduce it (minimal and precise)

Pair the device and try to make the valve work (heat) with a native `type: thermostat` Lovelace card or any other thermostat add-on. The temperature set update is respected and the sensor reading of the current temperature is proper, but no action is taken to start the heating process.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.3.1.0 build 0

### Adapter

EZSP v12

### Setup

Home assistant: Core 2024.10.2 Supervisor 2024.10.0 Operating System 13.1 Frontend 20241002.3

Klazzeman, 2024-10-28:
I have the same problem, have you solved it?

studio404pl, 2024-11-03:
> I have the same problem, have you solved it?

unfortunately no, I have only one piece installed and using it manually for now.

MrMEEE, 2024-12-16:
I'm having the same issue here with the _TZE200_p3dbf6qs...

speedycars, 2025-01-25:
My _TZE200_rxq4iti9 start doing the same today.

gazd25, 2025-03-27:
Haven't had chance to test any of this yet, but as far as I'm concerned it's still an issue.

gazd25, 2025-05-29:
Still a problem currently

closed 2025-08-07 by github-actions[bot]

luxandos, 2026-01-01:
Got the same issue here. o3x45p96