Koenkk/zigbee2mqtt #24304: [New device support]: INRR RB 247 T E14 candle

EugenKomi, 2024-10-12:
### Link

https://www.innr.com/en/product/innr-smart-candle-comfort/

### Database entry

{"id":16,"type":"Router","ieeeAddr":"0xa4c1381f143c81a6","nwkAddr":11663,"manufId":4454,"manufName":"innr","powerSource":"Mains (single phase)","modelId":"RB 247 T","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":268,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[10,25],"clusters":{"genBasic":{"attributes":{"modelId":"RB 247 T","manufacturerName":"innr","powerSource":1,"zclVersion":8,"appVersion":22,"stackVersion":54,"hwVersion":1,"dateCode":"20240109","swBuildId":"1.6.4"}},"lightingColorCtrl":{"attributes":{"colorCapabilities":16,"colorTempPhysicalMin":154,"colorTempPhysicalMax":500}},"genLevelCtrl":{"attributes":{"onLevel":255}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":22,"stackVersion":54,"hwVersion":1,"dateCode":"20240109","swBuildId":"1.6.4","zclVersion":8,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1728743581105}

### Zigbee2MQTT version

1.40.2

### Comments

This bulb works but is marked as unsupported

### External definition

```shell
const {light} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['RB 247 T'],
    model: 'RB 247 T',
    vendor: 'innr',
    description: 'Automatically generated definition',
    extend: [light({"colorTemp":{"range":[154,500]}})],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

It looks like it's fully functional, no ОТА update and not added to Dashboard.

EugenKomi, 2024-10-13:
It llooks like RB 247 T is an equivalent of supported RB 249 T.

Koenkk, 2024-10-13:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `RB-247-T.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-12-13 by EugenKomi