Koenkk/zigbee2mqtt #24305: [Wrong device]:

xdmore70, 2024-10-12:
### Link

https://zigbee.blakadder.com/Tuya_WL-19DWZ.html

### Model

_TZ3000_n2egfsli

### Description

Contact sensor

### Vendor

tuya

### Picture (link)

https://zigbee.blakadder.com/Tuya_WL-19DWZ.html

### Database entry

{"id":14,"type":"EndDevice","ieeeAddr":"0xa4c13897622309ae","nwkAddr":45993,"manufId":4742,"manufName":"_TZ3000_n2egfsli","powerSource":"Battery","modelId":"SNZB-04","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1,1280,32],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"ssIasZone":{"attributes":{"61441":1,"iasCieAddr":"0xb43522fffec19b2e","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":14400}},"genPowerCfg":{"attributes":{"batteryVoltage":30,"batteryPercentageRemaining":200}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0xb43522fffec19b2e","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0xb43522fffec19b2e","endpointID":1}],"configuredReportings":[{"cluster":1,"attrId":32,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0},{"cluster":1,"attrId":33,"minRepIntval":3600,"maxRepIntval":7200,"repChange":0}],"meta":{}}},"appVersion":113,"stackVersion":2,"hwVersion":0,"swBuildId":"0122052017","zclVersion":3,"interviewCompleted":true,"meta":{"configured":-14895530},"lastSeen":1728732694569,"checkinInterval":3600}

### Notes

ONENUO Tuya Zigbee Contact Sensor is view like a SONOFF SNZ-04
[image]

[image]

My configuration is: 
Zigbee2MQTT version
1.40.2-dev commit: [5f60187](https://github.com/Koenkk/zigbee2mqtt/commit/5f60187)
Coordinator type
EmberZNet
Coordinator revision
7.4.4 [GA]
Coordinator IEEE Address
0xb43522fffec19b2e
Frontend version
0.7.4
zigbee-herdsman-converters version
20.25.0
zigbee-herdsman version
2.1.3

Koenkk, 2024-10-13:
Is this only a cosmetic problem? (do all the features work?)

xdmore70, 2024-10-13:
Yes Koenkk, all features work.

Koenkk, 2024-10-13:
Fixed!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `WL-19DWZ.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

closed 2024-10-13 by Koenkk

xdmore70, 2024-10-13:
Thanks Koenkk , just done it.

albertinizao, 2024-11-02:
Now arbitrarily these devices are sending me incorrect messages with “contact=true” without the actual state of the device having changed. In this case, at 23:10:03 the device did not change his real status, without any movement, but sent these messages.

[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":102,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":99,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":false,"linkquality":96,"tamper":false,"voltage":3000}'
[2024-11-02 23:10:03] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Silla Bailey', payload '{"battery":100,"battery_low":false,"contact":true,"linkquality":99,"tamper":false,"voltage":3000}'

pushpinderbal, 2024-11-06:
same problem as @albertinizao . I will start a new issue.