Koenkk/zigbee2mqtt #24312: Regression since 1.40.2: Aqara SRTS-A01 generate false positive error messages

jan-di, 2024-10-13:
### What happened?

After updating zsm to v1.40.2, there are a lot of error messages coming from interaction with Aqara/Lumi SRTS-A01 TRV devices.

```
z2m: Publish 'get' 'calibrate' to 'friendlyname-redacted' failed: 'Error: Value: 'calibrate' not found in: [system_mode, preset, window_detection, valve_detection, child_lock, away_preset_temperature, calibrated, sensor, schedule, schedule_settings]'
z2m: Publish 'get' 'external_temperature_input' to 'friendlyname-redacted' failed: 'Error: Value: 'external_temperature_input' not found in: [system_mode, preset, window_detection, valve_detection, child_lock, away_preset_temperature, calibrated, sensor, schedule, schedule_settings]'
```

It seems, that everything is working the same way as before - the TRV devices are controllable and I did not found any regression in the functionality - it justs spams the error log.
Especially providing external_temperature_input still works, despite being mentioned in the error as non existing field.

### What did you expect to happen?

Not having the above mentioned error logs.

### How to reproduce it (minimal and precise)

1. Setup a working installation of z2m v1.40.1
2. Add at least one [Aqara SRTS-A01](https://www.zigbee2mqtt.io/devices/SRTS-A01.html)
3. Ensure everything works, there are no errors
4. Update zsm to v1.40.2

-> Every interaction with the TRV devices will trigger the error messages mentioned above

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.1.1.0 build 273

### Adapter

Skyconnect (EZSP v9)

### Setup

Docker

jan-di, 2024-11-07:
Seems fixed in v1.41.0. Possibly via https://github.com/Koenkk/zigbee-herdsman-converters/pull/8088

nurunet, 2025-01-11:
I still get the "calibrate" error message in Z2M 2.0.0-2 (HA add-on).

> z2m: Publish 'get' 'calibrate' to 'Radiator Thermostat Aqara' failed: 'Error: Value: 'calibrate' not found in: [system_mode, preset, window_detection, valve_detection, child_lock, away_preset_temperature, calibrated, sensor, schedule, schedule_settings]'

nurunet, 2025-01-17:
Scratch that, I was being plain stupid. I realized that the thermostate was not reacting to any input anymore. But I was confused because it still seemed to show like a normal, operational device in Z2M.

I finally tried setting a temperature manually - no luck. And found a battery had leaked inside the device...

After a quick cleanup and inserting a new set of AAs, it seems to work just like it used to. Sorry.

closed 2025-01-17 by jan-di