Koenkk/zigbee2mqtt #24317: Z2M Starts but cannot start pairing procedure (Sonoff ZBDongle-E 7.4.4 | HA OS 13.1 | z2m 1.40.2-1)

thomasvalette, 2024-10-13:
### What happened?

Hello, i have trouble to get Z2M to start properly.

**Here is my setup:**
Sonoff ZBDongle-E (V2 model, EFR32MG21)
HA.OS 13.1 which is running on RPI4 4GB

I flashed my sonoff dongle with this firmware (available on [silabs-firmware-builder/firmware_builds/zbdonglee at main · darkxst/silabs-firmware-builder · GitHub 3](https://github.com/darkxst/silabs-firmware-builder/tree/main/firmware_builds/zbdonglee)):
ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl

Z2M GUI is starting properly and my mqtt broker is receiving messages, but i got these errors (see logs) i cant start pairing procedure. Also my "coordinator revision" is unknown. Something is obviously going wrong but i can't find what.

Searched through the web (and also here) for this exact error but didn't find anything close to what i get.
Am i missing something or is it an Z2M bug? 

### What did you expect to happen?

Z2M nominal start without error logs

### How to reproduce it (minimal and precise)

Start Z2M addon with my setup **HA OS 13.1** on **RPI4**, with **Sonoff ZBDongle-E** dongle flashed with **ncp-uart-hw-v7.4.4.0** firmware (i can provide my configuration if needed)

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

unknown

### Adapter

ember

### Setup

Z2M Addon on HA OS 13.1

### Debug log

```
[14:53:55] INFO: Preparing to start...
[14:53:55] INFO: Socat not enabled
[14:53:58] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-10-04 14:54:04] info: 	z2m: Logging to console, file (filename: log.log)
[2024-10-04 14:54:04] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #unknown)
[2024-10-04 14:54:04] info: 	z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-04 14:54:05] info: 	zh:ember: Using default stack config.
[2024-10-04 14:54:05] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-10-04 14:54:05] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-10-04 14:54:05] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-04 14:54:05] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-10-04 14:54:05] info: 	zh:ember:uart:ash: Serial port opened
[2024-10-04 14:54:05] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-10-04 14:54:06] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-10-04 14:54:06] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-10-04 14:54:06] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-10-04 14:54:06] info: 	zh:ember: Adapter EZSP protocol version (13) lower than Host. Switched.
[2024-10-04 14:54:06] info: 	zh:ember: Adapter version info: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
[2024-10-04 14:54:06] info: 	zh:ember: [STACK STATUS] Network up.
[2024-10-04 14:54:06] info: 	zh:ember: [INIT TC] Adapter network matches config.
[2024-10-04 14:54:06] info: 	zh:ember: [CONCENTRATOR] Started source route discovery. 1248ms until next broadcast.
[2024-10-04 14:54:06] info: 	z2m: zigbee-herdsman started (resumed)
[2024-10-04 14:54:06] info: 	z2m: Coordinator firmware version: '{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":4,"revision":"7.4.4 [GA]","special":0,"type":170},"type":"EmberZNet"}'
[2024-10-04 14:54:06] info: 	z2m: Currently 0 devices are joined.
[2024-10-04 14:54:06] info: 	z2m: Zigbee: disabling joining new devices.
[2024-10-04 14:54:07] info: 	z2m: Connecting to MQTT server at mqtt://core-mosquitto:1883
[2024-10-04 14:54:07] info: 	z2m: Connected to MQTT server
[2024-10-04 14:54:07] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
[2024-10-04 14:54:07] error: 	z2m: Failed to call 'Bridge' 'start' (TypeError: Cannot read properties of undefined (reading 'getDevice')
    at Bridge.publishInfo (/app/lib/extension/bridge.ts:746:69)
    at Bridge.start (/app/lib/extension/bridge.ts:174:20)
    at Controller.callExtensions (/app/lib/controller.ts:399:17)
    at Controller.start (/app/lib/controller.ts:218:9)
    at start (/app/index.js:154:5))
[2024-10-04 14:54:07] info: 	z2m: Started frontend on port 8099
[2024-10-04 14:54:07] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"unknown","coordinator":{"meta":{"build":0,"ezsp":13,"major":7,"minor":4,"patch":4,"revision":"7.4.4 [GA]","special":0,"type":170},"type":"EmberZNet"},"log_level":"info","network":{"channel":11,"extendedPanID":13368757625167469000,"panID":11483},"permit_join":false,"version":"1.40.2"}'
[2024-10-04 14:54:07] error: 	z2m: Failed to call 'HomeAssistant' 'start' (TypeError: Cannot read properties of undefined (reading 'deviceIeeeAddress')
    at HomeAssistant.getBridgeEntity (/app/lib/extension/homeassistant.ts:2016:78)
    at HomeAssistant.start (/app/lib/extension/homeassistant.ts:476:28)
    at Controller.callExtensions (/app/lib/controller.ts:399:17)
    at Controller.start (/app/lib/controller.ts:218:9)
    at start (/app/index.js:154:5))
[2024-10-04 14:54:07] info: 	z2m: Zigbee2MQTT started!
```

thomasvalette, 2024-10-14:
Sure, it only contains this line: 

`{"id":1,"type":"Coordinator","ieeeAddr":"0xe8e07efffe7c5dcc","nwkAddr":0,"manufId":4169,"epList":[],"endpoints":{},"interviewCompleted":true,"meta":{}}`

chenyixi1997, 2024-10-16:
Hello, I also have a similar problem, although the adapter is different, but it is the same log:
**_error: 	z2m: Failed to call 'Bridge' 'start'_**
 I feel that the bridge startup failed.Can anyone help us?

Matoxius, 2024-10-16:
Hello, I had the same issue, after halve a day of troubleshooting I had a solution that might help.

I uninstalled the zigbee2mqtt plugin and went into the plugin folder via the `ssh/terminal` plugin. 
There I removed all files with `rm -rf *`( __make sure to be in the right folder, or you might destroy your installation__ ) and then reinstalled the mqtt plugin. 

For me it's working now, could be worth a try for you guys.

chenyixi1997, 2024-10-16:
Thank you, I also tried to reinstall zigbee2mqtt on my Raspberry PI today, and my problem can be solved after the re-installation.

chenyixi1997, 2024-10-17:
I found that whenever I re-installed zigbee2mqtt, my first npm start started normally, but after I closed zigbee2mqtt by ctrl+c, I started npm start again, and there was an error：
**_error:     z2m: Failed to call 'Bridge' 'start'_**

thomasvalette, 2024-10-17:
Removing the content of this folder somehow fixed my issue too, thanks a lot @Matoxius for this easy fix.
I tried rebooting to see if i get the same error as @chenyixi1997, but everything seems to work ok even after a full restart. 

> Hello, I had the same issue, after halve a day of troubleshooting I had a solution that might help.
> 
> I uninstalled the zigbee2mqtt plugin and went into the plugin folder with `cd /config/zigbee2mqtt/` via the `ssh/terminal` plugin. There I removed all files with `rm -rf *`( **make sure to be in the right folder, or you might destroy your installation** ) and then reinstalled the mqtt plugin.
> 
> For me it's working now, could be worth a try for you guys.

closed 2024-10-17 by thomasvalette

clrmsc, 2024-10-19:
Hello friends! I have an error when starting, could you help me?

configuration.yaml:
`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost:1883
serial:
  port: /dev/ttyACM0
  baudrate: 115200
  adapter: ember
  rtscts: false
frontend:
  port: 8081
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false`

Debug log:
`[2024-10-19 10:33:40] info: 	z2m: Logging to console, file (filename: log.log)
[2024-10-19 10:33:40] info: 	z2m: Starting Zigbee2MQTT version 1.40.2 (commit #e06848d)
[2024-10-19 10:33:40] info: 	z2m: Starting zigbee-herdsman (2.1.3)
[2024-10-19 10:33:41] info: 	zh:ember: Using default stack config.
[2024-10-19 10:33:41] info: 	zh:ember: ======== Ember Adapter Starting ========
[2024-10-19 10:33:41] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-10-19 10:33:41] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 10:33:41] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-10-19 10:33:41] info: 	zh:ember:uart:ash: Serial port opened
[2024-10-19 10:33:41] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 10:33:43] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 10:33:43] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 10:33:46] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 10:33:46] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 10:33:48] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 10:33:48] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 10:33:51] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-19 10:33:51] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-10-19 10:33:53] error: 	z2m: Error while starting zigbee-herdsman
[2024-10-19 10:33:53] error: 	z2m: Failed to start zigbee
[2024-10-19 10:33:53] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
[2024-10-19 10:33:53] error: 	z2m: Exiting...
[2024-10-19 10:33:53] error: 	z2m: Error: Failed to start EZSP layer with status=HOST_FATAL_ERROR.
    at EmberAdapter.initEzsp (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:684:19)
    at EmberAdapter.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/adapter/ember/adapter/emberAdapter.ts:1583:24)
    at Controller.start (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
    at Zigbee.start (/opt/zigbee2mqtt/lib/zigbee.ts:69:27)
    at Controller.start (/opt/zigbee2mqtt/lib/controller.ts:161:27)
    at start (/opt/zigbee2mqtt/index.js:154:5)
`

artemkovalyov, 2024-10-26:
@clrmsc , if you're on sonoff dongle plus (e), you have to use `adapter: ezsp` instead of ember. Otherwise your questions is off-topic for this issue.

artemkovalyov, 2024-10-26:
@Matoxius, your solution fixes the issue and is a "know-how", do you think the maintainers will accept smth like this onto a known issues or troubleshooting guide?

Heljick, 2024-11-10:
> Hello, I had the same issue, after halve a day of troubleshooting I had a solution that might help.
>
> I uninstalled the zigbee2mqtt plugin and went into the plugin folder with `cd [...]

HI, I had similar issue after updating SonOff Dongle-P to 20240710 Firmware but didn't wanna to loose my 127 devices and 16 groups. I did a backup of all the files in /config/zigbee2mqtt (coordinator_backup.json, configuration.yaml and database.db) run Z2M creating new files, took everything in configuration.yaml starting after "legacy: false" and deleted to database.db and copied the backup of the old one and I get everything backup showing up in Z2M with "permit join" working again and dongle versionning which was set as "Unknow" before

dampfkolben, 2025-02-16:
> Hello, I had the same issue, after halve a day of troubleshooting I had a solution that might help.
>
> I uninstalled the zigbee2mqtt plugin and went into the plugin folder with `cd [...]

Thx it seems to help me fix this issue ;-)