Koenkk/zigbee2mqtt #24328: HK-DIM-A - Power_On_Behavior unsupported

Telk0, 2024-10-14:
### What happened?

Trying to get Power-on Behavior gives this error. the error randomly spams aswell.

[image]
[image]

### What did you expect to happen?

i would set the power-on behavior.

### How to reproduce it (minimal and precise)

Try change the setting.

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

20220219

### Adapter

Sonoff Dongle-P - zStack3x0

### Setup

HA OS on Intel NUC x64 as add-on

### Debug log

[image]

Telk0, 2024-12-14:
@Koenkk

Koenkk, 2024-12-15:
The device does not support this (given the error), I removed it!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now. Note that the dev branch currently contains **breaking changes**, see https://github.com/Koenkk/zigbee2mqtt/discussions/24198

closed 2024-12-15 by Koenkk