Koenkk/zigbee2mqtt #24333: [Feature request]: Allow to set switch type for NodOn devices

acasajus, 2024-10-14:
### Is your feature request related to a problem? Please describe

Currently NodOn devices try to auto-detect if the switch is a rocker or normal switch. Once the switch has decided on one mode it cannot be changed via zigbee2mqtt. You need to turn off the power and turn on to make the switch to get into auto-detect mode. If you have a normal switch and switch too fast the device thinks it has a rocker switch.

[image]

### Describe the solution you'd like

To be able to set the switch mode (rocker or normal switch) from the configuration skipping the auto-detection mode.

### Describe alternatives you've considered

Changing switches. Having to power-cycle all the switches if somebody clicks any switch twice the first time is very anonying.

### Additional context

Since the device takes a bit to actually cycle the power to the light once the switch has been pressed, it has happened many times at home that since the lights don't switch immediately they click the switch again and then the relay thinks its in rocker mode

Th-Ar, 2024-10-15:
Wich model of Nodon device ?

AlexisPolegato, 2024-11-07:
@acasajus : It is now possible to change it on the SIN-4-1-20 : 

[image]

It's coming on SIN-4-RS-20 as well.

You need Firmware Version on V3.4.0 or superior on the Device.

Also, from this version, the Switch Type is saved and restored after Power Cycle.

acasajus, 2024-11-07:
That's great! Thanks!

acasajus, 2024-12-06:
Hello, do you know when it will land for SIN-4-RS-20? Thanks a lot again for that feature 👍

acasajus, 2025-02-05:
Sorry @AlexisPolegato  for the ping. I can do the change in the repo if the config is the same for SIN-4-RS-20 as the SIN-4-1-20.

w-marco, 2025-04-05:
Hi, I had the same problem and figured out the switch type option is inside the closuresWindowCovering Cluster for the SIN-4-RS-20. I've created a PR here to update the feature to support this model:

https://github.com/Koenkk/zigbee-herdsman-converters/pull/9126

Once this has been merged, it should work just fine.

w-marco, 2025-04-05:
Hello @acasajus, I have the same issue and found out through experimenting, that for the Nodon SIN-4-RS-20 the property is not in the genOnOff but in the closuresWindowCovering cluster. These are modified snippets of https://github.com/sjorge/zigbee-herdsman-converters/blob/214115fcb627ba6fc073df794bdf78a84955fd8b/src/devices/nodon.ts working to change the type on SIN-4-RS-20:

```
    switchType: (args) => m.enumLookup({                                                                         
        name: "switch_type",                                                                     
        lookup: { bistable: 0x00, monostable: 0x01, auto_detect: 0x02 },                       
        cluster: "closuresWindowCovering",                                                                       
        attribute: { ID: 0x1001, type: zigbee_herdsman_1.Zcl.DataType.ENUM8 },                   
        description: "Select the switch type wire to the device. " + "Available from version > V3.4.0",          
        zigbeeCommandOptions: { manufacturerCode: zigbee_herdsman_1.Zcl.ManufacturerCode.NODON },                
        ...args,                                                                                       
    }),     
```

```
    {                                                                                                                                    
        zigbeeModel: ["SIN-4-RS-20", "SIN-4-UNK"],                                                                     
        model: "SIN-4-RS-20",                                                                                                            
        vendor: "NodOn",                                                                                                                 
        description: "Roller shutter relay switch",                                                                                      
        extend: [                                                                                                                        
            m.windowCovering({ controls: ["tilt", "lift"], coverMode: true }),                                         
            nodonModernExtend.calibrationVerticalRunTimeUp(),                                                    
            nodonModernExtend.calibrationVerticalRunTimeDowm(),                                                  
            nodonModernExtend.calibrationRotationRunTimeUp(),                                                          
            nodonModernExtend.calibrationRotationRunTimeDown(),                                               
            nodonModernExtend.switchType(),                                                                            
        ],                                                                                                       
        ota: true,                                                                                            
    },  
```

So there would have to be either two switchType functions or some sort of if-case in the cluster depending on the model.  I tried adding it myself but the codebase has since changed a bit and it kept breaking. Sadly I didn't have the time to cleanly implement it as a PR.

acasajus, 2025-04-07:
Hey, Thanks a lot!!! Do you know how the switch manages tilt? I only have window blinds that go up and down and that works. But I always wonder how it manages two different functionalities with just one pair of cables.

acasajus, 2025-04-07:
BTW, i'm trying to find out some docs on what each cluster is or means. Can you direct to some docs that explain it?

AlexisPolegato, 2025-04-07:
> Hey, Thanks a lot!!! Do you know how the switch manages tilt? I only have window blinds that go up and down and that works. But I always wonder how it manages two different functionalities with just one pair of cables.

Hello,

It is for venetian blinds, it works with the UP/ DOWN wires, you don't need additional wires for the tilt.

> BTW, i'm trying to find out some docs on what each cluster is or means. Can you direct to some docs that explain it?

You can look for "Zigbee Cluster Library" online, you will find all the info there.

=> https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf

acasajus, 2025-04-08:
I've got two controls for the relay, one for the blinds and one for the blinds that go up and down. How does the switch know when to use each? or is it just that both do the same thing?

Liozon, 2025-05-13:
Hello everyone!
I'm just commenting on this issue/feature request to say that I'm also interested in this new feature.

One of my NodOn SIN-4-RS-20 has suddenly charged itself my switch type, maybe because I'm using Home Assistant more than my physical switches around the house.

However, this is not the first time that this happens, and I had to completely remove the device from Z2M, open my entire switches fixture just to physically reset the NodOn and repair it again, just for it to correctly recognise my switch type at the first boot (I'm with firmware 3.5.0 from the beginning)

To sum up: this feature request has my vote too 👋
Thank you in advance and if can I be of any help, do not hesitate!

Liozon, 2025-05-16:
Hello @acasajus ,
I updated Z2M to version 2.3.0, and I now have the switch type for all my SIN-4-RS-20 too !
I was able to correct my problem I had on one of my NodOn, who changed itself the switch type for some reason.
My SIN-4-RS-20 is on firmware 3.5.0-1.3.0 and this setting perfectly works

[image]

AlexisPolegato, 2025-05-17:
> Thanks for the answer. But i'm still confused. I've got two controls for the relay, one for the blinds and one for the blinds that go up and down. How does the switch know when to use each? or is it just that both do the same thing? and in that case, why have two different controls?

I don't get it : you have two types of switches connected to the Relay Switch ?

On the Relay Switch you have 2 inputs : 1 for Switch Up and 1 for Switch Down. Of course you need to wire your Switch Up to the input Up and your Switch Down to the input Down, then the Device knows which one is which one.

AlexisPolegato, 2025-05-17:
> Hello [@acasajus](https://github.com/acasajus) , I updated Z2M to version 2.3.0, and I now have the switch type for all my SIN-4-RS-20 too ! I was able to fix my problem I had on one of my NodOn, who changed itself the switch type for some reason. My SIN-4-RS-20 is on firmware 3.5.0-1.3.0 and this setting perfectly works
> 
> [image]

Great !

Roxxor91, 2025-06-05:
Thanks! Couldnt get the switch type to work in one room, now I could select it!
Sorry for the offtopic, but you are the only other NodOn users I can find.
2 Questions:
- Is it possible to disable Bluetooth? My local Bluetooth surrounding is spammed by my 10 NodOn devices.
- I have one room where all of a sudden (worked correctly before) this happens: When closing the blind, it switches to the other direction, goes up a bit and then stops there. So automatically it never closes but opens again. I have to "interrupt" closing, so it stays closed. Anybody of you got that before?