Koenkk/zigbee2mqtt #24334: 2024-10-14 19:16:22zh:controller:touchlink: Set InterPAN channel to '16' Warning 2024-10-14 19:16:23zh:controller:touchlink: Scan request failed or was not answered: 'Error: {"apsFrame":{"profileId":49246,"clusterId":4096,"sourceEndpoint":1,"destinationEndpoint":1,"options":0,"groupId":65535,"sequence":0},"zclSequence":0,"commandIdentifier":1} timed out after 500ms'

cpttrooper, 2024-10-14:
### What happened?

I have installed a new Zigbee coordinator when I want to connect my Philips Hue play bars via touchline I got this error

> 2024-10-14 19:16:22zh:controller:touchlink: Set InterPAN channel to '16'
> Warning 2024-10-14 19:16:23zh:controller:touchlink: Scan request failed or was not answered: 'Error: {"apsFrame":{"profileId":49246,"clusterId":4096,"sourceEndpoint":1,"destinationEndpoint":1,"options":0,"groupId":65535,"sequence":0},"zclSequence":0,"commandIdentifier":1} timed out after 500ms'

I have searched online and tried everything. If I want to add a devices in normal join mode all works

The driver I use is the new Ember

### What did you expect to happen?

Touchlink connecting my Philips Hue Play bar

### How to reproduce it (minimal and precise)

Click on touchlink

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

20240510

### Adapter

SLZB-06M

### Setup

Home Assistant Add-on

closed 2024-12-22 by github-actions[bot]

wunni1, 2024-12-29:
I have the same problem, with the same configuration like @cpttrooper .

Any ideas?

TommySharpNZ, 2025-02-11:
I also have the exact same problem :-(

Arreslee, 2025-05-06:
Same problem here, SLZB-06M in combination with  touchlink