Koenkk/zigbee2mqtt #24336: [New device support]: Legrand 067696 module to open garage doors and gates

dsigmoun, 2024-10-14:
### Link

https://www.legrand.fr/pro/catalogue/module-pour-porte-de-garage-et-portail-pour-installation-connectee-with-netatmo

### Database entry

{     "linkquality": 255,     "position": 100,     "state": "OPEN",     "device": {         "applicationVersion": 0,         "dateCode": " \u000020240117\u0000\u0000\u0000\u0000\u0000",         "friendlyName": "Contact sec portail Legrand Netatmo",         "hardwareVersion": 1,         "ieeeAddr": "0x0004740000280f1e",         "manufacturerID": 4129,         "manufacturerName": " Legrand",         "model": " NLJ - Garage door\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000",         "networkAddress": 53309,         "powerSource": "Mains (single phase)",         "softwareBuildID": "0007",         "stackVersion": 67,         "type": "Router",         "zclVersion": 2     } }

### Zigbee2MQTT version

1.40.2

### Comments

Hello,

I use the Legrand 067696 module to open my garage doors and gates. In Z2M, the module is indicated unmanaged command.
The Open, close, stop commands work correctly.
[image]

When I update it, I get error messages.

```
info 2024-10-14 21:53:11z2m: Configuring 'Contact sec portail Legrand Netatmo'
info 2024-10-14 21:53:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/Contact sec portail Legrand Netatmo', payload '{"device":{"applicationVersion":0,"dateCode":" \u000020240117\u0000\u0000\u0000\u0000\u0000","friendlyName":"Contact sec portail Legrand Netatmo","hardwareVersion":1,"ieeeAddr":"0x0004740000280f1e","manufacturerID":4129,"manufacturerName":" Legrand","model":" NLJ - Garage door\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000\u0000","networkAddress":53309,"powerSource":"Mains (single phase)","softwareBuildID":"0007","stackVersion":67,"type":"Router","zclVersion":2},"linkquality":255,"position":0,"state":"CLOSE"}'
error 2024-10-14 21:53:11z2m: Failed to configure 'Contact sec portail Legrand Netatmo', attempt 4 (Error: ZCL command 0x0004740000280f1e/1 closuresWindowCovering.configReport([{"minimumReportInterval":1,"maximumReportInterval":65000,"reportableChange":1,"attribute":"currentPositionTiltPercentage"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:112:13) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:157:17) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:205:21) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:130:13) at Configure.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/configure.ts:57:21) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:206:17))
info 2024-10-14 21:53:11z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/configure', payload '{"data":{"id":"Contact sec portail Legrand Netatmo"},"error":"Failed to configure (ZCL command 0x0004740000280f1e/1 closuresWindowCovering.configReport([{\"minimumReportInterval\":1,\"maximumReportInterval\":65000,\"reportableChange\":1,\"attribute\":\"currentPositionTiltPercentage\"}], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE'))","status":"error","transaction":"ebjku-2"}'
```

### External definition

```shell
const {windowCovering, binary, commandsWindowCovering} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: [' NLJ - Garage door
```

### What does/doesn't work with the external definition?

[image]

The Open, close, stop commands work correctly.

I don't understand what the "tilt" command is for. It generates errors.
[image]

```
error 2024-10-14 21:56:44z2m: Publish 'get' 'tilt' to 'Contact sec portail Legrand Netatmo' failed: 'Error: ZCL command 0x0004740000280f1e/1 closuresWindowCovering.read(["currentPositionTiltPercentage"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'
info 2024-10-14 21:56:44z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'get' 'tilt' to 'Contact sec portail Legrand Netatmo' failed: 'Error: ZCL command 0x0004740000280f1e/1 closuresWindowCovering.read([\"currentPositionTiltPercentage\"], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"reservedBits\":0,\"writeUndiv\":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE')'","meta":{"friendly_name":"Contact sec portail Legrand Netatmo"},"type":"zigbee_publish_error"}'
```

Same thing for "Binary input 1"
[image]

`error 2024-10-14 22:02:14z2m: No converter available for 'binary_input_1_1' ("")`

Thanks for your help!

BBilly156, 2024-10-25:
Hello
I have the same device here, and also tried the procedure, but i don't have enough knowledge to complete it.
I saw you messages on jeedom forum, is it working fine now ?
thanks a lot

dsigmoun, 2024-10-25:
I still get error messages, but otherwise the commands work fine.

closed 2025-01-01 by github-actions[bot]