Koenkk/zigbee2mqtt #24340: TS0505B bulb power status unknown

gfrantsau, 2024-10-15:
### What happened?

The TS0505B bulb power status is shown as unknown (a question mark is displayed next to Power).

[image]

### What did you expect to happen?

Power status displayed as Mains.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

7.4.4 [GA]

### Adapter

Sonoff Dongle E

### Setup

Add-on on HA OS

Koenkk, 2024-10-16:
Fixed! After updating re-configure the light by pressing the yellow refresh icon

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

closed 2024-10-16 by Koenkk

gfrantsau, 2024-11-01:
@Koenkk Doesn't work. I've tried reconfiguring then restarting the add-on, and even removing the device altogether and re-adding it. Still displays a question mark next to Power.

KG3RK3N, 2024-11-01:
Same behavior as @gfrantsau

Koenkk, 2024-11-02:
- Could you provide the `data/databse.db` value of this device?
- After the re-configure did you already restart z2m?

gfrantsau, 2024-11-03:
@Koenkk yes, restarted z2m after reconfiguring. Then tried removing and re-adding the device. No change.

{"id":11,"type":"Router","ieeeAddr":"0xa4c138c4e5a77fbd","nwkAddr":58550,"manufId":4107,"manufName":"_TZ3210_mja6r5ix","powerSource":"Unknown","modelId":"TS0505B","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":269,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"hwVersion":0}},"lightingColorCtrl":{"attributes":{"57344":1000,"57345":4294984794153,"61453":1,"colorCapabilities":29,"tuyaRgbMode":0}},"genLevelCtrl":{"attributes":{"61440":1000}},"genOnOff":{"attributes":{"61440":0,"onOff":0}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":112,"stackVersion":2,"hwVersion":0,"swBuildId":"z.1.0","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1730608625677}

Koenkk, 2024-11-03:
Found the issue, pushed a fix.

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.