Koenkk/zigbee2mqtt #24346: [New device support]: Heiman HS2SA-1 Photoelectrit Smoke Alarm

Vaarinvaliliha, 2024-10-15:
### Link

https://www.heimantech.com/product/?type=detail&id=139

### Database entry

},             "0xa4c138f065fbc22d": {                 "friendly_name": "0xa4c138f065fbc22d"

### Zigbee2MQTT version

1.40.2

### Comments

Hi, I'd like to have this supported, I have tried to make converter but I'm also a dummy.

### External definition

```shell
const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601',
    vendor: '_TZE200_ai4rqhky',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

z2m does not support

Vaarinvaliliha, 2024-10-15:
[image]

Oggymator, 2024-10-21:
be careful, seems you have the one with TUYA Chip. Heiman has some products with TUYA and some with his own ZigBee Chip

murimaa, 2024-11-17:
I got this alarm and has success copying the code from Moes ZSS-HM-SSD01, which looks similar to this.

Smoke attribute was triggered when actual smoke was detected, Silence was set to true correctly when I pushed the button to silence, and also Self test property responded to holding the button triggering the alarm test sound.

Not sure how to go about doing a PR though, never done one. The Moes converter I copied is in `moes.ts`, but I think it should be moved to `tuya.ts` and use the whitelabel property to define both devices (assuming they are the same)?

Anyways, here is the converter:

```
const tuya = require('zigbee-herdsman-converters/lib/tuya');
const exposes = require('zigbee-herdsman-converters/lib/exposes');

const e = exposes.presets;
const ea = exposes.access;

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_ai4rqhky',
        },
    ],
    model: 'HS2SA-1',
    vendor: 'HEIMAN',
    description: 'Photoelectric Smoke Alarm',
    fromZigbee: [tuya.fz.datapoints],
    toZigbee: [tuya.tz.datapoints],
    onEvent: tuya.onEventSetTime,
    configure: tuya.configureMagicPacket,
    exposes: [
        e.smoke(),
        e.battery(),
        tuya.exposes.batteryState(),
        e.binary('silence', ea.STATE_SET, 'ON', 'OFF'),
        e.enum('self_test', ea.STATE, ['checking', 'check_success', 'check_failure']),
    ],
    extend: [],
    meta: {
        tuyaDatapoints: [
                [1, 'smoke', tuya.valueConverter.trueFalse0],
                [9, 'self_test', tuya.valueConverterBasic.lookup({checking: 0, check_success: 1, check_failure: 2})],
                [14, 'battery_state', tuya.valueConverter.batteryState],
                [15, 'battery', tuya.valueConverter.raw],
                [16, 'silence', tuya.valueConverter.onOff],
        ],
    },
};

module.exports = definition;
```

[image]

murimaa, 2024-11-17:
Here are some log outputs when triggering the alarm (with smoke):

```
zigbee2mqtt           | [2024-11-17 10:49:27] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":255,"self_test":"check_success","silence":"OFF","smoke":true}'
zigbee2mqtt           | [2024-11-17 10:49:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":255,"self_test":"check_success","silence":"OFF","smoke":true}'
zigbee2mqtt           | [2024-11-17 10:49:30] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":255,"self_test":"check_success","silence":"OFF","smoke":true}'
zigbee2mqtt           | [2024-11-17 10:49:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":255,"self_test":"check_success","silence":"ON","smoke":true}'
zigbee2mqtt           | [2024-11-17 10:49:32] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":255,"self_test":"check_success","silence":"ON","smoke":true}'
zigbee2mqtt           | [2024-11-17 10:49:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":255,"self_test":"check_success","silence":"ON","smoke":false}'
zigbee2mqtt           | [2024-11-17 10:49:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":255,"self_test":"check_success","silence":"ON","smoke":false}'
zigbee2mqtt           | [2024-11-17 10:49:33] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":255,"self_test":"check_success","silence":"ON","smoke":false}'
```

`silence` was reset to OFF after 10 minutes:
```
zigbee2mqtt           | [2024-11-17 10:59:31] info: 	z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0xa4c138a7ae07202d', payload '{"battery":100,"battery_state":"high","linkquality":167,"self_test":"check_success","silence":"OFF","smoke":false}'
```

`self_test` seems to never reset, it has been `check_success` since yesterday.

closed 2025-01-24 by github-actions[bot]

SmoothMC, 2025-03-21:
Hello there, why is this device marked as not planned?