Koenkk/zigbee2mqtt #24365: Error: ZCL command genOnOff.on

MopNGlow88, 2024-10-16:
### What happened?

Devices are lagging and not properly communicating with the coordinator and Zigbee2MQTT. Similar to closed issue 22189. In relation to the SLZB-06, I have flashed the Zigbee firmware from 20230507 forward and would get the errors within an hour. Kept core firmware at v2.5.6. Mains devices in particular randomly drop and come back after a short time. This did not happen previously. Tried uninstalling and reinstalling Zigbee2MQTT as well to no avail.

### What did you expect to happen?

Devices to communicate and work in a timely manner.

### How to reproduce it (minimal and precise)

The error does not seem to be consistent, other than as devices send information to or from the mains or endpoint devices I get the error. There are also delays on the communication between the devices up to a minute. 

### Zigbee2MQTT version

1.40.2-1

### Adapter firmware version

20240710

### Adapter

SLZB-06

### Setup

HAOS 2024.10.2 Sup 2024.10.2 OS 13.2 FE 20241002.3

### Debug log

[2024-10-16 11:41:22] error: 	z2m: Failed to read state of 'Deep Freezer Plug' after reconnect (ZCL command 0x70b3d52b6001ba61/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 14517 - 1 - 225 - 6 - 1 after 10000ms))
[2024-10-16 11:52:15] error: 	z2m: Failed to read state of 'Living Room Gaming Device Plug' after reconnect (ZCL command 0x70b3d52b6001b02d/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 24085 - 1 - 149 - 6 - 1 after 10000ms))
[2024-10-16 12:14:38] error: 	z2m: Failed to read state of 'Air Fryer Plug' after reconnect (ZCL command 0x70b3d52b6001b7f3/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 47174 - 1 - 28 - 6 - 1 after 10000ms))
[2024-10-16 13:24:14] error: 	z2m: Failed to read state of 'Outdoors Front Plug' after reconnect (ZCL command 0x70b3d52b6001b88e/1 genOnOff.read(["onOff"], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 46224 - 1 - 197 - 6 - 1 after 10000ms))
[2024-10-16 14:17:38] error: 	zh:controller:device: Read response to 0xa4c1380a1cbc94c9 failed (ZCL command 0xa4c1380a1cbc94c9/1 genTime.readRsp({"localTime":782403449}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":135,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
[2024-10-16 16:05:37] error: 	zh:controller:device: Read response to 0x94deb8fffea3f828 failed (ZCL command 0x94deb8fffea3f828/1 genTime.readRsp({"localTime":782409929}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":72,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
[2024-10-16 16:05:46] error: 	zh:controller:device: Read response to 0x94deb8fffea3f828 failed (ZCL command 0x94deb8fffea3f828/1 genTime.readRsp({"localTime":782409931}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":72,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
[2024-10-16 16:33:30] error: 	zh:controller:device: Read response to 0xa4c138f9cdd884cf failed (ZCL command 0xa4c138f9cdd884cf/1 genTime.readRsp({"localTime":782411602}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":182,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
[2024-10-16 16:33:41] error: 	zh:controller:device: Read response to 0xa4c138f9cdd884cf failed (ZCL command 0xa4c138f9cdd884cf/1 genTime.readRsp({"localTime":782411604}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":1,"srcEndpoint":1,"reservedBits":0,"transactionSequenceNumber":182,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225)))
[2024-10-16 16:59:59] error: 	z2m: Publish 'set' 'state' to 'Living Room Night Light' failed: 'Error: ZCL command 0xb40ecfd3a1d00000/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
[2024-10-16 17:12:38] error: 	z2m: Publish 'set' 'state' to 'Living Room Reading Lamp' failed: 'Error: ZCL command 0x00124b0025203a94/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
[2024-10-16 17:20:09] error: 	z2m: Publish 'set' 'state' to 'Living Room Reading Lamp' failed: 'Error: ZCL command 0x00124b0025203a94/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'
[2024-10-16 17:20:18] error: 	z2m: Publish 'set' 'color' to 'Kitchen Night Light' failed: 'Error: ZCL command 0xb40ecfd3ade80000/1 lightingColorCtrl.moveToColor({"transtime":0,"colorx":8978,"colory":2884}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'

Koenkk, 2024-10-17:
> MAC channel access failure' (225)

This means there is wireless interference, see https://www.zigbee2mqtt.io/guide/faq/#common-error-codes

MopNGlow88, 2024-10-17:
Thanks! Is the ZCL also related to that as well?

Yesterday I switched my 2.4ghz to only do channel 1 on my router. I have about 12 devices on it. For the coordinator I'm using channel 20 with the transmit power has been at 20. I have 25 mains powered routers. 

Should I be reducing the transmit power? The article seems to imply there could be complications at higher transmit powers. 

I moved the coordinator up higher in the room for 12 hours and then placed it at the original location. It seems fine now. I've only gotten one ZCL error/ MAC error.

Not really sure what caused the issue as it was working fine before then I just started having major issues yesterday.

MopNGlow88, 2024-10-17:
One additional thing I did to test things was to do a dual setup with ZHA and Zigbee2mqtt. Didn't have to use any of the entities through ZHA but everything did stabilize. No idea why becuase I used Zigbee2mqtt entities for automations and to test the devices. To see if that was actually helping I removed ZHA and reset everything. This caused the entire Zigbee network to slow with more ZCL errors.

[2024-10-17 19:25:43] error: 	z2m: Publish 'set' 'state' to 'Living Room Reading Lamp' failed: 'Error: ZCL command 0x00124b0025203a94/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 63500 - 1 - 90 - 6 - 11 after 10000ms)'

MopNGlow88, 2024-10-31:
As requested I moved the coordinator to another part of the room and that fixed it. The device was there for months, not sure what changed but moving it worked. Thanks!

closed 2024-10-31 by MopNGlow88