Koenkk/zigbee2mqtt #24375: Error: ZCL command genOnOff.on (continuation of #22189)

majkers, 2024-10-17:
### What happened?

This is a continuation of IMHO mistakenly closed issue #22189. From time to time. It is hard to tell exactly when but one of my coordinators ie. [Xiaomi ZNCZ02LM](https://www.zigbee2mqtt.io/devices/ZNCZ02LM.html#xiaomi-zncz02lm) fails with an error:

`[2024-10-17 12:10:31] error: 	z2m: Publish 'set' 'state' to 'gniazdko_xiaomi' failed: 'Error: ZCL command 0x00158d0002347282/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Timeout - 18793 - 1 - 88 - 6 - 11 after 10000ms)'`

And since it is a coordinator, all of my devices connected to it fall out off the mesh network. As it is continuation of https://github.com/Koenkk/zigbee2mqtt/issues/22189 you can see that it is a problem of many coordinators with many hardwares.

I tested it with Conbee II then moved to Sonoff ZBDongle-P with firmware version 20240710 and then moved back to 20230507 but it doesn't help.

I've also noticed that trying to set state via frontend to my socket sometimes do change it as I hear the click sound but there is no confirmation in z2m state and then I see an error mentioned above.

### What did you expect to happen?

I expect those errors not to happen

### How to reproduce it (minimal and precise)

There is no real reporducer. Maybe setting on/off state quickly.

### Zigbee2MQTT version

1.40.2

### Adapter firmware version

tested with 20230507 and 20240710 

### Adapter

Conbee II and Sonoff ZBDongle-P

### Setup

Add on on Home Assistant OS

majkers, 2024-10-17:
I know from https://github.com/Koenkk/zigbee2mqtt/issues/22961
that this:
`Timeout - 18793 - 1 - 88 - 6 - 11 after 10000ms`
can mean a lot. But. My netowrk is many years old and I never had any issues with it. It all startek more or less 6 months ago. I even migrated to Sonoff from Conbee to stabilize it since it was not reommended adapter.

grokkerrj, 2024-10-26:
error 2024-10-26 13:25:33z2m: Publish 'set' 'state' to 'Ikea Trådfri LED 1055 lm Taklampe bulb 5' failed: 'Error: ZCL command 0x385b44fffe0433b6/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Delivery failed for '32706'.)'
[image]
[image]

driek12, 2024-11-01:
Same issue here, config:
Zigbee2MQTT versie
[1.41.0](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.41.0) commit: [unknown](https://github.com/Koenkk/zigbee2mqtt/commit/unknown)
Coördinator type
zStack3x0
Coordinator revisie
20211217
Coordinator IEEE adres
0x00124b0024c95416
Versie frontend
0.7.4
zigbee_herdsman_converters_version
20.38.0
zigbee_herdsman_version
2.1.7

majkers, 2024-11-07:
Guys. Please try this that seems to help a bit with my socket/router. Move away all the wires that are close to it. Plugged in to that socket or other. It seems like they are somewhat interfere with zigbee signal.

denostr, 2024-11-07:
> Guys. Please try this that seems to help a bit with my socket/router. Move away all the wires that are close to it. Plugged in to that socket or other. It seems like they are somewhat interfere with zigbee signal.

This is strange, but I moved my Sonoff coordinator 1 meter away, rebooted Home Assistant, and the error disappeared. On my shelf, besides the Raspberry Pi (with HAOS) and the coordinator, I have another Raspberry Pi, a mini PC, and a switch. Maybe they somehow interfere with the coordinator’s work.

rsirigu, 2024-11-10:
I had the same problem and I solved it moving the zigbee stick away from the shelf where my mini PC, switch and NAS are, with a USB extension cable. Wifi routers and especially USB 3.0 ports can cause interferences with 2.4 GHz radio devices. You can get informations about this on [this page](https://community.home-assistant.io/t/zigbee-network-optimization-a-how-to-guide-for-avoiding-radio-frequency-interference-adding-zigbee-router-devices-repeaters-extenders-to-get-a-stable-zigbee-network-mesh-with-best-possible-range-and-coverage-by-fully-utilizing-zigbee-mesh-networking/515752).

Inshakoor, 2024-11-14:
I just wanted to chime in that although I did have my coordinator extended via USB, it was right next to my switch and router.  I moved it 2 meters away just on the floor and I devices are responding again.

peterchs, 2024-11-20:
For my device (Sonoff ZBMINIL2) it automatically seemed to have a binding to genOnOff seen on the bind tab of the devices page.  I removed this by clicking the 'unbind' button.  If your device is online this should work.  After this I no longer receive the errors, and the device functions as I expect it to.  

I'm not sure exactly why it has this binding by default, but z2m receives updates to the state and can be controlled fine.

makanimike, 2024-12-24:
geez. After months of troubleshooting and struggling with this (and similar) issues (after it had worked flawlessly for almost 2 years), buying a new NUC and a new zigbee dongle, all that was needed was a 1,5m USB extension cable.... :) what a relief!

Smily35, 2025-01-13:
I resolve it by clicking on the yellow button refresh, or the blue button next to it.
In the zigbee2mqtt gui, click on your équipement, in the bottom of the first page, you will find this 2 buttons yellow and blue.
It seems that the yellow button resend binding and configuration to the équipement.
Sorry for my poor english! Have a nice day!

steilz, 2025-01-30:
My experience and solution: 
I was using a Conbee stick without any issues on the front I/O of my server (a standard ATX PC). However, when I plugged it into the back, it started behaving strangely - it could receive data but struggled to send on/off commands to lights, switches, etc. 

Using a 5 cm USB extension was enough to fix the issue completely.

best-74, 2025-02-01:
> For my device (Sonoff ZBMINIL2) it automatically seemed to have a binding to genOnOff seen on the bind tab of the devices page. I removed this by clicking the 'unbind' button. If your device is [...]

This "fixed" my errors.. I had no issues with controlling the devices, but got these errors in the log

Apuru, 2025-02-05:
I'm having the same issue.

Tried updating firmware on my Sonoff ZBDongle-P from 202107 to 2024.
Tried unplugging the adapter and replugging it in after a couple minutes.
Tried USB extension cable (several USB3.0 1feet/2.4feet and USB2.0 2m). 
Tried restarting HA and Z2M.
Tried downgrading Z2M and HA Core.
Tried regenerating the map.

None of the above worked. I'm starting to suspect my adapter is broken. Its able to receive Zigbee info but not send which is weird. I've tried un-pairing and re-pairing a device but the interview fails. 

The error I get when an interview failed is "Error: Interview failed because can not get active endpoints"

And the error I get when trying to turn on/off something already paired is "Data request failed with error: 'No network route' (205)".

JoelHer, 2025-05-06:
I still have this issue. Enabling or disabling availability doesn't help. None of my devices work anymore, I can only receive data. Can we reopen this issue?

`error 2025-05-06 18:33:03z2m: Publish 'set' 'state' to 'Waschmaschine ' failed: 'Error: ZCL command 0x94ec32fffe1ffab8/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Data request failed with error: 'MAC channel access failure' (225))'`

andlinux007, 2025-05-24:
Tried the things above and they also do not work. 
I'm using the SLZB-06 as a coordinator and my guess is that the problem lies with this device.
But I'm still trying more things.

progval, 2025-08-02:
I had the same issue after updating my SLZB-07 from firmware '7.3.0.0 build 131' to '7.4.1.0 build 0', where it wouldn't be able to talk to any of my devices.

Updating Z2M from 2.4.0 to 2.6.0 made it able to talk to half of them (*SonOff* end devices and *Heimann* router, but not *Nous* or *LiXee* routers). I did not try to re-handshake any device.

I plugged a *SonOff* smartplug (so a router) and suddenly *Nous* and *LiXee* devices were reachable. This persists across Z2M restarts, even after I unplugged the *SonOff* smartplug. So... that's an interesting workaround.

Alternatively, re-handshaking the devices might work, but I did not have the opportunity to try.