Koenkk/zigbee2mqtt #24376: [New device support]: ZigBee Smart Thermostat Avatto - brother of EARU TRV06

barc343, 2024-10-17:
### Link

https://pl.aliexpress.com/item/1005006306054040.html?spm=a2g0o.order_list.order_list_main.47.195e1c24WRU4my&gatewayAdapt=glo2pol

### Database entry

{"id":15,"type":"EndDevice","ieeeAddr":"0xa4c138e82a55865a","nwkAddr":24047,"manufId":4417,"manufName":"_TZE200_hvaxb2tc","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_hvaxb2tc","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1729165072293}

### Zigbee2MQTT version

1.40.2

### Comments

This is a copy of https://www.zigbee2mqtt.io/devices/TRV06.html, but distributed by other vendor with name Avatto

### External definition

```shell
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE200_hvaxb2tc',
        },
    ],
    model: 'TRV06',
    vendor: 'Avatto',
        description: 'Thermostatic radiator valve',
        fromZigbee: [tuya.fz.datapoints],
        toZigbee: [tuya.tz.datapoints],
        onEvent: tuya.onEventSetTime,
        configure: tuya.configureMagicPacket,
        exposes: [
            e.child_lock(),
            e.battery_low(),
            e
                .climate()
                .withSetpoint('current_heating_setpoint', 5, 35, 1, ea.STATE_SET)
                .withLocalTemperature(ea.STATE)
                .withSystemMode(['auto', 'heat', 'off'], ea.STATE_SET)
                .withRunningState(['idle', 'heat'], ea.STATE)
                .withLocalTemperatureCalibration(-9, 9, 1, ea.STATE_SET),
            ...tuya.exposes.scheduleAllDays(ea.STATE_SET, 'HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C'),
            e
                .binary('scale_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription(
                    'If the heat sink is not fully opened within ' +
                        'two weeks or is not used for a long time, the valve will be blocked due to silting up and the heat sink will not be ' +
                        'able to be used. To ensure normal use of the heat sink, the controller will automatically open the valve fully every ' +
                        'two weeks. It will run for 30 seconds per time with the screen displaying "Ad", then return to its normal working state ' +
                        'again.',
                ),
            e
                .binary('frost_protection', ea.STATE_SET, 'ON', 'OFF')
                .withDescription(
                    'When the room temperature is lower than 5 °C, the valve opens; when the temperature rises to 8 °C, the valve closes.',
                ),
            e.numeric('error', ea.STATE).withDescription('If NTC is damaged, "Er" will be on the TRV display.'),
        ],
        meta: {
            tuyaDatapoints: [
                [2, 'system_mode', tuya.valueConverterBasic.lookup({auto: tuya.enum(0), heat: tuya.enum(1), off: tuya.enum(2)})],
                [3, 'running_state', tuya.valueConverterBasic.lookup({heat: tuya.enum(0), idle: tuya.enum(1)})],
                [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [5, 'local_temperature', tuya.valueConverter.divideBy10],
                [7, 'child_lock', tuya.valueConverter.lockUnlock],
                [28, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [29, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [30, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [31, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [32, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [33, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [34, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [35, null, tuya.valueConverter.errorOrBatteryLow],
                [36, 'frost_protection', tuya.valueConverter.onOff],
                [39, 'scale_protection', tuya.valueConverter.onOff],
                [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
            ],
        },
    }

module.exports =  definition;
```

### What does/doesn't work with the external definition?

All works fine

dszymczuk, 2024-10-17:
Big thank you for support this device!

I've tried to change the schedule, but the schedule doesn't work.
I've changed values from `06:00/21.0 08:00/16.0 12:00/21.0 14:00/16.0` to `06:00/21.0 08:00/16.0 12:00/21.0 22:42/18.0`.
Initial `Current heating setpoint` is 23*C.
But at `22:42` TRV doesn't change value.

Also, another question.
In description, you have `Schedule for saturday, format: "HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C HH:MM/C"`.
When I tried to add 5th or 6th value, I've got:
`z2m: Publish 'set' 'schedule_wednesday' to '0xa4c138b78b8bdcaa' failed: 'Error: Invalid schedule: there should be 4 transitions'` where `0xa4c138b78b8bdcaa` is my device name.

dszymczuk, 2024-10-18:
Next issue or missing setup.

I also found issue in automatization.

My name of device is `Termostat.Salon`

When I set `Termostat.Salon Temperatura` it takes entity `number.termostat_salon_local_temperature_calibration` instead of `number.termostat_salon_current_heating_setpoint`.

When I've changes entity `Temperatura` to take `termostat_salon_current_heating_setpoint` instead of `termostat_salon_local_temperature_calibration`
I've got validation error `Error: Value 26.0 for number.termostat_salon_current_heating_setpoint is outside valid range -9.0 - 9.0`

So temporary changed to
`.withLocalTemperatureCalibration(5, 35, 1, ea.STATE_SET),`

And...
It changes calibration :D
Currently, I have 47*C :D

### So I can't change from automation perspective `current_heating_setpoint`.

Barley194, 2024-10-19:
Will this device be available in the next release?

vrflyer, 2024-10-19:
regarding the auto or programmed mode issue with temperature jumping to 156 degrees celsius when switching from heating to auto mode I modified my external converter script according to a calculation fix which I found here: https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs/blob/main/avatto_me167.js I adopted the modification to my script so that the calculation for the schedule entries (datapoint 28 to 34) use the fixed routine in 'fixedValueConverter'. This really did the trick for me :-)

vrflyer, 2024-10-19:
... and you have to provide 6 entries per day schedule !

dszymczuk, 2024-10-19:
@Barley194 

> Will this device be available in the next release?

If you want to test it, you can extend your configuration using this manual https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_2-1-extending-the-external-definition

dszymczuk, 2024-10-19:
@vrflyer do you have any idea why automation takes `termostat_salon_local_temperature_calibration` instead of `termostat_salon_current_heating_setpoint` ?

vrflyer, 2024-10-19:
No idea... I didn't see this problem yet. Maybe it is also related to the calculation issue. Did you apply the fix and test again?

⁣Gesendet mit Blue ​

dszymczuk, 2024-10-19:
> No idea... I didn't see this problem yet. Maybe it is also related to the calculation issue. Did you apply the fix and test again? ⁣Gesendet mit Blue ​Am 19. Okt. 2024, 22:16, um 22:16, Damian Szymczuk ***@***.***> schrieb:
> […](#)

It's not related with calculation fixes.
It looks like the wrong entity.

dszymczuk, 2024-10-29:
I found another bug.
I've set up a scheduler for Tuesday, but the scheduler was applied on Sunday.
So days from the scheduler do not match with real days.

amadeo-alex, 2024-10-30:
Looks like those devices come in all sorts of flavours - https://github.com/Koenkk/zigbee2mqtt/issues/20732
I also received the "_TZE200_hvaxb2tc" model and stumbled upon this threads and others.
This implementation appears to be functioning and contain ?more logic? that could potentially fix issues reported by @dszymczuk - https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs
The only adjustment that needs to be done is replacing "_TZE200_p3dbf6qs" with "_TZE200_hvaxb2tc" although I'm yet to test everything.

dszymczuk, 2024-10-30:
@amadeo-alex thank you for the different definition of device. I've updated only the part related to the schedule.

amadeo-alex, 2024-11-03:
Looks like the "_TZE200_hvaxb2tc" has been included in one of the merged PRs - https://github.com/Koenkk/zigbee-herdsman-converters/pull/8151

amadeo-alex, 2024-11-03:
> Looks like the "_TZE200_hvaxb2tc" has been included in one of the merged PRs - [Koenkk/zigbee-herdsman-converters#8151](https://github.com/Koenkk/zigbee-herdsman-converters/pull/8151)
> 
> I'll update the Zigbee2MQTT as two days ago there was a new release with that PR included (zigbee-herdsman-converters update that includes that PR) and report back :)

Works, I removed the external converter and updated the container.
[image]

tykovec, 2024-11-07:
Issue with temperature 156 is still present i just use fix from https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs (thanks for it), if weekdays are correct in schedule i cannot confirm.

dszymczuk, 2024-11-11:
I can see that in this script we have 6 scheduleAllDays. In official, they are 4 days.

How many days they supported?

copernic-us, 2024-11-14:
Well I ordered from Avatto and got one _TZE200_hvaxb2tc and 5 from EARU with _TZE200_rxq4iti9 as  manufacturer. Issue with local temperature spining is present.

tykovec, 2024-11-14:
> Have You had any problems with this fix? Does weekdays work as they should?

Honestly I didn't checked it, I have every day same settings. It should be
easy to check one day after and compare with json. I think that temperature
sensor is also not working well. It is measuring lower temperature that
realy is.

dszymczuk, 2024-11-14:
I think that this part:

```
[28, 'schedule_wednesday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
[29, 'schedule_thursday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
[30, 'schedule_friday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
[31, 'schedule_saturday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
[32, 'schedule_sunday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
[33, 'schedule_monday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
[34, 'schedule_tuesday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],  
```

Works correctly.

copernic-us, 2024-11-14:
Maybe but i've switched it to 4 days and managed to fix schedule.

barc343, 2024-11-21:
> ### Link
> https://pl.aliexpress.com/item/1005006306054040.html?spm=a2g0o.order_list.order_list_main.47.195e1c24WRU4my&gatewayAdapt=glo2pol
>
> ### Database entry [...]

Hi this is updated version with correct temperature and scheduler:

`const exposes = require('zigbee-herdsman-converters/lib/exposes');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const fixedValueConverter = {
    thermostatScheduleDayMultiDP: {
        from: (v) => {
            const schedule = [];
            for (let index = 1; index < 24; index = index + 4) {
                schedule.push(
                    String(parseInt(v[index+0])).padStart(2, '0') + ':' +
                    String(parseInt(v[index+1])).padStart(2, '0') + '/' +
                    // @ts-ignore
                    (parseFloat((v[index+2] << 8) + v[index+3]) / 10.0).toFixed(1),
                );
            }
            return schedule.join(' ');
        },
        to: (v) => {
            const payload = [0];
            const transitions = v.split(' ');
            if (transitions.length != 6) {
                throw new Error('Invalid schedule: there should be 6 transitions');
            }
            for (const transition of transitions) {
                const timeTemp = transition.split('/');
                if (timeTemp.length != 2) {
                    throw new Error('Invalid schedule: wrong transition format: ' + transition);
                }
                const hourMin = timeTemp[0].split(':');
                const hour = parseInt(hourMin[0]);
                const min = parseInt(hourMin[1]);
                const temperature = Math.floor(parseFloat(timeTemp[1]) * 10);
                if (hour < 0 || hour > 24 || min < 0 || min > 60 || temperature < 50 || temperature > 300) {
                    throw new Error('Invalid hour, minute or temperature of: ' + transition);
                }
                payload.push(
                    hour,
                    min,
                    (temperature & 0xff00) >> 8,
                    temperature & 0xff,
                );
            }
            return payload;
        },
    },
    thermostatScheduleDayMultiDPWithDayNumber: (dayNum) => {
        return {
            from: (v) => fixedValueConverter.thermostatScheduleDayMultiDP.from(v),
            to: (v) => {
                const data = fixedValueConverter.thermostatScheduleDayMultiDP.to(v);
                data[0] = dayNum;
                return data;
            },
        };
    }
};

const definition = {
    fingerprint: [
        {
            modelID: 'TS0601',
[... 35 lines cut ...]
            e.numeric('error', ea.STATE).withDescription('If NTC is damaged, "Er" will be on the TRV display.'),
        ],
        meta: {
            tuyaDatapoints: [
                [2, 'system_mode', tuya.valueConverterBasic.lookup({auto: tuya.enum(0), heat: tuya.enum(1), off: tuya.enum(2)})],
                [3, 'running_state', tuya.valueConverterBasic.lookup({heat: tuya.enum(0), idle: tuya.enum(1)})],
                [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
                [5, 'local_temperature', tuya.valueConverter.divideBy10],
                [7, 'child_lock', tuya.valueConverter.lockUnlock],
                [28, 'schedule_wednesday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
                [29, 'schedule_thursday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
                [30, 'schedule_friday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
                [31, 'schedule_saturday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
                [32, 'schedule_sunday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
                [33, 'schedule_monday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
                [34, 'schedule_tuesday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
                [35, null, tuya.valueConverter.errorOrBatteryLow],
                [36, 'frost_protection', tuya.valueConverter.onOff],
                [39, 'scale_protection', tuya.valueConverter.onOff],
                [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
            ],
        },
    }

module.exports =  definition;`

barc343, 2024-11-21:
i updated script by fix from - https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs , now temperature is correct and schedule works (for 4 sets)

barc343, 2024-11-21:
> ### Link
> https://pl.aliexpress.com/item/1005006306054040.html?spm=a2g0o.order_list.order_list_main.47.195e1c24WRU4my&gatewayAdapt=glo2pol
> 
> ### Database entry
> {"id":15,"type":"EndDevice","ieeeAddr":"0xa4c138e82a55865a","nwkAddr":24047,"manufId":4417,"manufName":"_TZE200_hvaxb2tc","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65534":0,"modelId":"TS0601","manufacturerName":"_TZE200_hvaxb2tc","powerSource":3,"zclVersion":3,"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":67,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":217189513},"lastSeen":1729165072293}
> 
> ### Zigbee2MQTT version
> 1.40.2
> 
> ### Comments
> This is a copy of https://www.zigbee2mqtt.io/devices/TRV06.html, but distributed by other vendor with name Avatto
> 
> ### External definition
> ```js
> const exposes = require('zigbee-herdsman-converters/lib/exposes');
> const e = exposes.presets;
> const ea = exposes.access;
> const tuya = require('zigbee-herdsman-converters/lib/tuya');
> 
> const fixedValueConverter = {
>     thermostatScheduleDayMultiDP: {
>         from: (v) => {
>             const schedule = [];
>             for (let index = 1; index < 24; index = index + 4) {
>                 schedule.push(
>                     String(parseInt(v[index+0])).padStart(2, '0') + ':' +
>                     String(parseInt(v[index+1])).padStart(2, '0') + '/' +
>                     // @ts-ignore
>                     (parseFloat((v[index+2] << 8) + v[index+3]) / 10.0).toFixed(1),
>                 );
>             }
>             return schedule.join(' ');
>         },
>         to: (v) => {
>             const payload = [0];
>             const transitions = v.split(' ');
>             if (transitions.length != 6) {
>                 throw new Error('Invalid schedule: there should be 6 transitions');
>             }
>             for (const transition of transitions) {
>                 const timeTemp = transition.split('/');
>                 if (timeTemp.length != 2) {
>                     throw new Error('Invalid schedule: wrong transition format: ' + transition);
>                 }
>                 const hourMin = timeTemp[0].split(':');
>                 const hour = parseInt(hourMin[0]);
>                 const min = parseInt(hourMin[1]);
>                 const temperature = Math.floor(parseFloat(timeTemp[1]) * 10);
>                 if (hour < 0 || hour > 24 || min < 0 || min > 60 || temperature < 50 || temperature > 300) {
>                     throw new Error('Invalid hour, minute or temperature of: ' + transition);
>                 }
>                 payload.push(
>                     hour,
>                     min,
>                     (temperature & 0xff00) >> 8,
>                     temperature & 0xff,
>                 );
>             }
>             return payload;
>         },
>     },
>     thermostatScheduleDayMultiDPWithDayNumber: (dayNum) => {
>         return {
>             from: (v) => fixedValueConverter.thermostatScheduleDayMultiDP.from(v),
>             to: (v) => {
>                 const data = fixedValueConverter.thermostatScheduleDayMultiDP.to(v);
>                 data[0] = dayNum;
>                 return data;
>             },
>         };
[... 48 lines cut ...]
>                 [4, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
>                 [5, 'local_temperature', tuya.valueConverter.divideBy10],
>                 [7, 'child_lock', tuya.valueConverter.lockUnlock],
>                 [28, 'schedule_wednesday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(1)],
>                 [29, 'schedule_thursday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(2)],
>                 [30, 'schedule_friday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(3)],
>                 [31, 'schedule_saturday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(4)],
>                 [32, 'schedule_sunday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(5)],
>                 [33, 'schedule_monday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(6)],
>                 [34, 'schedule_tuesday', fixedValueConverter.thermostatScheduleDayMultiDPWithDayNumber(7)],
>                 [35, null, tuya.valueConverter.errorOrBatteryLow],
>                 [36, 'frost_protection', tuya.valueConverter.onOff],
>                 [39, 'scale_protection', tuya.valueConverter.onOff],
>                 [47, 'local_temperature_calibration', tuya.valueConverter.localTempCalibration2],
>             ],
>         },
>     }
> 
> module.exports =  definition;
> ```
> 
> ### What does/doesn't work with the external definition?
> All works fine

i updated script by fix from - https://github.com/sychu/avatto_me167_TZE200_p3dbf6qs , now temperature is correct and schedule works (for 4 sets)

copernic-us, 2024-11-21:
> @barc343 so 4 or 6 sets are available?
> Code says that for 6 sets, but you said `now temperature is correct and schedule works (for 4 sets)`
> I'm confused 🤔 

I've tried to change schedule to 4 instances but it gives me an error. But managed to fix corresponding days.

barc343, 2024-11-25:
> > @barc343 so 4 or 6 sets are available?
> > Code says that for 6 sets, but you said `now temperature is correct and schedule works (for 4 sets)`
> > I'm confused 🤔 
> 
> I've tried to change schedule to 4 instances but it gives me an error. But managed to fix corresponding days.

Sorry, description changed, on this config, all works in auto mode
[image]

copernic-us, 2024-11-25:
> Sorry, description changed, on this config, all works in auto mode
> [image]

You don't have problems with temperature spining to 156° on auto mode?

approde, 2024-11-26:
Can anyone help me with this? I don't understand the instructions and would like to integrate mine.

Skyr, 2024-12-05:
I got yet another twin, it seems... manufacturer string is "_TZE284_o3x45p96", works with config above (at least at first glance).

closed 2025-02-12 by github-actions[bot]