Koenkk/zigbee2mqtt #24378: [New device support]: Tuya BSD17

conta1988, 2024-10-17:
### Link

https://it.aliexpress.com/item/1005007087155796.html?spm=a2g0o.order_list.order_list_main.99.6e623696jdKYgL&gatewayAdapt=glo2ita

### Database entry

{"id":24,"type":"Router","ieeeAddr":"0x70b3d52b60090f36","nwkAddr":8735,"manufId":4660,"epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":1}},"manuSpecificTuya_3":{"attributes":{"powerOnBehavior":2,"switchType":0}},"genBasic":{"attributes":{"appVersion":100,"zclVersion":2}}},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":100,"zclVersion":2,"interviewCompleted":false,"meta":{},"lastSeen":1729174889514}

### Zigbee2MQTT version

1.22.1

### Comments

i need help, if you can. thank you very much

### External definition

```shell
const definition = {
    zigbeeModel: [''],
    model: '',
    vendor: '',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

it can't work

ChrysMa, 2024-10-20:
Here it is wrongly reported by zigbee2mqtt as SWITCH-ZR03-1 which is not. BSD17 is 2 switch module that looks similar to AVATTO N-ZLWSM01

martin-badin, 2024-11-21:
Any update? I see this device in HA but it is reported wrong. 

[image]

hohodyret, 2024-11-26:
I have the same request. Hoping for a manual fix, until it gets fixed in main branch 🤞

hohodyret, 2024-11-27:
Yes, I would like an update too!
[image]
Firmware: 1.41.0-dev

omarahmed786, 2024-12-17:
Would also love an upd

cpsancha, 2025-03-25:
Any update on this one?

Aida3000, 2025-03-26:
Unfortunately not, as fast as i see  :-/

RHCPNG, 2025-05-28:
Still interested!

the-guitarman, 2025-06-09:
eWelink Modle: BSD17
I press the one and only button at the top side for nearly 5 seconds and it reacts with a switch "click", than the button is blinking for some minutes. But I don't see any interview. I use koenkk/zigbee2mqtt:1.41.0, the log_level is debug and permit_join is true. This is the first zigbee device, where I have no output in the log file while I try to pair it.

555MT, 2025-07-26:
to Aida3000
ordered from Aliexpress same as your picture hereis for the switches

[image]

555MT, 2025-08-01:
physically  wired switch to turn on/off. could be a single switch or "tree way"  as per picture

hecromo, 2025-08-01:
> Physically wired switch to turn on/off. It can be an individual or three-way switch, as shown in the image. 
Not that, home assistant he recognizes it as a plug.

o9qieezda, 2025-08-21:
I have the exact same. Recognized as the wrong device and as such missing full functionality.

hecromo, 2025-08-21:
> Tengo exactamente lo mismo. Lo reconocieron como un dispositivo incorrecto y, por lo tanto, no funciona correctamente.

Necesitamos, una solución.

555MT, 2025-08-24:
S1 and S2 are (optionnal) manual on-off.

hecromo, 2025-08-24:
S1 and S2 are manually turned on and off (optional).

 It refers to the type of switch. In the first image I leave the one with the problem, in the second the same type of device but another model. You know what I'm saying? 
If you notice, it detects it as a plug. 

[image]

[image]

4nd0rn0t, 2025-08-24:
But how does it detect the BSD17 or Zigbee2MQTT module?
For me, it detects it as SWITCH-ZR03-1, and the S1 and S2 functions don't work.

555MT, 2025-08-24:
here is how it is reported on my HA
I have no clue on your issue..

[image]

4nd0rn0t, 2025-08-24:
My problem is that I have the BSD17 model and it detects it as a SWITCH-ZR03-1 model plug and I cannot get the external signal extraction option to work (switch, commutator, etc.) because for Zigbee2MQTT it is a plug base, I think I have explained myself better now, see photos

[image]

[image]

hecromo, 2025-08-24:
This is how it is reported in my HA. I have no idea about his problem.
>
> ![Image] (https://github.com/user-attachments/assets/c2e580e8-cbe2-425c-ace0-d05dc46c2a24) 

 
What model?

hecromo, 2025-08-24:
Yes, I'm the same model.

555MT, 2025-08-25:
is it an original or a copy ?

bitblaster, 2025-09-19:
> Any solution?

I tried writing a custom external converter copying the "switchType" attribute from other devices, but non worked...
So at last I simply rewired the last terminal block to the upper (rest button), a simple mod, and now behaves as a push button.

[image]

hecromo, 2025-09-19:
But manual, or can you handle it from zigbee2mqtt? screenshot

hecromo, 2025-09-19:
> > ¿Alguna solución?
> 
> Intenté escribir un convertidor externo personalizado copiando el atributo "switchType" de otros dispositivos, pero ninguno funcionó... Así que finalmente simplemente recableé el último bloque de terminales al superior (botón de descanso), una modificación simple, y ahora se comporta como un botón pulsador.
> 
> [image]

But manual, or can you handle it from zigbee2mqtt? screenshot

hecromo, 2025-09-20:
To me the s1/s2 push button works manually, I have a problem with it that I can't handle it from zigbee2mqttt

hecromo, 2025-09-20:
> Disculpa, no entiendo tu problema: zigbee2mqtt también detecta este dispositivo como un enchufe SWITCH-ZR03-1. Sin embargo, la función de encendido/apagado funciona correctamente en HA. Lo único que no funcionó es que, por defecto, el dispositivo está configurado para manejar el interruptor externo (conectado a S1/S2) como un interruptor "momentáneo", mientras que yo tengo un interruptor pulsador, así que quería que el dispositivo alternara cada vez que pulsaba el botón. Mi modificación solucionó exactamente este problema (que pensé que también era tuyo).

It works for me with a physical switch, without any modifications. The problem I have is that I have smart lights, and it disconnects them. What I wanted was to be able to manage S1S2 from Home Assistant. I don't know if I'm making myself clear.

bitblaster, 2025-09-20:
> It works for me with a physical switch, without any modifications. The problem I have is that I have smart lights, and it disconnects them. What I wanted was to be able to manage S1S2 from Home Assistant. I don't know if I'm making myself clear.

Oh now I understand, you want the device to send a signal to home assistant when you press the physical switch but without switching locally the relay. This is not possible with this device, unless you are able to reprogram the firmware. The local relay in this device is linked to the S1/S2 switch.
A workaround in your use case is that you connect your smart light directly to the mains voltage, then keep this device powered on without anything on the output but with the S1/S2 input connected to your physical switch. Then you can use the status in your HA automations to control your smart lights.

Koenkk, 2025-10-22:
@robefernandez Refer to the [documentation](https://www.zigbee2mqtt.io/advanced/support-new-devices/02_support_new_tuya_devices.html#fixing-tuya-device-detection) to fix device detection issues.

Koenkk, 2026-02-02:
> does that make it impossible to properly integrate with Z2M?

No it should work, what's not working?

Koenkk, 2026-02-02:
To make it detected as a ZBMINIR2, stop z2m, in the `database.db` change `SWITCH-ZR03-1` to `ZBMINIR2` and see if it works.

uk3k, 2026-03-13:
@Koenkk I double checked today all the ideas since your reply:

- today i removed the cover from one device, the zigbee controller ist labeled "EWL-ZB502-LG"
- it's not sonoff, it's the same BSD17 labed device as all the other people have, got multiple of them from different batches, so i also expect it's not only wrong firmware
- tried to change model/manufacturer/etc in database to several sonoff/tuya devices. Also "faking" it to "_TZ3000_fdxihpp7" doesn't work.
 Still no luck with changing the switch behaviour (Attribute unsupported). 
[image]

Database entry for BSD17:

> {"id":68,"type":"Router","ieeeAddr":"0xa81710b90f0d0000","nwkAddr":21648,"manufId":4742,"manufName":"eWeLink","powerSource":"Mains (single phase)","modelId":"SWITCH-ZR03-1","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":9,"inClusterList":[0,3,4,5,6,4096,64599,64529],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"SWITCH-ZR03-1","manufacturerName":"eWeLink","powerSource":0,"zclVersion":8,"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"1.2.1"}},"genOnOff":{"attributes":{"onOff":0,"startUpOnOff":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029daead6","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":65000,"repChange":1}],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":1,"stackVersion":0,"hwVersion":0,"dateCode":"20210610-BL","swBuildId":"1.2.1","zclVersion":8,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1773385647019}
<

"Expose" should be same as TS001 (_TZ3000_fdxihpp7) Model WHD02(This works as intended):
> {"id":79,"type":"Router","ieeeAddr":"0x70b3d52b600b92bb","nwkAddr":59592,"manufId":4660,"manufName":"_TZ3000_fdxihpp7","powerSource":"Mains (single phase)","modelId":"TS0001","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,6,3,4,5,57345,2820,1794],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"manufacturerName":"_TZ3000_fdxihpp7","zclVersion":2,"appVersion":100,"modelId":"TS0001","powerSource":1}},"genOnOff":{"attributes":{"onOff":1}},"manuSpecificTuya3":{"attributes":{"switchType":0,"powerOnBehavior":2}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0029daead6","endpointID":1}],"configuredReportings":[{"cluster":6,"attrId":0,"minRepIntval":0,"maxRepIntval":3600,"repChange":0}],"meta":{}}},"appVersion":100,"stackVersion":1,"hwVersion":2,"dateCode":"20210625","swBuildId":"500+TZTDQ11Z3101","zclVersion":2,"interviewCompleted":true,"interviewState":"SUCCESSFUL","meta":{"configured":332242049},"lastSeen":1773386070483}

RHCPNG, 2026-09-15:
It would still be nice to have support for this....

fuuman, 2026-09-15:
Would be cool, yes. This device is just a relais without the ability to work with a switch...

nh2, 2026-11-16:
Love it when bots consume human time :(