Koenkk/zigbee2mqtt #24379: [New device support]: SLZB-07p7 (as a router)

Great-Chart, 2024-10-17:
### Link

https://smlight.tech/product/slzb-07p7/

### Database entry

{"id":143,"type":"Router","ieeeAddr":"0x00124b0028f55bd4","nwkAddr":22659,"manufId":0,"manufName":"SMLIGHT","powerSource":"Mains (single phase)","modelId":"SLZB-0xp7","epList":[8,242],"endpoints":{"8":{"profId":260,"epId":8,"devId":255,"inClusterList":[0,3],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"modelId":"SLZB-0xp7","manufacturerName":"SMLIGHT","powerSource":1,"zclVersion":3,"hwVersion":1,"swBuildId":"20240513"}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"hwVersion":1,"swBuildId":"20240513","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1729178221809}

### Zigbee2MQTT version

1.40.2

### Comments

I have the p7 variant of the SM Light (SLZB-07p7) that I have flashed as a router (via https://smlight.tech/flasher/#SLZB-07p7) and it appears to be functioning correctly - but shows as unsupported.  
Somewhat akin to the example for the SLZB-07 covered under https://github.com/Koenkk/zigbee2mqtt/issues/23076; grateful if this can be added as a supported device

### External definition

```shell
const definition = {
    zigbeeModel: ['SLZB-0xp7'],
    model: 'SLZB-0xp7',
    vendor: 'SMLIGHT',
    description: 'Automatically generated definition',
    extend: [],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Device appears to work OK as a router (flashed with FW 20240513)

Koenkk, 2024-10-18:
Added!

Changes will be available in the [dev branch](https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html) in a few hours from now.

Could you also submit a picture for the docs? This can be done by:
1. Create a fork by clicking [here](https://github.com/Koenkk/zigbee2mqtt.io/fork)
2. Go to the `public/images/devices` directory, *Add file* -> *Upload files*
3. Upload the files and press *Commit changes*
4. Press *Contribute* -> *Open pull request* -> update title/description -> *Create pull request*

Make sure that:
- The filename is `SLZB-0xp7.png`
- The size is 512x512
- The background is transparent (use e.g. [Adobe remove background](https://new.express.adobe.com/tools/remove-background))

Great-Chart, 2024-11-06:
Included in Z2M 1.41.0