Koenkk/zigbee2mqtt #24387: all my powered devices (routers) down after HA update

silkyclouds, 2024-10-18:
### What happened?

Hi,

I noticed some slowness when trying to use my zigbee devices this morning, and took that opportunity to update HA to latest version, knowing it would have to be restarted, sometimes, a restart fixes such things...

I now run HA version 2024.10.2, and Z2M Current version: 1.40.2-1. My controller is a sonoff zigbee dongle (the P model) and is up-to-date as well, latest FW flashed on it from @Koenkk )

weird issue, I can see all my "non-router" devices listed. but all the non battery powered ones are gone. here is what I can see in Z2M logs at start : 

````
[08:10:24] INFO: Preparing to start...
[08:10:25] INFO: Socat not enabled
[08:10:28] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-10-18 08:10:44] error: 	zh:zstack:znp: Failed to determine if path is valid: 'Error: spawn udevadm ENOENT'
[2024-10-18 08:12:12] error: 	z2m: Failed to call 'Groups' 'start' (AssertionError [ERR_ASSERTION]: GroupID must be at least 1
    at Function.create (/app/node_modules/zigbee-herdsman/src/controller/model/group.ts:129:15)
    at Controller.createGroup (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:505:22)
    at Zigbee.createGroup (/app/lib/zigbee.ts:409:23)
    at Groups.syncGroupsWithSettings (/app/lib/extension/groups.ts:81:118)
    at Groups.start (/app/lib/extension/groups.ts:52:20)
    at Controller.callExtensions (/app/lib/controller.ts:399:42)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Controller.start (/app/lib/controller.ts:218:9)
    at start (/app/index.js:154:5))
    `````
the logs whill the keep looping on ping failure messages. Restarting Z2M keeps displaying the same error, and I can't get my devices to get back up. 

here is how it looks in the GUI : 

![image](https://github.com/user-attachments/assets/5771b306-65e6-4f3b-ad36-93b8733ddef5)

As you might imagine, this is really, really problematic. Do anyone else encounter the same issue ? any clue about that error message ? 

    

### What did you expect to happen?

my line powered / router devices should show up ! 

### How to reproduce it (minimal and precise)

no idea, it started this morning after latest HA update. 

### Zigbee2MQTT version

1.40.2-1 

### Adapter firmware version

CC1352P2_CC2652P_launchpad_coordinator_20240710

### Adapter

Sonoff dongle (P model)

### Setup

HASS in a VM, passed coordinator, Z2M as an add-on. 

### Debug log

[08:24:14] INFO: Preparing to start...
[08:24:14] INFO: Socat not enabled
[08:24:15] INFO: Starting Zigbee2MQTT...
Starting Zigbee2MQTT without watchdog.
[2024-10-18 08:24:18] error: 	zh:zstack:znp: Failed to determine if path is valid: 'Error: spawn udevadm ENOENT'
[2024-10-18 08:24:18] error: 	z2m: Failed to set permit join to false (--> 'SREQ: AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":1,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,2,2,10,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)'))
[2024-10-18 08:24:19] error: 	z2m: Failed to call 'Groups' 'start' (AssertionError [ERR_ASSERTION]: GroupID must be at least 1
    at Function.create (/app/node_modules/zigbee-herdsman/src/controller/model/group.ts:129:15)
    at Controller.createGroup (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:505:22)
    at Zigbee.createGroup (/app/lib/zigbee.ts:409:23)
    at Groups.syncGroupsWithSettings (/app/lib/extension/groups.ts:81:118)
    at Groups.start (/app/lib/extension/groups.ts:52:20)
    at Controller.callExtensions (/app/lib/controller.ts:399:42)
    at processTicksAndRejections (node:internal/process/task_queues:95:5)
    at Controller.start (/app/lib/controller.ts:218:9)
    at start (/app/index.js:154:5))
[2024-10-18 08:24:25] warning: 	z2m: Failed to ping 'Chambre Raph et Cel - luminaire - tradfri led light' (attempt 1/1, ZCL command 0x90fd9ffffed4466f/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":51290,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":9,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,10,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
[2024-10-18 08:24:27] warning: 	z2m: Failed to ping 'Chambre d'amis - luminaire (led centre) - tradfri led light' (attempt 1/1, ZCL command 0x90fd9ffffe175760/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":47180,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":10,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,11,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
[2024-10-18 08:24:29] warning: 	z2m: Failed to ping 'Chambre d'amis - luminaire (led gauche) - tradfri led light' (attempt 1/1, ZCL command 0x000b57fffeb89236/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":25054,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":11,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,12,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
[2024-10-18 08:24:31] warning: 	z2m: Failed to ping 'Salon - étagère de droite - tradfri led light' (attempt 1/1, ZCL command 0x040d84fffed2e6cd/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":56423,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":12,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,13,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
[2024-10-18 08:24:33] warning: 	z2m: Failed to ping 'Salon - barre son / etc. - prise tuya' (attempt 1/1, ZCL command 0xa4c13801c3d05d22/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":17038,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":13,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,14,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
[2024-10-18 08:24:35] warning: 	z2m: Failed to ping 'Salon - étagère de gauche - tradfri led light' (attempt 1/1, ZCL command 0x84b4dbfffe770a09/1 genBasic.read(["zclVersion"], {"timeout":10000,"disableResponse":false,"disableRecovery":true,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (--> 'SREQ: AF - dataRequest - {"dstaddr":1987,"destendpoint":1,"srcendpoint":1,"clusterid":0,"transid":14,"options":0,"radius":30,"len":5,"data":{"type":"Buffer","data":[16,15,0,0,0]}}' failed with status '(0x11: BUFFER_FULL)' (expected '(0x00: SUCCESS)')))
[2024-10-18 08:24:37]

silkyclouds, 2024-10-18:
One more time, I could not love proxmox more : 

[image]

I will let you know if things get back working after restoring my nightly backup... 

But I think there is something wrong with last HASS / Z2M combo.

silkyclouds, 2024-10-18:
I've rolled back to previous version of HA, same issue, do you think my coordinator died ? Is it possible ??? 

I've check Z2M changelog, and it was updated 2 weeks ago, meaning it worked without any issue for approx 2 weeks, and suddenly stopped working. 

As a reminder, this morning, I COULD act on my bulbs and other devices, but it was crazy "slow". and after a reboot, they were all gone.

silkyclouds, 2024-10-18:
OK, it turned ou that the BUFFER_FULL was due to interferences. 

here is my full story : https://community.home-assistant.io/t/troubleshooting-zigbee2mqtt-issues-with-tp-link-deco-x50-routers-my-experience/783892

Great-Chart, 2024-10-18:
Hi - I've been suffering random drop out issues that I suspect are more interference related than FW related problems with coordinator. Reading your summary (via linked article) can I ask about the process you followed for moving your zigbee network to another channel and how that went for any re-pairing etc?

Aware a recent Z2M update added this feature in Tools/Setting under the advanced tab. Does it just require the add on restarting or HA etc and do the battery / end devices get restored without intervention?

Welcome feedback from your experiences her; including any devices not liking ch25

silkyclouds, 2024-10-18:
> Hi - I've been suffering random drop out issues that I suspect are more interference related than FW related problems with coordinator. Reading your summary (via linked article) can I ask about the [...]

I'd like to tell you removing the coordinator backup file , changing the channel to 25, and rebooting was the solution. But damn, tonight, everything broke again. 

No more zigbee device available tonight.

silkyclouds, 2024-10-19:
This is hell on earth, really. all the devices get back, and will suddenly disappear, with no reason. As all my lights are controller through zigbee buttons, I can't control any lights anymore... @Koenkk  do you have any idea on how I could debug this issue ?

silkyclouds, 2024-10-19:
by the way I've changed the conf in order to lower the amount of updates that way : 
````
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://core-mosquitto:1883
  user: addons
  password: ke2ohphoh9diesi8cee4yek7oongohgahh7reechofaef6tiewezeeShaiChahk4
  keepalive: 60
  reject_unauthorized: true
  version: 4

serial:
  port: >-
    /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0

frontend:
  port: 8099

advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
  log_level: warning
  log_syslog:
    app_name: Zigbee2MQTT
    eol: /n
    host: localhost
    localhost: localhost
    path: /dev/log
    pid: process.pid
    port: 123
    protocol: tcp4
    type: '5424'
  transmit_power: 10
  channel: 25
  adapter_delay: 1000    log_directory: /app/data/log/

device_options:
  legacy: false
  last_seen: false  
availability:
  active:
    timeout: 60  
    timeout: 1200 
````

Koenkk, 2024-10-20:
> FYI, I updated to latest Z-stack FW about a week ago, could it be that there was a regression and that this buffer issue appeared due to this ?

That could be, try reverting to the previous fw

Nerivec, 2024-10-20:
About the two original errors mentioned:

> error: zh:zstack:znp: Failed to determine if path is valid: 'Error: spawn udevadm ENOENT'

Shouldn't be a problem as long as the adapter starts after that, it's just an optional validation during startup.

> error: z2m: Failed to call 'Groups' 'start' (AssertionError [ERR_ASSERTION]: GroupID must be at least 1

Looks like you have a group with ID zero (in all likelihood) in your `configuration.yaml` and/or `database.db`. You might have to dig in and remove it manually, then re-create it with a proper ID (_note: if you don't set an ID at all in the UI (or via MQTT) when creating the group, just the name, it will automatically increment from the previous ID and ensure it is proper._).

silkyclouds, 2024-10-21:
> > FYI, I updated to latest Z-stack FW about a week ago, could it be that there was a regression and that this buffer issue appeared due to this ?
> 
> That could be, try reverting to the previous fw

now that : 

(venv) root@proxmox:~/cc2538-bsl-master# python cc2538-bsl.py -p /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0 -evw CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
Opening port /dev/serial/by-id/usb-Silicon_Labs_Sonoff_Zigbee_3.0_USB_Dongle_Plus_0001-if00-port0, baud 500000
Reading data from CC1352P2_CC2652P_launchpad_coordinator_20230507.hex
Your firmware looks like an Intel Hex file
Connecting to target...
ERROR: Timeout waiting for ACK/NACK after 'Synch (0x55 0x55)'

silkyclouds, 2024-10-21:
> [2024-10-21 08:48:23] error: 	z2m: Error while starting zigbee-herdsman
> [2024-10-21 08:48:23] error: 	z2m: Failed to start zigbee
> [2024-10-21 08:48:23] error: 	z2m: Check https://www.zigbee2mqtt.io/guide/installation/20_zigbee2mqtt-fails-to-start.html for possible solutions
> [2024-10-21 08:48:23] error: 	z2m: Exiting...
> [2024-10-21 08:48:24] error: 	z2m: Error: network commissioning timed out - most likely network with the same panId or extendedPanId already exists nearby (Error: AREQ - ZDO - stateChangeInd after 60000ms
>     at Object.start (/app/node_modules/zigbee-herdsman/src/utils/waitress.ts:59:23)
>     at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:370:31)
>     at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:91:21)
>     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
>     at Zigbee.start (/app/lib/zigbee.ts:69:27)
>     at Controller.start (/app/lib/controller.ts:161:27)
>     at start (/app/index.js:154:5))
>     at ZnpAdapterManager.beginCommissioning (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:372:23)
>     at ZnpAdapterManager.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/manager.ts:91:21)
>     at ZStackAdapter.start (/app/node_modules/zigbee-herdsman/src/adapter/z-stack/adapter/zStackAdapter.ts:158:16)
>     at Controller.start (/app/node_modules/zigbee-herdsman/src/controller/controller.ts:137:29)
>     at Zigbee.start (/app/lib/zigbee.ts:69:27)
>     at Controller.start (/app/lib/controller.ts:161:27)
>     at start (/app/index.js:154:5)

sorry I didn't see your answer on these points. 

I've checked the error and indeed did read koen comment about it being a message I could skip. 

I've also edited my config file where I indeed had a calibration thing, so second error message is not there anymore.

Great-Chart, 2024-10-22:
Hi @silkyclouds - wondered if you'd had improvements with reverting to earlier firmware here?  I've been trying to fathom out issues with my network using the Sonoff Dongle-P and experienced buffer full related issues with 20240710 and worked my way back to 20221226 which seemed the most stable.
But it goes through periods of needing some devices re-paired, some re-interviewed etc and nominally spacing apart the times when the devices get pinged before it stabilises.

I can get a day and a half if lucky but at other times its much less.
Z2M certainly seems to hang when pinging devices (availability @ 15 mins) as if its pre-occupied with a device that's non-responsive temporarily and cannot keep up with the others.  I've removed a number of devices completely and disabled several more to try and improve / vary things but nothing is a step in the right direction as yet.

There are numerous posts covering concerns over firmware and/or reporting devices going offline but nothing seems to have surfaced as yet to indicate what course of action to take.  I had hoped that you might have seen some improvements with the recent revisions you applied for availability / timeout etc

Nerivec, 2024-10-22:
> I also now run into a log of warning messages about WRONG_ROUTE, ... definitely not wonderful.

What's the exact message?

It sounds like interferences might be your issue if you experience the same kind of instability with two completely different adapters and brand new networks.
That, or one (or several) of the devices on your network is messing things up; some are known to be capable of basically bringing down the whole network because of terrible design (most often Tuya, like [this one](https://www.zigbee2mqtt.io/devices/TS0207_repeater.html)).

Did you try running a [network scan](https://github.com/Nerivec/ember-zli/wiki/Stack#scan-network) with the Skyconnect with [Ember ZLI](https://github.com/Nerivec/ember-zli/wiki#installation) to see if you have issues with channel interference?

With `ember`, once you get several hours worth of logs, you can also try to put them through https://nerivec.github.io/z2m-ember-helper/ it might point you in the direction of trouble devices.

silkyclouds, 2024-10-22:
> What's the exact message?
>
> It sounds like interferences might be your issue if you experience the same kind of instability with two completely different adapters and brand new networks. That, or one [...]

This is really helpful. I will gather some hours of logs and pass them through your tool. I'll be off for a few days but this will definitely be helping !

silkyclouds, 2024-10-23:
by the way, I noticed that as soon I do pair a new device I get this error : 

zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=9 Len=24].

The device gets detected correctly most of the time, though.

Nerivec, 2024-10-30:
That WRONG_DIRECTION error is a bit strange sometimes... still trying to get more info on that one. It doesn't appear to behave as intended sometimes.
Do you have a fuller log (`debug`), so I can take a look?

silkyclouds, 2024-10-30:
> That WRONG_DIRECTION error is a bit strange sometimes... still trying to get more info on that one. It doesn't appear to behave as intended sometimes. Do you have a fuller log (`debug`), so I can take a look?

I just did enable debug again. 

Lets wait for a day or two, and i'll send you the debug logs somewhere after that. :) 

thanks !

Nerivec, 2024-10-30:
If you feel up to the task, you can try to modify your install a bit to see if removing the check causes problem or not.
- Go in `/app/node_modules/zigbee-herdsman/dist/adapter/ember/ezsp/`
- Open `ezsp.js` with a text editor (make a backup of it first if you prefer, though reverting is just a matter of running the following 2 steps in reverse)
- Find the line that says `status = enums_1.EzspStatus.ERROR_WRONG_DIRECTION;`
- Change it to `// status = enums_1.EzspStatus.ERROR_WRONG_DIRECTION;`
- Save the file
- Restart Z2M

silkyclouds, 2024-10-30:
Sure, but as I told you, there are LOADS of other warnings and error messages. 

 I've spent several days re-organising my whole HA setup, rebuilding my dashboards, automations, scenes... as I had to re-pair all my devices (I've moved from the zonoff P dongle to a skyconnect I had on a shelf). 

Let me share my debug logs first as I'm afraid there are way more warnings and errors. 

it's pretty weird, as these messages were NOT existing with the sonoff (non ember) coordinator, though it was unstable as well, and seemed to loose the end devices way easier. 

This is a fresh new zigbee network created from scratch. 

[log.log](https://github.com/user-attachments/files/17576099/log.log)

I'm obviously OK to disable any check you want after that check. :)

silkyclouds, 2024-10-30:
I also see a lot of these ones popping up when in debug mode: 

````
zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "38264".
````

silkyclouds, 2024-10-31:
> Ember is more verbose than zStack for sure. zStack doesn't have as many options to report "general errors" like this, so, Z2M just can't know about them. _Hiding these errors would be counterproductive since they are available here; very useful in optimizing networks, and allows for tools like [the one I mentioned before](https://nerivec.github.io/z2m-ember-helper/)._
> 
> Route errors are common enough though (unless you have a very stable, very "uninterfered-with" network). Check with the above tool, once you have a decent size log file (several hours), it should tell you if there are any "particularly offending" devices, points worth taking a closer look at, etc..

Oh true, I totally forgot about your previous message. 

Thing is, I think ember is way way more verbose, indeed : 

[image]

meaning that, when I enter my amount of routers / end devices and load a 10MB file, it actualy only covers a few minutes of logs : 

[image]

I will put    log_rotation: false for a while and then see if I can inject the log again in the webui. but as it created 10MB logs every 5 minutes, I wonder if this will work.

silkyclouds, 2024-10-31:
bythe way, I started playing with ember-zli, but I got scared at that point : 

````
root@proxmox:~# ember-zli router
[2024-10-31 09:56:49.634] info: 	cli: Data folder: /root/ember-zli.
✔ Adapter connection type Serial
✔ Adapter firmware baudrate 115200
✔ Serial port Nabu Casa  usb-Nabu_Casa_SkyConnect_v1.0_329ebaaaed56ed11959652ca5720eef3-if00-port0 (/dev/ttyUSB0)
✔ Flow control Software Flow Control (rtscts=false)
[2024-10-31 09:56:49.843] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-10-31 09:56:49.844] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-10-31 09:56:49.845] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-10-31 09:56:49.852] info: 	zh:ember:uart:ash: Serial port opened
[2024-10-31 09:56:49.853] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-10-31 09:56:49.950] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-10-31 09:56:49.966] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-10-31 09:56:49.967] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-10-31 09:56:49.987] info: 	ember: NCP EZSP protocol version (13) lower than Host. Switched.
[2024-10-31 09:56:49.999] info: 	ember: NCP version: {"ezsp":13,"revision":"7.4.3 [GA]","build":0,"major":7,"minor":4,"patch":3,"special":0,"type":170}
[2024-10-31 09:56:49.000] info: 	cli: Using default stack config.
✔ Menu Join network
✔ A network is present in the adapter. Leave and continue join? no
[2024-10-31 09:56:49.771] info: 	cli: Join cancelled.
````
Could you eventually let me know which commands I should run to avoid resetting the existing network of my skyconnect, and simply scan the environment to see if it interferes with my (single, with only a few IoT devices connected to 2.4ghz) wifi network?

Nerivec, 2024-10-31:
You can use `info` level for the helper, it should drastically reduce the size of your logs. You can also append each file to the previous one, they should follow each other (can check the last line timestamp vs the first line timestamp of the next).

For ember-zli, you used the `router` [command](https://github.com/Nerivec/ember-zli/wiki/Router). You have to use the `stack` command instead https://github.com/Nerivec/ember-zli/wiki/Stack#scan-network

silkyclouds, 2024-11-01:
> You can use `info` level for the helper, it should drastically reduce the size of your logs. You can also append each file to the previous one, they should follow each other (can check the last line [...]

OK, I've ran into issues yesterday which totally screwed my HASS VM, but this morning I've managed to inject info level logs. 

[image]

So far, it keeps telling me I should wait a little more to get relevant stats. I'll update you with an up-to-date log. 

The ping issues are "normal" as it is normal these devices are powered off at the moment. 

But what about these already ? 

Network/Router Errors: 1682

This seems like a lot of errors, right ?

Nerivec, 2024-11-01:
Note: as mentioned on the site, the "ideals" are based on a running network with nearly zero issues, so, for a network with a bit of troubles, it is expected that the numbers be higher.

Are you regularly taking routers offline? That can hugely impact the network. Routers are not meant to be powered off completely as it can disrupts the surrounding neighbors for a while.

Route errors do not necessarily mean you will "see" issues, just that the backbone isn't very strong. Devices may however start looking for new connections, and disrupt things for a little while (which might be perceived as latencies in some cases I guess).

About the logs duration, did you append logs together? It can mess things up if the order of the timestamps is not respected. If not, check the timestamp of the first line, and last line, to be sure the log file matches what you'd expect.

silkyclouds, 2024-11-01:
> Note: as mentioned on the site, the "ideals" are based on a running network with nearly zero issues, so, for a network with a bit of troubles, it is expected that the numbers be higher.
>
> Are you [...]

So, no, I only have two out of all my routers which goes "offline" and this is due to some recent changes. But all the other ones are plugged and powered on 24/7. 

so approx 64 devices acting as routers. 

I also agree that if routes can't be "fixed", and devices search their way to the coordinator, then I could perceive the time needed for a device to find its route as a latency. 

About the log duration, I've simply disabled log rotation :   log_rotation: false. I therefore have one single log. 

here is the new version of the log
[log.log](https://github.com/user-attachments/files/17601699/log.log)

Do you see anything special here that I should investigate to avoid these error messages ? and improve routing ? I always felt that routing in a zigbee network is a serious point of failure, I would expect the devices to stick to their best route once they found the shortest route to the coordinator.

Nerivec, 2024-11-01:
There's a decent spike (x2-3) in errors for 10 and 11AM (10 to 12), can you see anything that could affect it (particularly the failed CCAs, which means interference on the 2.4GHz band, wifi, thread, zigbee, some proprietary wireless -like Logitech Unify-). Could be just increased usage, but checking to be sure.

These are way off on the route errors:
16687 / 0x412F
29145 / 0x71D9
31087 / 0x796F
33037 / 0x810D
38264 / 0x9578

silkyclouds, 2024-11-02:
> There's a decent spike (x2-3) in errors for 10 and 11AM (10 to 12), can you see anything that could affect it (particularly the failed CCAs, which means interference on the 2.4GHz band, wifi, [...]

I think it's all increased usage, because nothing automatic is happening at these hours. 

Also, as I said, my IoT network (that has unfortunately to remain in 2.4ghz, thanks ESPs...) only has a limited set of devices connected. 

By the way, All the problematic outlets (all the ones you listed are tuya plugs) are located in my kitchen, which is actually situated in an annex, placed behind the perimeter wall of my house. The perimeter wall is approximately 1.5 meters wide. I have the impression that this area might be poorly serviced.

Nerivec, 2024-11-02:
I believe you can flash the Dongle-P to router mode, than you could place it near that perimeter wall (but on that side), so it picks up a better signal, and hopefully, relays it better for the other side of the wall.

silkyclouds, 2024-11-03:
> I believe you can flash the Dongle-P to router mode (not that familiar with zStack things), than you could place it near that perimeter wall (but on that side), so it picks up a better signal, and hopefully, relays it better for the other side of the wall.

a few questions though. if I do flash it as a router, what will be the added value of using it as I already do have 66 routers in the house ? 

And would I have to flash it so it can manage ember ?

Nerivec, 2024-11-04:
I figured you maybe did not have a router very close to that wall you mentioned. It takes the old adapter out of the shelf and puts it to use 😉
Once flashed, it will become a regular router device (just like a light bulb, but without the light part 😬), so, it shouldn't matter what coordinator type you have. Its only job becomes routing traffic.

https://github.com/Koenkk/Z-Stack-firmware/tree/master/router/Z-Stack_3.x.0/bin

silkyclouds, 2024-11-04:
> I figured you maybe did not have a router very close to that wall you mentioned. It takes the old adapter out of the shelf and puts it to use 😉 Once flashed, it will become a regular router device [...]

I understand this but back to my comment, I do have several plugs connected behind that wall, and they all do have very good LQI values. I think I really have to go deeper in the analysis of the network using your ember-zli tool.

silkyclouds, 2024-11-04:
there we go : 

````
root@proxmox:~/ember-zli# ember-zli stack
[2024-11-04 18:20:24.170] info: 	cli: Data folder: /root/ember-zli.
✔ Path: /dev/ttyUSB0, Baudrate: 115200, RTS/CTS: false. Use this config? yes
[2024-11-04 18:20:24.374] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-11-04 18:20:24.375] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-04 18:20:24.376] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-11-04 18:20:24.381] info: 	zh:ember:uart:ash: Serial port opened
[2024-11-04 18:20:24.381] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-04 18:20:24.479] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-11-04 18:20:24.495] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-11-04 18:20:24.496] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-11-04 18:20:24.515] info: 	ember: NCP EZSP protocol version (13) lower than Host. Switched.
[2024-11-04 18:20:24.524] info: 	ember: NCP version: {"ezsp":13,"revision":"7.4.3 [GA]","build":0,"major":7,"minor":4,"patch":3,"special":0,"type":170}
✔ Menu Get stack info
[2024-11-04 18:20:24.820] info: 	cli: EmberZNet: 7.4.3 [GA]. EZSP: 13
✔ Menu Get stack info
[2024-11-04 18:20:24.454] info: 	cli: EmberZNet: 7.4.3 [GA]. EZSP: 13
✔ Menu Get stack config (firmware defaults)
✔ Save to file? (Only print if not) no
[2024-11-04 18:20:24.084] info: 	cli: CONFIG.PACKET_BUFFER_HEAP_SIZE = 255.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.NEIGHBOR_TABLE_SIZE = 16.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.APS_UNICAST_MESSAGE_COUNT = 32.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.BINDING_TABLE_SIZE = 32.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.ADDRESS_TABLE_SIZE = 8.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.MULTICAST_TABLE_SIZE = 8.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.ROUTE_TABLE_SIZE = 16.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.DISCOVERY_TABLE_SIZE = 8.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.STACK_PROFILE = 0.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.SECURITY_LEVEL = 5.
[2024-11-04 18:20:24.085] info: 	cli: CONFIG.MAX_HOPS = 30.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.MAX_END_DEVICE_CHILDREN = 32.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.INDIRECT_TRANSMISSION_TIMEOUT = 3000.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.END_DEVICE_POLL_TIMEOUT = 8.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.TX_POWER_MODE = 0.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.DISABLE_RELAY = 0.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.TRUST_CENTER_ADDRESS_CACHE_SIZE = 0.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.SOURCE_ROUTE_TABLE_SIZE = 200.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.FRAGMENT_WINDOW_SIZE = 1.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.FRAGMENT_DELAY_MS = 0.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.KEY_TABLE_SIZE = 12.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.APS_ACK_TIMEOUT = 1600.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.BEACON_JITTER_DURATION = 3.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.PAN_ID_CONFLICT_REPORT_THRESHOLD = 2.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.REQUEST_KEY_TIMEOUT = 0.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.CERTIFICATE_TABLE_SIZE = ZIGBEE_EZSP_ERROR.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.APPLICATION_ZDO_FLAGS = 0.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.BROADCAST_TABLE_SIZE = 64.
[2024-11-04 18:20:24.086] info: 	cli: CONFIG.MAC_FILTER_TABLE_SIZE = 15.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.SUPPORTED_NETWORKS = 1.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.SEND_MULTICASTS_TO_SLEEPY_ADDRESS = 0.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.ZLL_GROUP_ADDRESSES = 0.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.ZLL_RSSI_THRESHOLD = 216.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.MTORR_FLOW_CONTROL = 1.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.RETRY_QUEUE_SIZE = 16.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.NEW_BROADCAST_ENTRY_THRESHOLD = 58.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.TRANSIENT_KEY_TIMEOUT_S = 300.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.BROADCAST_MIN_ACKS_NEEDED = 255.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.TC_REJOINS_USING_WELL_KNOWN_KEY_TIMEOUT_S = 300.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.CTUNE_VALUE = 140.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.ASSUME_TC_CONCENTRATOR_TYPE = 1.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.GP_PROXY_TABLE_SIZE = 5.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.GP_SINK_TABLE_SIZE = 0.
[2024-11-04 18:20:24.087] info: 	cli: CONFIG.END_DEVICE_CONFIGURATION = ZIGBEE_EZSP_ERROR.
[2024-11-04 18:20:24.087] info: 	cli: POLICY.TRUST_CENTER_POLICY = ALLOW_JOINS,ALLOW_UNSECURED_REJOINS.
[2024-11-04 18:20:24.087] info: 	cli: POLICY.BINDING_MODIFICATION_POLICY = DISALLOW_BINDING_MODIFICATION.
[2024-11-04 18:20:24.087] info: 	cli: POLICY.UNICAST_REPLIES_POLICY = HOST_WILL_NOT_SUPPLY_REPLY.
[2024-11-04 18:20:24.088] info: 	cli: POLICY.POLL_HANDLER_POLICY = POLL_HANDLER_IGNORE.
[... 105 lines cut ...]
[2024-11-04 18:20:24.824] error: 	cli: Failed to scan 26 with status=OBJECT_READ.
✔ Menu Backup network
✔ Network backup save file Use default (/root/ember-zli/coordinator_backup.json)
[2024-11-04 18:20:24.673] info: 	cli: Retrieved 0 link keys.
[2024-11-04 18:20:24.697] info: 	cli: Network backup written to '/root/ember-zli/coordinator_backup.json'.
✔ Restart? (If no, exit) yes
[2024-11-04 18:20:24.376] info: 	zh:ember:uart:ash: Port closed.
[... cut ...]
[2024-11-04 18:20:24.488] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-11-04 18:20:24.501] info: 	ember: NCP EZSP protocol version (13) lower than Host. Switched.
[2024-11-04 18:20:24.509] info: 	ember: NCP version: {"ezsp":13,"revision":"7.4.3 [GA]","build":0,"major":7,"minor":4,"patch":3,"special":0,"type":170}
````
I'm not totally sure how to interpret the channel scan. My coordinator is configured to use channel 25 in Z2M. 

as you can see it failed to scan for existing networks. see errors. 

Is there anything going wrong ?

Nerivec, 2024-11-04:
Channel 25 is probably good for you indeed. Good balance between well-supported channel, and lower overall interference. Definitely avoid 11.

Not quite sure about that error. Is it persistent?

> I do have several plugs connected behind that wall

You have good routers on both side of the wall, near the wall, correct? Large walls can sometimes be a problem, so having a router close-by with good signal on each side usually helps. (That's what I meant to say.)

Have you tried the [higher configs 8.0.2 build](https://github.com/Nerivec/silabs-firmware-builder/releases/tag/v2024.6.2-update1)?

silkyclouds, 2024-11-05:
> Channel 25 is probably good for you indeed. Good balance between well-supported channel, and lower overall interference. Definitely avoid 11.
>
> Not quite sure about that error. Is it persistent? [...]

Now this is crazy, I am 100% confident I did pick channel 25 when I flashed and started to configure this all new skyconnect coordinator, but I double checked and can see channel 11 is indeed used in Z2M config. 

What is best practice to change this again ? same as dongle P? removing the coordinator.backup file, changing the channel, and restart z2M ?

Nerivec, 2024-11-05:
Hmm, strange indeed. You can just change the channel in `configuration.yaml`, restart Z2M, and it should broadcast a channel change to the network (most usually follow, might have to wake up a few end devices by triggering them). Should be the same for all adapters now that the ZDO is standard (though only really tested on z-stack and ember).
https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#changing-the-zigbee-channel

silkyclouds, 2024-11-05:
> Hmm, strange indeed. You can just change the channel in `configuration.yaml`, restart Z2M, and it should broadcast a channel change to the network (most usually follow, might have to wake up a few [...]

yeah I did change the channel several times before (using the P-dongle) and I always did remove the coordinator.backup file. I'll do this in the afternoon and will see if all these routing errors keeps displaying.

silkyclouds, 2024-11-05:
I've changed the channel without any issue. 

But today I am pairing my moes devices and now this : 

````
error 2024-11-05 12:11:19z2m: Publish 'set' 'state' to 'Salon - prise VARMBLIXT ' failed: 'Error: ZCL command 0xa4c1388b5595cf19/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"reservedBits":0,"writeUndiv":false}) failed ({"target":54201,"apsFrame":{"profileId":260,"clusterId":6,"sourceEndpoint":1,"destinationEndpoint":1,"options":4416,"groupId":0,"sequence":204},"zclSequence":191,"commandIdentifier":11} timed out after 10000ms)'
error 2024-11-05 12:47:33zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-11-05 12:47:33zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-11-05 12:47:33zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-11-05 12:47:33zh:ember: Adapter fatal error: HOST_FATAL_ERROR
error 2024-11-05 12:47:33z2m: Adapter disconnected, stopping
error 2024-11-05 12:48:55zh:ember:ezsp: ERROR Transaction failure; status=ERROR_WRONG_DIRECTION. Last Frame: [FRAME: ID=52:"SEND_UNICAST" Seq=115 Len=24].
error 2024-11-05 12:50:19zh:ember:uart:ash: Received ERROR from adapter, with code=ERROR_EXCEEDED_MAXIMUM_ACK_TIMEOUT_COUNT.
error 2024-11-05 12:50:19zh:ember:uart:ash: ASH disconnected | Adapter status: ASH_NCP_FATAL_ERROR
error 2024-11-05 12:50:19zh:ember:uart:ash: Error while parsing received frame, status=ASH_NCP_FATAL_ERROR.
error 2024-11-05 12:50:19zh:ember: Adapter fatal error: HOST_FATAL_ERROR
error 2024-11-05 12:50:19z2m: Adapter disconnected, stopping
````

So many errors...

Nerivec, 2024-11-05:
Trying updating to 7.4.4, some users were having weird issues like this with versions before, for some reason, 7.4.4 seemed to fix it. You can also try the firmware I mentioned above, 8.0.2 with higher configs.
Something's definitely not very stable in your setup, but so many factors to take into account... 😅

silkyclouds, 2024-11-06:
> Trying updating to 7.4.4, some users were having weird issues like this with versions before, for some reason, 7.4.4 seemed to fix it. You can also try the firmware I mentioned above, 8.0.2 with [...]

Considering I’m constantly dealing with a barrage of errors—especially routing issues—I think it’s time to try some experimental firmware. If it could help resolve things, I’m all in. Right now, it’s like a daily ritual: some plug or another fails to start in the morning when triggered by my buttons (which are bound through Z2M).

And, every morning, I get the delightful experience of my wife saying, “Oh, that light didn’t come on,” with an unmistakable tone of condescension.

Having things not work is one thing, but enduring that daily “I told you so” is a matter of honor. I need to put an end to it.

I’m using Silicon Labs Flasher. Is there a way to force an update to 8.0.2?

Nerivec, 2024-11-06:
You can use ember-zli, but I believe there's a hidden field somewhere to put a custom URL in that flasher addon too? You can give it a link to github.
https://github.com/NabuCasa/silabs-firmware-builder/releases/download/v2024.10.21/skyconnect_zigbee_ncp_7.4.4.0.gbl

silkyclouds, 2024-11-06:
> You can use ember-zli, but I believe there's a hidden field somewhere to put a custom URL in that flasher addon too? You can give it a link to github. https://github.com/NabuCasa/silabs-firmware-builder/releases/download/v2024.10.21/skyconnect_zigbee_ncp_7.4.4.0.gbl

There is indeed a FW url field in the advances options ! Well seen ! :) 

Would you recommend me to go for 8.0.2 though ? is there any risk ?

Nerivec, 2024-11-06:
Looks like the add-on is outdated. Use universal-silabs-flasher directly (make sure it's up-to-date), or ember-zli.
Try 7.4.4 first, see if it stabilizes a bit.

silkyclouds, 2024-11-06:
> Looks like the add-on is outdated. Use universal-silabs-flasher directly (make sure it's up-to-date), or ember-zli. Try 7.4.4 first, see if it stabilizes a bit.

nope, it's up-to-date: Current version: 0.3.0 

I see no 0.3.1 at all, no way to use it to update to 7.4.4. :(

Nerivec, 2024-11-06:
> nope, it's up-to-date: Current version: 0.3.0

The actual flasher behind it is outdated. I've notified puddly.

Nerivec, 2024-11-06:
You can use [ember-zli](https://github.com/Nerivec/ember-zli/wiki/Bootloader#update-firmware) (it does what you mentioned 😉) or [universal-silabs-flasher](https://github.com/NabuCasa/universal-silabs-flasher) from another machine to flash the firmware you want though.

silkyclouds, 2024-11-06:
> You can use [ember-zli](https://github.com/Nerivec/ember-zli/wiki/Bootloader#update-firmware) (it does what you mentioned 😉) or [universal-silabs-flasher](https://github.com/NabuCasa/universal-silabs-flasher) from another machine to flash the firmware you want though.

Sorry but I've searched on the wiki and didn't find the right command to flash the FW sing ember-zli, can you point me in the right direction eventually ?

silkyclouds, 2024-11-06:
ok I just flashed it : 

````
root@proxmox:~/ember-zli# ember-zli stack
[2024-11-06 14:29:26.273] info: 	cli: Data folder: /root/ember-zli.
✔ Path: /dev/ttyUSB0, Baudrate: 115200, RTS/CTS: false. Use this config? yes
[2024-11-06 14:29:26.758] info: 	zh:ember:ezsp: ======== EZSP starting ========
[2024-11-06 14:29:26.759] info: 	zh:ember:uart:ash: ======== ASH Adapter reset ========
[2024-11-06 14:29:26.760] info: 	zh:ember:uart:ash: RTS/CTS config is off, enabling software flow control.
[2024-11-06 14:29:26.764] info: 	zh:ember:uart:ash: Serial port opened
[2024-11-06 14:29:26.765] info: 	zh:ember:uart:ash: ======== ASH starting ========
[2024-11-06 14:29:26.861] info: 	zh:ember:uart:ash: ======== ASH connected ========
[2024-11-06 14:29:26.879] info: 	zh:ember:uart:ash: ======== ASH started ========
[2024-11-06 14:29:26.879] info: 	zh:ember:ezsp: ======== EZSP started ========
[2024-11-06 14:29:26.898] info: 	ember: NCP EZSP protocol version (13) lower than Host. Switched.
[2024-11-06 14:29:26.907] info: 	ember: NCP version: {"ezsp":13,"revision":"7.4.4 [GA]","build":0,"major":7,"minor":4,"patch":4,"special":0,"type":170}
✔ Menu Get stack info
[2024-11-06 14:29:26.947] info: 	cli: EmberZNet: 7.4.4 [GA]. EZSP: 13
✔ Menu Get network info
[2024-11-06 14:29:26.765] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "48198".
[2024-11-06 14:29:26.807] info: 	cli: Node EUI64=0xe0798dfffe06a3c5 type=COORDINATOR.
[2024-11-06 14:29:26.807] info: 	cli: Network parameters:
[2024-11-06 14:29:26.808] info: 	cli:   - PAN ID: 59415 (0xE817)
[2024-11-06 14:29:26.808] info: 	cli:   - Extended PAN ID: 202,74,231,39,106,233,255,50
[2024-11-06 14:29:26.808] info: 	cli:   - Radio Channel: 25
[2024-11-06 14:29:26.808] info: 	cli:   - Radio Power: 5 dBm
[2024-11-06 14:29:26.808] info: 	cli:   - Preferred Channels: 11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26
[2024-11-06 14:29:26.808] info: 	cli: Network key info:
[2024-11-06 14:29:26.808] info: 	cli:   - Set? yes
[2024-11-06 14:29:26.808] info: 	cli:   - Sequence Number: 0
[2024-11-06 14:29:26.808] info: 	cli:   - Frame Counter: 2981889
[2024-11-06 14:29:26.808] info: 	cli:   - Alt Set? no
[2024-11-06 14:29:26.809] info: 	cli:   - Alt Sequence Number: 0
[2024-11-06 14:29:26.809] info: 	cli: Trust Center link key info:
[2024-11-06 14:29:26.809] info: 	cli:   - Properties: HAS_OUTGOING_FRAME_COUNTER|HAS_PARTNER_EUI64|IS_AUTHORIZED
[2024-11-06 14:29:26.809] info: 	cli:   - Incoming Frame Counter: 0
[2024-11-06 14:29:26.809] info: 	cli:   - Outgoing Frame Counter: 90112
? Restart? (If no, exit) (Y/n) [2024-11-06 14:29:26.810] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27906".
[2024-11-06 14:29:26.816] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55305".
[2024-11-06 14:29:26.822] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "48198".
[2024-11-06 14:29:26.373] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55305"
````
but even in 7.4.4, it instantly starts displaying all these route errors again. 

[image]

````
✔ Menu Scan network
✔ Radio transmit power [-128-127] 5
✔ Type of scan Scan each channel for existing networks
✔ Duration of scan per channel 998 msec
[2024-11-06 14:29:26.371] info: 	cli: Found network:
[2024-11-06 14:29:26.371] info: 	cli:   - PAN ID: 55841
[2024-11-06 14:29:26.371] info: 	cli:   - Ext PAN ID: 17,152,84,92,170,226,11,31
[2024-11-06 14:29:26.371] info: 	cli:   - Channel: 20
[2024-11-06 14:29:26.371] info: 	cli:   - Allowing join: no
[2024-11-06 14:29:26.371] info: 	cli:   - Node RSSI: -90 dBm | LQI: 84
[2024-11-06 14:29:26.371] info: 	cli: Found network:
[2024-11-06 14:29:26.371] info: 	cli:   - PAN ID: 55841
[2024-11-06 14:29:26.372] info: 	cli:   - Ext PAN ID: 17,152,84,92,170,226,11,31
[2024-11-06 14:29:26.372] info: 	cli:   - Channel: 20
[2024-11-06 14:29:26.372] info: 	cli:   - Allowing join: no
[2024-11-06 14:29:26.372] info: 	cli:   - Node RSSI: -82 dBm | LQI: 116
[2024-11-06 14:29:26.372] info: 	cli: Found network:
[2024-11-06 14:29:26.372] info: 	cli:   - PAN ID: 59415
[2024-11-06 14:29:26.372] info: 	cli:   - Ext PAN ID: 202,74,231,39,106,233,255,50
[2024-11-06 14:29:26.372] info: 	cli:   - Channel: 25
[2024-11-06 14:29:26.372] info: 	cli:   - Allowing join: no
[2024-11-06 14:29:26.372] info: 	cli:   - Node RSSI: -62 dBm | LQI: 196
[2024-11-06 14:29:26.372] info: 	cli: Found network:
[... 76 lines cut ...]
[2024-11-06 14:29:26.783] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "55305".
[2024-11-06 14:29:26.822] info: 	cli: Node EUI64=0xe0798dfffe06a3c5 type=COORDINATOR.
[2024-11-06 14:29:26.822] info: 	cli: Network parameters:
[2024-11-06 14:29:26.822] info: 	cli:   - PAN ID: 59415 (0xE817)
[2024-11-06 14:29:26.822] info: 	cli:   - Extended PAN ID: 202,74,231,39,106,233,255,50
[2024-11-06 14:29:26.822] info: 	cli:   - Radio Channel: 25
[2024-11-06 14:29:26.822] info: 	cli:   - Radio Power: 5 dBm
[2024-11-06 14:29:26.823] info: 	cli:   - Preferred Channels: 11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26
[2024-11-06 14:29:26.823] info: 	cli: Network key info:
[2024-11-06 14:29:26.823] info: 	cli:   - Set? yes
[2024-11-06 14:29:26.823] info: 	cli:   - Sequence Number: 0
[2024-11-06 14:29:26.823] info: 	cli:   - Frame Counter: 2985985
[... cut ...]
[2024-11-06 14:29:26.630] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27906".
[2024-11-06 14:29:26.867] info: 	zh:ember:ezsp: Received network/route error ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE for "27906"
````
Again, if I do get network info, I can see ROUTE_ERROR_MANY_TO_ONE_ROUTE_FAILURE all the way. + I wonder why it tells me "preferred" channels in such a large range, as I've picked 25 in Z2M options ??

I'm out of ideas on how to get rid of these errors, really.

silkyclouds, 2024-11-06:
I've google this routing issue a bit, and fell on a french forum post stating the tuya plugs could be so bad at routing they would actually generate the issues. As I do have a ton of ikea bulbs, I'd like t try something crazy. Is there a way to disable the routing feature of all the tuya plugs, and see if these errors keeps displaying ? it's radical, but sometimes, you got to give things a try for the sake of humanity.

Nerivec, 2024-11-07:
https://github.com/Nerivec/ember-zli/wiki/Stack#existing-networks
As noted in the wiki, scanning networks picks up any device around that is communicating, that means all devices in range, even if on the same network. You do appear to have another network nearby (Thread/Matter? Another ZigBee?); seems a bit farther considering the LQIs.

Preferred channels are the channels the device will scan first during network operations (in this case, all).

> a way to disable the routing feature

_Most desired feature from the ZigBee spec, but unfortunately, no..._
Some TuYa routers are indeed terrible.
Some users have the errors, but it does not impact the network "for actual use", so, they've just been ignoring them.
You don't have any TuYa dedicated repeaters, do you? Because [some of these](https://www.zigbee2mqtt.io/devices/TS0207_repeater.html) are actually damaging to the network... they are meant to be used with only TuYa devices, and just ignore the rest... 😓

I'd suggest you try to narrow down an area, or a device (or two) that is generating more errors ("for actual use") and focus on that first. _Trying to "remove all errors" with TuYa devices on the network will be near impossible..._
If you want to be drastic, you could move all TuYa devices to a secondary network (requires a second Z2M, and adapter).

silkyclouds, 2024-11-20:
> https://github.com/Nerivec/ember-zli/wiki/Stack#existing-networks As noted in the wiki, scanning networks picks up any device around that is communicating, that means all devices in range, even if [...]

Well, I do have one of these PoE coordinators from Ukraine. I will install it, start a second instance of Z2M, and pair all the tuya plugs to it. I think it's a good way to see if I keep noticing delays (which I believe may be happening due to bad routing coming from these tuya plugs). It's drastic, really, but it's worth a try as I do have the resources and hardware anyway. the only "annoying" thing is I'll have to run two instances of Z2M to do so, but not such a big deal...at the end of the day. 

Two questions regarding this : 

1. as I am using channel 25 for my current zigbee network, which would be  best channel to use for the other one ?
2. Do I need to run Z2M in a container ? or i there a way to load two instances of Z2M using the HA add-on store ?

Nerivec, 2024-11-20:
You can use https://github.com/Nerivec/ember-zli/wiki/Stack#zigbee2mqtt-onboard-auto-configuration to try to find the best configuration for your environment (make sure you are in the location where the coordinator will stay), it will run a scan and identify the best channel/PAN ID, then setup the network on the adapter, and create a matching `configuration.yaml` for you to start with. You feed it to Z2M (make the necessary changes based on your setup), and Z2M will pick up where `ember-zli` left off!

_Commonly, 20 or 25 are best. If you don't use the onboard from ember-zli, make sure to set [network options](https://www.zigbee2mqtt.io/guide/configuration/zigbee-network.html#network-config) to `GENERATE` so it creates a non-conflicting network (with previous one)._

Bit of a workaround to get multiple Z2M on same HA (see [example](https://community.home-assistant.io/t/solved-mqtt-with-multiple-zigbee2mqtt-instances/667932))... in any case, make sure to setup MQTT properly to differentiate your instances (`base_topic`...), same for Z2M settings (`data_path`...).

I'd recommend you pull Tuya devices one by one (or by groups of model ID+firmware version), and leave some time pass in-between, you may just identify a single device that's causing a huge mess... 😅

silkyclouds, 2024-11-22:
> You can use https://github.com/Nerivec/ember-zli/wiki/Stack#zigbee2mqtt-onboard-auto-configuration to try to find the best configuration for your environment (make sure you are in the location where [...]

Just my two cents but, it would be nice if Z2M could handle several zigbee networks in a single instance ;)

crowbarsolutions, 2024-12-05:
I can also confirm experiencing this. I have a Sonoff Zigbee 3.0 P (CC2652P) running 20240710 and came back home to all my zigbee devices being unresponsive. I have a mix of mostly IKEA devices. 

Tried restarting Z2MQTT, HA, and even my entire server with no change. Trying to issue any commands (permit join all, loading network map, repairing device, interviewing device, etc.) in the Z2MQTT dashboard resulted in the same BUFFER_FULL errors you were getting. I was able to track down the time it happened and couldn't see any errors reported in the logs. I hadn't even done any recent updates or changes to my configuration, it had been running solid since it was released in August, but today decide to die.

After trying everything I could to rescue my zigbee network I eventually flashed 20221226 back onto the dongle, and can confirm that my network returned back to normal.

silkyclouds, 2025-01-03:
@Nerivec as the network ran "kind" of stable lately, I did leave this topic aside for a while. But today, I see the IKEA tretakt devices are on sale (6,99 instead of 9,99). I am seriously tempted to buy 20 of these to replace all the tuya plugs reported as being terrible all over the internet. 

I might then want to re-re-re-re-re-re-re-re-pair my devices, again, using the SLZB06-P7 coordinator I did buy months ago and never took out of its box, due to my lazyness. 

Did you hear anything bad about these IKEA plugs router wise ? or are these safe to go for ? 
And, is there an added value to replace the skyconnect that was running with the SLZB-06-P7 ? better coverage, right ? It has such a larger antenna, I cannot believe the skyconnect could compete ?

Nerivec, 2025-01-03:
The 06p7 is POE, and zStack-based, very different from EmberZNet-based.
I don't use IKEA stuff, can't tell you much, but you can check Github issues: https://github.com/Koenkk/zigbee2mqtt/issues?q=is%3Aissue+is%3Aopen+ikea

silkyclouds, 2025-01-03:
> The 06p7 is POE, and zStack-based, very different from EmberZNet-based. I don't use IKEA stuff, can't tell you much, but you can check Github issues: https://github.com/Koenkk/zigbee2mqtt/issues?q=is%3Aissue+is%3Aopen+ikea

Well, you might remember I was coming from a Sonoff P coordinator (zStack), that had the main topic BUFFER FULL issue, I left it on a shelve and used a zbt-1 I had. Since then I indeed moved to ember, but got a lot more errors, while the LQI's raised incredibly when reported in Z2M, i started to see all these route errors on my network. 

As I do have the 06p7 anyway, and it has way more memory, can be updated directly from its webui, and has PoE, I am wondering if I might "loose" features or anything if I decided to give it a go. 

Now back to the IKEA plugs, I have approx 5 of these plugs, and they work flawlessly. But to be honest, so do the tuya plugs. I simply wonder if they couldn't be the reason I am loosing devices, or notice latency, as so many people do complain about them sucking at routing.

silkyclouds, 2025-01-03:
ah ! well... the plugs I was looking at are the ones that do not monitor power consumption. so it's a no go for me. I'll stick to the tuya plugs for now and will try to save 200 bucks. 

can I eventually run the 06p7 on another channel, using ZHA, and move the tuya plugs to this zigbee network ? for the sake of testing if my main zigbee network becomes more reliable ?

SkIIr, 2025-01-04:
Have exactly same problem (again). It comes up out of nothing and connections of the devices are dying. I indeed have many Tuyas and the TS0207 repeater as well. I will try with the old 20221226 again to see if it helps. Maybe trying with other repeaters but on the other side I have a lot of TSxxx with router function so don't know if before mentioned steps will help. Pitty as Tuya has reasonable prices in general...

SkIIr, 2025-01-04:
Just went to 20221226 and removed all TS207. Instantly about 30 devices back online, hope rest will show up after a while as well. Observing how net will behave next days.

[3 more replies]

closed 2025-04-29 by github-actions[bot]