Koenkk/zigbee2mqtt #24389: [New device support]:

Bart-VanDerGucht, 2024-10-18:
### Link

https://www.becaenergy.com/zh-hans/product/beca-bht-9000wifi%E8%AF%AD%E9%9F%B3%E6%8E%A7%E5%88%B6%E6%B0%B4%E9%87%87%E6%9A%96-%E6%B0%B4%E9%94%85%E7%82%89-%E7%87%83%E6%B0%94%E9%94%85%E7%82%89-%E7%94%B5%E5%8A%A0%E7%83%AD%E5%AE%A4%E5%86%85%E6%B8%A9/

### Database entry

{"id":3,"type":"Router","ieeeAddr":"0xa4c1389e2fc9548e","nwkAddr":50863,"manufId":4417,"manufName":"_TZE204_amybcd8o","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[4,5,61184,0],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65503":"\u0011��.i\u0012��.i\u0012��.i","65506":56,"65508":0,"65534":0,"manufacturerName":"_TZE204_amybcd8o","zclVersion":3,"appVersion":74,"modelId":"TS0601","powerSource":1,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":74,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":332242049},"lastSeen":1729241871008}

### Zigbee2MQTT version

1.40.2

### Comments

Most of the tuyaDatapoints were identified, not to sure about DP 101, 104 and 113 , all other changes made on the device are nicely reported back to Z2M by using fromZigbee : [tuya.fz.datapoints]

i'm stuck at the toZigbee: [tuya.tz.datapoints] part:
current_heating_setpoint needs to be multiplied by 10 before sending
min_temperature needs to be multiplied by 10 before sending
max_temperature needs to be multiplied by 10 before sending

### External definition

```shell
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const modernExtend = require('zigbee-herdsman-converters/lib/modernExtend');
const e = exposes.presets;
const ea = exposes.access;
const tuya = require('zigbee-herdsman-converters/lib/tuya');

const tz_min_temperature_limit = {
        key: ['min_temperature'],
        convertSet: async (entity, key, value, meta) => {
            const temp = Math.round(value*10);
            await tuya.sendDataPointValue(entity, 26, temp);
        },
};

const definition = {
	fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_amybcd8o',
        },
    ],
    model: 'BHT-9000GCLZB',
    vendor: 'Beca',
	description: 'Smart Zigbee Thermostat Beca BHT-9000GCLZB Expert4house Edition',
	fromZigbee: [tuya.fz.datapoints],
	toZigbee: [tuya.tz.datapoints],
	onEvent: tuya.onEventSetTime, // Add this if you are getting no converter for 'commandMcuSyncTime'
    configure: tuya.configureMagicPacket,
	exposes: [
	    e.child_lock(),
	    e.deadzone_temperature().withValueMin(0).withValueMax(5).withValueStep(1),
	    e.max_temperature(),
	    e.min_temperature(),
	    exposes.climate()
	        .withSetpoint('current_heating_setpoint', 10, 35, 0.5, ea.STATE_SET)
	        .withLocalTemperature(ea.STATE)
	        .withLocalTemperatureCalibration(-9, 9, 1, ea.STATE_SET)
	        .withSystemMode(['off', 'heat'], ea.STATE_SET)
	        .withRunningState(['idle', 'heat'], ea.STATE)
			.withPreset(['hold', 'program']),
	    ],
    meta: {
        tuyaDatapoints: [
            [24, 'local_temperature', tuya.valueConverter.divideBy10],
            [40, 'child_lock', tuya.valueConverter.lockUnlock],
            [27, 'local_temperature_calibration', tuya.valueConverter.localTemperatureCalibration],
            [1, 'state', tuya.valueConverter.onOff],
            [2, 'system_mode', tuya.valueConverterBasic.lookup({'off': tuya.enum(0), 'heat': tuya.enum(1),'cool': tuya.enum(2)})],
            [16, 'current_heating_setpoint', tuya.valueConverter.divideBy10],
            [19, 'max_temperature', tuya.valueConverter.divideBy10],
            [26, 'min_temperature', tuya.valueConverter.divideBy10],
            [101, 'unknown_0-99', tuya.valueConverter.raw],
            [36, 'running_state', tuya.valueConverterBasic.lookup({'heat': tuya.enum(0), 'idle': tuya.enum(1)})],
            [103, 'deadzone_temperature', tuya.valueConverter.raw],
            [102, 'sensor', tuya.valueConverterBasic.lookup({'IN': tuya.enum(0),'AL': tuya.enum(1),'OU': tuya.enum(2)})],
            [4, 'eco_mode', tuya.valueConverter.onOff],
            [105, 'eco_temperature', tuya.valueConverter.divideBy10],
            [115, 'standby_brightness', tuya.valueConverter.raw],
            [114, 'backlight_standby', tuya.valueConverter.onOff],
            [113, 'preset', tuya.valueConverterBasic.lookup({'hold': tuya.enum(0), 'program': tuya.enum(1) })],
            [104, 'day_of_the_week', tuya.valueConverterBasic.lookup({'Mon': tuya.enum(1), 'Tue': tuya.enum(2),'Wed': tuya.enum(3), 'Thu': tuya.enum(4), 'Fri': tuya.enum(5), 'Sat': tuya.enum(6),'Sun': tuya.enum(7)}) ],
            [112, 'schedule_monday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [111, 'schedule_tuesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [110, 'schedule_wednesday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [109, 'schedule_thursday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [108, 'schedule_friday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [107, 'schedule_saturday', tuya.valueConverter.thermostatScheduleDayMultiDP],
            [106, 'schedule_sunday', tuya.valueConverter.thermostatScheduleDayMultiDP],
        ],
    },
    extend: [
        // A preferred new way of extending functionality.
    ],
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

Setting the current_heating_setpoint, min_temperature & max_temperature

Bart-VanDerGucht, 2024-12-04:
Received detailed information from supplier https://developer.tuya.com/en/docs/iot/tuya-cloud-universal-serial-port-access-protocol?id=K9hhi0xxtn9cb

nygma2004, 2024-12-14:
If I may add to this issue. I have a few BAC-006AZB units (2 pipe fain coil thermostats).
Z2M recognizes them as BAC-003 and almost everything works. I mean everything I need.
I get the output as:
```
{
   "child_lock":"UNLOCK",
   "current_cooling_setpoint":29,
   "current_heating_setpoint":20,
   "fan_mode":"auto",
   "linkquality":255,
   "local_temperature":24.5,
   "preset":"manual",
   "state":"ON",
   "system_mode":"fan_only",
   "system_mode_device":"fan_only"
}
```
What I notice that sometimes both the `current_cooling_setpoint` and `current_heating_setpoint` gets updated, but for example today, only `current_cooling_setpoint` is changed, even if the mode is set to heating or cooling. The unit does not store a different value for heating and cooling, I don't think it makes much sense have two separate properties. 

I worked with modbus versions of this device and it reports whether the value is open or not. I think that would be important to add to Z2M as well. Based on post above, that appears to be dpID 36 . The current `state` attribute only reports if the thermostat is turned off or turned on.

Bart-VanDerGucht, 2024-12-14:
I can confirm that, on my device, dpID 36 gives the state of the integrated contact switch: heat (or valve Closed) means the contact for the boiler is triggered: boiler gets request to produce warm water for the heating circuit. 
Idle (or valve Open) means that the contact for the boiler is untriggered: boiler gets no request to produce warm water for the heating

nygma2004, 2024-12-14:
OK, I just realized mine is a motorized valve output, so it has a separate "close" and "open" terminal. But I don't think this drives the valve to propotial open/closed state, so it either fully opens or fully closes the valve. So for me the same information would be useful.
How can I do the same?

Bart-VanDerGucht, 2024-12-19:
@nygma2004 the fingerprint in my template is the one for my device:

fingerprint: [
        {
            modelID: 'TS0601',
            manufacturerName: '_TZE204_amybcd8o',
        },
  
I presume if you fill in the one you have it will take precedent over the internal files and the adjustments will be used then.

nygma2004, 2024-12-23:
Thanks a lot @Bart-VanDerGucht, I think I start to understand this a bit better. I changed the `manufacturerName: '_TZE204_dzuqwsyg',` and I can see this now:
[image]
So it is definitely using the new definition.
What I don't understand that in the exposes section, I only have `current_hearing_setpoint` defined, but in MQTT I am still getting this:
```
{
   "child_lock":"UNLOCK",
   "current_cooling_setpoint":29,
   "current_heating_setpoint":35,
   "deadzone_temperature":1,
   "fan_mode":"auto",
   "linkquality":255,
   "local_temperature":29,
   "local_temperature_calibration":0,
   "operation_mode":"holiday",
   "preset":"manual",
   "schedule":"06:00/20 11:30/21 13:30/22 17:30/23 06:00/24 12:00/23 14:30/22 17:30/21 06:00/19 12:30/20 14:30/21 18:30/20",
   "state":"ON",
   "system_mode":"fan_only",
   "system_mode_device":"fan_only"
}
```
I am not sure where the cooling setpoint comes from. It is not in the definition.
Mine is a BAC-006AZB which is a 2 pipe version, so I would also need the settings for heat/cool/fan-only and also the fan speed. These are missing from your definition. Is there a way to see the definition for the supported model, so I can copy/paste the details?
Infortunately ID 36 as `running_state` is shown as N/A and not included in the MQTT message.

Also, I am not seeing anything in the debug log what is stated in the documentation. Or is it because the external definition already has the `exposes` and `tuyaDatapoints` defined, so the raw details are not included in the debug log any more?

nygma2004, 2024-12-23:
Sorry for highjacking your issue with my questions, I started a new discussion on the forum for my specific questions: https://github.com/Koenkk/zigbee2mqtt/discussions/25308

Bart-VanDerGucht, 2025-01-20:
@Koenkk is something else missing to this for it to be used?

closed 2025-03-30 by github-actions[bot]