Koenkk/zigbee2mqtt #24392: [New device support]: water dual water timer

ad7524, 2024-10-18:
### Link

https://www.aliexpress.com/item/1005007541732936.html?spm=a2g0o.productlist.main.13.26edk1Umk1Umi9&algo_pvid=31860577-8f5f-4562-9d01-9f8e58529367&algo_exp_id=31860577-8f5f-4562-9d01-9f8e58529367-6&pdp_npi=4%40dis%21USD%21130.65%2137.73%21%21%21927.26%21267.78%21%402102f6bd17292533481141852e60ae%2112000041227003848%21sea%21QA%210%21ABX&curPageLogUid=EWnkgN4BhpZN&utparam-url=scene%3Asearch%7Cquery_from%3A

### Database entry

Last seen 2 hours ago Availability Online Device type EndDevice Zigbee Model TS0601 Zigbee Manufacturer _TZE284_eaet5qt5 Support status Not supported (how_to_add_support)  IEEE Address 0x30fb10fffe17da41 Network address 0x7C18 / 31768 Power Battery 

### Zigbee2MQTT version

 1.40.2-1

### Comments

request support for this device

### External definition

```shell
i have used below

Last seen
2 hours ago
Availability
Online
Device type
EndDevice
Zigbee Model
TS0601
Zigbee Manufacturer
_TZE284_eaet5qt5
Support status
Not supported (how_to_add_support)

IEEE Address
0x30fb10fffe17da41
Network address
0x7C18 / 31768
Power
Battery
```

### What does/doesn't work with the external definition?

no exposes
info 2024-10-18 15:31:06z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe17da41', payload '{"last_seen":"2024-10-18T15:31:05+03:00","linkquality":58}'
info 2024-10-18 15:31:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe17da41', payload '{"last_seen":"2024-10-18T15:31:09+03:00","linkquality":58}'
info 2024-10-18 15:31:10z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe17da41', payload '{"last_seen":"2024-10-18T15:31:09+03:00","linkquality":58}'

ad7524, 2024-10-19:
the device is not supported. if you have away to make supported with zigbee2mqtt it will be great

ad7524, 2024-10-21:
please can you help 
i have done this and device become supported

const {} = require('zigbee-herdsman-converters/lib/modernExtend');
// Add the lines below
const fz = require('zigbee-herdsman-converters/converters/fromZigbee');
const tz = require('zigbee-herdsman-converters/converters/toZigbee');
const exposes = require('zigbee-herdsman-converters/lib/exposes');
const reporting = require('zigbee-herdsman-converters/lib/reporting');
const ota = require('zigbee-herdsman-converters/lib/ota');
const utils = require('zigbee-herdsman-converters/lib/utils');
const globalStore = require('zigbee-herdsman-converters/lib/store');
const e = exposes.presets;
const ea = exposes.access;

const definition = {
    zigbeeModel: ['TS0601'],
    model: 'TS0601', // Update this with the real model of the device (written on the device itself or product page)
    vendor: '_TZE284_eaet5qt5', // Update this with the real vendor of the device (written on the device itself or product page)
    description: 'smart dual water timer', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
    extend: [],
    fromZigbee: [], // We will add this later
    toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    exposes: [e.battery(), e.temperature(), e.humidity()], // Defines what this device exposes, used for e.g. Home Assistant discovery and in the frontend
};

module.exports = definition;

debug data as below
info 2024-10-21 15:52:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe17da41', payload '{"battery":null,"humidity":null,"last_seen":"2024-10-21T15:52:25+03:00","linkquality":65,"temperature":null}'
info 2024-10-21 15:52:27z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe17da41', payload '{"battery":null,"humidity":null,"last_seen":"2024-10-21T15:52:26+03:00","linkquality":65,"temperature":null}'
info 2024-10-21 15:52:38z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe17da41', payload '{"battery":null,"humidity":null,"last_seen":"2024-10-21T15:52:37+03:00","linkquality":69,"temperature":null}'
info 2024-10-21 15:52:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe17da41', payload '{"battery":null,"humidity":null,"last_seen":"2024-10-21T15:52:45+03:00","linkquality":69,"temperature":null}'
info 2024-10-21 15:52:47z2m:mqtt: MQTT publish: topic 'zigbee2mqtt/0x30fb10fffe17da41', payload '{"battery":null,"humidity":null,"last_seen":"2024-10-21T15:52:46+03:00","linkquality":69,"temperature":null}'

ad7524, 2024-11-12:
i will try latter

closed 2024-11-12 by ad7524

Greeve99, 2024-11-13:
Did you manage to get this working? I have the same device and have made some progress in that I can turn one of the valves on and off but I don’t know any of the other tuyadatapoints for the other valve and functions

mrreebdoog, 2024-11-26:
Check #24927 for a semi working file