Koenkk/zigbee2mqtt #24393: [New device support]: Bosch ISW-ZDL1-WP11G

tiker, 2024-10-18:
### Link

https://commerce.boschsecurity.com/us/en/Wireless-TriTech-Motion-Detector/p/F.01U.283.249/

### Database entry

{"id":29,"type":"EndDevice","ieeeAddr":"0x00155f006c246c9a","nwkAddr":31022,"manufId":0,"manufName":"Bosch","powerSource":"Battery","modelId":"ISW-ZDL1-WP11G","epList":[5],"endpoints":{"5":{"profId":260,"epId":5,"devId":1026,"inClusterList":[0,3,1280,32,1026,1,2821],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"modelId":"ISW-ZDL1-WP11G","manufacturerName":"Bosch","powerSource":3,"zclVersion":1,"appVersion":41,"stackVersion":34,"hwVersion":19,"dateCode":"20140929"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0029dc07aa","zoneState":1}},"genPollCtrl":{"attributes":{"checkinInterval":6480}},"msTemperatureMeasurement":{"attributes":{"measuredValue":1895}},"genPowerCfg":{"attributes":{"batteryVoltage":65}}},"binds":[{"cluster":32,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc07aa","endpointID":1},{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc07aa","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0029dc07aa","endpointID":1}],"configuredReportings":[{"cluster":1026,"attrId":0,"minRepIntval":10,"maxRepIntval":3600,"repChange":100}],"meta":{}}},"appVersion":41,"stackVersion":34,"hwVersion":19,"dateCode":"20140929","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1729258627861,"checkinInterval":1620}

### Zigbee2MQTT version

 1.40.2 commit: e06848d

### Comments

I moved to a new home which had some left-over security stuff.  Found that it was Zigbee based so I'm trying to make use of it.  This first Bosch sensor appears as "Unsupported Device".  The detected stuff mostly works.  Having an issue with the battery voltage where it quickly flashes a value then throws an error.  Not sure what this means.

### External definition

```shell
const {iasZoneAlarm, temperature, battery} = require('zigbee-herdsman-converters/lib/modernExtend');

const definition = {
    zigbeeModel: ['ISW-ZDL1-WP11G'],
    model: 'ISW-ZDL1-WP11G',
    vendor: 'Bosch',
    description: 'Automatically generated definition',
    extend: [iasZoneAlarm({"zoneType":"generic","zoneAttributes":["alarm_1","alarm_2","tamper","battery_low"]}), temperature(), battery()],
    meta: {},
};

module.exports = definition;
```

### What does/doesn't work with the external definition?

On the device page, power source is usually shown as flashing red "%" but sometimes shows as "Battery OK" for a brief moment but an error follows (shown below) and goes back to the "%".  At one point I saw "65" as a value but the unit has brand new batteries (6 * 1.5v AA) so it should be 100%.

I'm still trying to figure this out but posting here in case there's an obvious solution that I have not found yet.

The error:
`Error 2024-10-18 10:31:12z2m: Failed to configure '0x00155f006c246c9a', attempt 1 (Error: ZCL command 0x00155f006c246c9a/5 genPowerCfg.configReport([{"minimumReportInterval":3600,"maximumReportInterval":65000,"reportableChange":10,"attribute":"batteryPercentageRemaining"}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"reservedBits":0,"writeUndiv":false}) failed (Status 'UNSUPPORTED_ATTRIBUTE') at Endpoint.checkStatus (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:350:28) at Endpoint.zclCommand (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:956:26) at Endpoint.configureReporting (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/src/controller/model/endpoint.ts:688:9) at setupAttributes (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:112:13) at configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/lib/modernExtend.ts:157:17) at Object.configure (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/src/index.ts:205:21) at Configure.configure (/opt/zigbee2mqtt/lib/extension/configure.ts:130:13) at EventEmitter.wrappedCallback (/opt/zigbee2mqtt/lib/eventBus.ts:206:17))`

tiker, 2024-12-20:
Don't close yet Mr. Bot.  I'm still figuring this out.

closed 2025-03-01 by github-actions[bot]

Bultreys, 2026-10-20:
I'm running two of these, via a converter made by Claude AI, that work without the battery error by using the battery voltage because the sensor doesn't support battery percentage. Here's the block:

```js
import * as m from 'zigbee-herdsman-converters/lib/modernExtend';

export default {
    zigbeeModel: ['ISW-ZDL1-WP11G'],
    model: 'ISW-ZDL1-WP11G',
    vendor: 'Bosch',
    description: 'Wireless TriTech motion detector (custom: battery voltage)',
    extend: [
        m.identify(),
        // Identical to the generated definition, so existing entities (alarm_1, tamper, battery_low) keep their IDs
        m.iasZoneAlarm({zoneType: 'generic', zoneAttributes: ['alarm_1', 'alarm_2', 'tamper', 'battery_low']}),
        m.temperature(),
        // This sensor supports batteryVoltage but not batteryPercentageRemaining
        m.battery({
            percentage: false,
            percentageReporting: false,
            voltage: true,
            voltageReporting: true,
            // Report at most hourly, at least every 6 hours, or whenever voltage changes by 0.1 V
            voltageReportingConfig: {min: 3600, max: 21600, change: 1},
        }),
    ],
};
```

The upstream version should use occupancy instead of alarm_1, since it's a motion sensor. Happy to open a PR if a maintainer is keen.