Koenkk/zigbee2mqtt #24395: The interface for viewing the status of the Aqara E1 (lumi.airrtc.agl001) device does not display the 'External temperature input' value.

almirus, 2024-10-18:
### What happened?

The interface for viewing the status of the Aqara E1 (lumi.airrtc.agl001) device does not display the 'External temperature input' value.
After clicking the "Refresh" button, an error is displayed. '`Error: Value: 'external_temperature_input' not found in: [system_mode, preset, window_detection, valve_detection, child_lock, away_preset_temperature, calibrated, sensor, schedule, schedule_settings]'
`

[image]

### What did you expect to happen?

The set value should be displayed in this field.

### Zigbee2MQTT version

 1.40.2

### Adapter firmware version

20240710

### Adapter

egony

### Setup

ha os

### Debug log

error 2024-10-18 18:51:21z2m: Publish 'get' 'external_temperature_input' to 'bedroom' failed: 'Error: Value: 'external_temperature_input' not found in: [system_mode, preset, window_detection, valve_detection, child_lock, away_preset_temperature, calibrated, sensor, schedule, schedule_settings]'

sp0ok1, 2024-10-25:
Same problem! No solution found? Isn't this a problem with the device itself?

maxicasa, 2024-11-09:
Same problem with Zigbee2MQTT version 1.41.0. Local temperature seems to work, but I don't understand if the valve works as expected if this value is not set. How can I check if it works fine or not?

closed 2025-02-02 by github-actions[bot]