Koenkk/zigbee2mqtt #24404: Cannot add Sonoff ZBDongle-E

StimpyMGS, 2024-10-19:
### What happened?

I've bought a Sonoff ZBDongle-E but since can't get Z2M to startup. It was working with a zstack adapter.
I have reset the configuration to default and the following lines to the configuration:
`serial:
  port: >-
    /dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240217100851-if00
  adapter: ember`

I can't save this configuration because of the following error:
`Failed to save add-on configuration, does not match regular expression ^zstack|deconz|zigate$. Got {'data_path': '/config/zigbee2mqtt', 'external_converters': [], 'devices': 'devices.yaml', 'groups': 'groups.yaml', 'homeassistant': True, 'permit_join': False, 'mqtt': {'base_topic': 'zigbee2mqtt'}, 'serial': {'port': '/dev/serial/by-id/usb-ITEAD_SONOFF_Zigbee_3.0_USB_Dongle_Plus_V2_20240217100851-if00', 'adapter': 'ember'}, `

### What did you expect to happen?

I expected that, with adding "adapter: ember" it would succesfully start. 

### Zigbee2MQTT version

1.18.1-1

### Adapter firmware version

ncp-uart-hw-v7.4.4.0-zbdonglee-115200.gbl

### Adapter

Sonoff ZBDongle-E

### Setup

Proxmox, HAOS VM, USB device passthrough. Device is recognized by HA and shows up i the hardware list. 

### Debug log

debug log not applicable because z2m does not start

clrmsc, 2024-10-19:
Hi, try this configuration.yaml

homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: mqtt://localhost:1883
serial:
  port: /dev/ttyACM0
  baudrate: 115200
  adapter: ember
  rtscts: false
frontend:
  port: 8081
advanced:
  homeassistant_legacy_entity_attributes: false
  legacy_api: false
  legacy_availability_payload: false
device_options:
  legacy: false

clockbrain, 2024-10-20:
1.18.1 is an ancient version. It was released in 2021 https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.18.1.
Current version of zigbee2mqtt is 1.40.2

Nerivec, 2024-10-20:
What clockbrain said; the version you are using is much too old, it pre-dates the addition of the driver for the Dongle-E, so the configuration doesn't know about `ember` or `ezsp`, hence the failure when saving.
You will have to update to latest version first: https://github.com/Koenkk/zigbee2mqtt/releases

MathijsG, 2024-10-22:
Sounds like you're still on the ancient repo from Daniel Welch (https://github.com/danielwelch/hassio-zigbee2mqtt) that many of us used back in the day before the official Add-on came.

StimpyMGS, 2024-10-22:
*face palm* yes that was it! I added this repo a million years ago when I started with HA. I never would've considered this because "I was on the latest version of Z2M" or at least I thought I was..

closed 2024-10-22 by StimpyMGS